MichaIng/DietPi #6307: Problem with ha-av==10.0

maury77, 2023-04-12:
Required Information
DietPi version | `8.16
Distro version | bullseye
Linux sdomotica 6.1.21-v7+ #1642 SMP Mon Apr  3 17:20:52 BST 2023 armv7l GNU/Linux
SBC model | RPi 3 Model B+ (armv7l)
Power supply used | (EG: 5V 3A)

after upgrade to 

Steps to reproduce

2023-04-12 14:08:22.958 ERROR (SyncWorker_7) [homeassistant.util.package] Unable to install package ha-av==10.0.0: error: subprocess-exited-with-error
  
  × Getting requirements to build wheel did not run successfully.
  │ exit code: 1
  ╰─> [29 lines of output]
      Package libavformat was not found in the pkg-config search path.
      Perhaps you should add the directory containing `libavformat.pc'
      to the PKG_CONFIG_PATH environment variable
      No package 'libavformat' found
      Package libavcodec was not found in the pkg-config search path.
      Perhaps you should add the directory containing `libavcodec.pc'
      to the PKG_CONFIG_PATH environment variable
      No package 'libavcodec' found
      Package libavdevice was not found in the pkg-config search path.
      Perhaps you should add the directory containing `libavdevice.pc'
      to the PKG_CONFIG_PATH environment variable
      No package 'libavdevice' found
      Package libavutil was not found in the pkg-config search path.
      Perhaps you should add the directory containing `libavutil.pc'
      to the PKG_CONFIG_PATH environment variable
      No package 'libavutil' found
      Package libavfilter was not found in the pkg-config search path.
      Perhaps you should add the directory containing `libavfilter.pc'
      to the PKG_CONFIG_PATH environment variable
      No package 'libavfilter' found
      Package libswscale was not found in the pkg-config search path.
      Perhaps you should add the directory containing `libswscale.pc'
      to the PKG_CONFIG_PATH environment variable
      No package 'libswscale' found
      Package libswresample was not found in the pkg-config search path.
      Perhaps you should add the directory containing `libswresample.pc'
      to the PKG_CONFIG_PATH environment variable
      No package 'libswresample' found
      pkg-config could not find libraries ['avformat', 'avcodec', 'avdevice', 'avutil', 'avfilter', 'swscale', 'swresample']
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
error: subprocess-exited-with-error

× Getting requirements to build wheel did not run successfully.
│ exit code: 1
╰─> See above for output.

note: This error originates from a subprocess, and is likely not a problem with pip.

maury77, 2023-04-12:
I install manualy 

apt-get install libavdevice-dev
apt-get install libavfilter-dev
apt-get install libavdevice

I solve it , but is it correct?

Joulinar, 2023-04-12:
according your error message, following packages are required by `ha-av`

```
libswscale-dev
libavutil-dev
libavcodec-dev
libavformat-dev
libavdevice-dev
libavfilter-dev
```

MichaIng, 2023-04-12:
Looks like `libavdevice-dev` is pulling all others as dependencies.