MichaIng/DietPi #6310: [dietpi-services] show service status on summary screen

TDuffinNTU, 2023-04-12:
### Creating a feature request

#### Is your feature request related to a problem? Please describe:
I can't remember if this was the case before the most recent dietpi-services overhaul, but I think that it used to show the status of services on the summary screen, without needing to select a specific service to know its status.

The other columns are usually static and stay at their default settings in my use, but status is the one I usually open the utility to check so I'd like that one to be displayed foremost. It could perhaps replace something like CPU Affinity?

#### Describe the solution you'd like:
(re?)add service status to the dietpi-services summary screen. Potentially replacing CPU affinity if space was needed.

#### Describe alternatives you've considered:
Right now I just open each service I need to check manually

#### Additional context

MichaIng, 2023-04-12:
It does still show it in my case:
```sh
│                                                ●─ Single Service Options ─────────────●                              │
│                              redis-server      : enabled | included | Affinity 0-3                                   │
│                              mariadb           : enabled | included | Affinity 0-3                                   │
│                              php8.2-fpm        : enabled | included | Affinity 0-3                                   │
│                              apache2           : enabled | included | Affinity 0-3                                   │
│                              coturn            : enabled | included | Affinity 0-3                                   │
│                              cron              : enabled | included | Affinity 0-3                                   │
│                              dropbear          : enabled | included | Affinity 0-3                                   │
│                              dietpi-vpn        : disabled | included | Affinity 0-3                                  │
│                              dietpi-cloudshell : disabled | included | Affinity 0-3                                  │
│                              nextcloud_push    : enabled | included | Affinity 0-3                                   │
│                              Add               : Add missing service to DietPi-Services                              │
│                                                ●─ Global Service Options ─────────────●                              │
│                              Stop              : Stop ALL above services                                             │
│                              Restart           : (Re)start ALL enabled services                                     
```
Or do you mean "active/inactive" running state? That wasn't shown here before either and is probably better checked like this:
```sh
dietpi-services status
```

TDuffinNTU, 2023-04-12:
Of course this is up to your discretion, but having a look at dietpi-services.sh I think I might know how to make the change. I'd be happy to open a PR :)

TDuffinNTU, 2023-04-12:
Ah yes, that's the one I meant:) Good spot btw!

MichaIng, 2023-04-12:
Actually I agree that the running state would be better in the overview menu than the CPU affinity. Currently it is estimated separately when the single service menu is entered.

It could be instead obtained as an array in the `Load_Process_Tool_Arrays()` function where also `aSERVICE_MODE[]` (enabled/disabled/masked) is obtained and then updated when starting/stopping/restarting one or all services.