MichaIng/DietPi #6315: CPU TEMP / banner TEMP mismatch - Lenovo M93p

duledxb, 2023-04-15:
### Creating a bug report/issue

#### Required Information
- DietPi version | `cat /boot/dietpi/.version`
- G_DIETPI_VERSION_CORE=8
G_DIETPI_VERSION_SUB=16
G_DIETPI_VERSION_RC=2
G_GITBRANCH='master'
G_GITOWNER='MichaIng'
G_LIVE_PATCH_STATUS[0]='not applicable'

- Distro version | bullseye

- Kernel version | "Linux DietPi 5.10.0-21-amd64 # 1 SMP Debian 5.10.162-1 (2023-01-21) x86_64 GNU/Linux"
- 
- SBC model |  Native PC (x86_64)
Lenovo ThinkCentre M93p, I3-4150, 4GB DDR3, 120GB SSD

- Power supply used | Original power supply
- SD card used | none 

#### Additional Information (if applicable)
- Software title | (EG: Nextcloud)
- Was the software title installed freshly or updated/migrated?
- Can this issue be replicated on a fresh installation of DietPi?

- Bug report ID | `echo $G_HW_UUID`

#### Steps to reproduce
`root@DietPi:~/Downloads# cpu`

 ─────────────────────────────────────────────────────
 DietPi CPU Info
 Use dietpi-config to change CPU / performance options
 ─────────────────────────────────────────────────────
 Architecture |     x86_64
 Temperature  |     27 °C / 80 °F : Who put me in the freezer!
 Governor     |     schedutil

Current Freq    Min Freq   Max Freq
 CPU0         |      2394 MHz      900 MHz    3500 MHz
 CPU1         |      2395 MHz      900 MHz    3500 MHz
 CPU2         |      2395 MHz      900 MHz    3500 MHz
 CPU3         |      2394 MHz      900 MHz    3500 MHz

...

#### Expected behaviour
`root@DietPi:~/Downloads# sensors`
acpitz-acpi-0
Adapter: ACPI interface
temp1:        +27.8°C  (crit = +105.0°C)
temp2:        +29.8°C  (crit = +105.0°C)

coretemp-isa-0000
Adapter: ISA adapter
Package id 0:  +46.0°C  (high = +80.0°C, crit = +100.0°C)
Core 0:        +41.0°C  (high = +80.0°C, crit = +100.0°C)
Core 1:        +46.0°C  (high = +80.0°C, crit = +100.0°C)

#### Actual behaviour

- ...

#### Extra details
```

root@DietPi:~# for i in /sys/class/thermal/thermal_zone[0-9]/temp /sys/class/hwmon/hwmon[0-9]/temp[0-9]_input /sys/devices/platform/coretemp.[0-9]/hwmon/hwmon[0-9]/temp[0-9]_input
do
[[ -e $i ]] && echo "$i : $(<$i)"
done

```
/sys/class/thermal/thermal_zone0/temp : 27800
/sys/class/thermal/thermal_zone1/temp : 29800
/sys/class/thermal/thermal_zone2/temp : 46000
/sys/class/hwmon/hwmon0/temp1_input : 27800
/sys/class/hwmon/hwmon0/temp2_input : 29800
/sys/class/hwmon/hwmon1/temp1_input : 47000
/sys/class/hwmon/hwmon1/temp2_input : 42000
/sys/class/hwmon/hwmon1/temp3_input : 46000
/sys/devices/platform/coretemp.0/hwmon/hwmon1/temp1_input : 47000
/sys/devices/platform/coretemp.0/hwmon/hwmon1/temp2_input : 42000
/sys/devices/platform/coretemp.0/hwmon/hwmon1/temp3_input : 46000
root@DietPi:~# 

`root@DietPi:~# ls -Al /sys/devices/platform/coretemp*`
total 0
lrwxrwxrwx 1 root root    0 Apr 15 13:27 driver -> ../../../bus/platform/drivers/coretemp
-rw-r--r-- 1 root root 4096 Apr 15 13:27 driver_override
drwxr-xr-x 3 root root    0 Apr 15 13:19 hwmon
-r--r--r-- 1 root root 4096 Apr 15 13:27 modalias
drwxr-xr-x 2 root root    0 Apr 15 13:27 power
lrwxrwxrwx 1 root root    0 Apr 15 13:19 subsystem -> ../../../bus/platform
-rw-r--r-- 1 root root 4096 Apr 15 13:27 uevent

```

root@DietPi:~# for i in /sys/devices/platform/coretemp*/hwmon/hwmon*/temp*
do
[[ -e $i ]] && echo "$i : $(<$i)"
done
```

/sys/devices/platform/coretemp.0/hwmon/hwmon1/temp1_crit : 100000
/sys/devices/platform/coretemp.0/hwmon/hwmon1/temp1_crit_alarm : 0
/sys/devices/platform/coretemp.0/hwmon/hwmon1/temp1_input : 45000
/sys/devices/platform/coretemp.0/hwmon/hwmon1/temp1_label : Package id 0
/sys/devices/platform/coretemp.0/hwmon/hwmon1/temp1_max : 80000
/sys/devices/platform/coretemp.0/hwmon/hwmon1/temp2_crit : 100000
/sys/devices/platform/coretemp.0/hwmon/hwmon1/temp2_crit_alarm : 0
/sys/devices/platform/coretemp.0/hwmon/hwmon1/temp2_input : 42000
/sys/devices/platform/coretemp.0/hwmon/hwmon1/temp2_label : Core 0
/sys/devices/platform/coretemp.0/hwmon/hwmon1/temp2_max : 80000
/sys/devices/platform/coretemp.0/hwmon/hwmon1/temp3_crit : 100000
/sys/devices/platform/coretemp.0/hwmon/hwmon1/temp3_crit_alarm : 0
/sys/devices/platform/coretemp.0/hwmon/hwmon1/temp3_input : 46000
/sys/devices/platform/coretemp.0/hwmon/hwmon1/temp3_label : Core 1
/sys/devices/platform/coretemp.0/hwmon/hwmon1/temp3_max : 80000
root@DietPi:~# 

I read tickets #3172, #3412 and #4848 but couldn't find a solution. Tnx.

MichaIng, 2023-04-15:
Many thanks for reporting.

Found the issue and fixed with: https://github.com/MichaIng/DietPi/commit/4807e59
Try this:
```sh
sed -i 's/"\${afp_temperature\[@\]}"/\${afp_temperature\[@\]}/' /boot/dietpi/func/dietpi-globals
. /boot/dietpi/func/dietpi-globals
cpu
```

duledxb, 2023-04-15:
root@DietPi:~# cpu

 ─────────────────────────────────────────────────────
 DietPi CPU Info
 Use dietpi-config to change CPU / performance options
 ─────────────────────────────────────────────────────
 Architecture |     x86_64
 Temperature  |     48 °C / 118 °F : Optimal temperature
 Governor     |     schedutil

 Current Freq    Min Freq   Max Freq
 CPU0         |      2492 MHz      900 MHz    3500 MHz
 CPU1         |      2492 MHz      900 MHz    3500 MHz
 CPU2         |      2494 MHz      900 MHz    3500 MHz
 CPU3         |      2496 MHz      900 MHz    3500 MHz

[ INFO ] DietPi-CPU_info | The current CPU frequency may be affected by processing this script itself.

root@DietPi:~# sensors
acpitz-acpi-0
Adapter: ACPI interface
temp1:        +27.8°C  (crit = +105.0°C)
temp2:        +29.8°C  (crit = +105.0°C)

coretemp-isa-0000
Adapter: ISA adapter
Package id 0:  +49.0°C  (high = +80.0°C, crit = +100.0°C)
Core 0:        +45.0°C  (high = +80.0°C, crit = +100.0°C)
Core 1:        +46.0°C  (high = +80.0°C, crit = +100.0°C)

root@DietPi:~# 

Thank you for the quick response!

closed 2023-04-15 by duledxb

Black6spdZ, 2023-05-13:
doesn't report temp correctly on S905 device running v8.17.2

─────────────────────────────────────────────────────
 DietPi CPU Info
 Use dietpi-config to change CPU / performance options
 ─────────────────────────────────────────────────────
 Architecture |     aarch64
 Temperature  |     WARNING: 4294967295 °C / 7730941163 °F : Reducing the life of your device
 Governor     |     schedutil

                 Current Freq    Min Freq   Max Freq
 CPU0         |      2016 MHz      100 MHz    2016 MHz
 CPU1         |      2016 MHz      100 MHz    2016 MHz
 CPU2         |      2016 MHz      100 MHz    2016 MHz
 CPU3         |      2016 MHz      100 MHz    2016 MHz

[ INFO ] DietPi-CPU_info | The current CPU frequency may be affected by processing this script itself.

MichaIng, 2023-05-13:
If Armbian shows the temp somewhere, then it is obtained with a different mechanism.

Which SBC exactly do you use? And can you show the output of:
```sh
for i in /sys/devices/platform/coretemp*/hwmon/hwmon*/temp*; do [[ -e $i ]] && echo "$i : $(<$i)"; done
```

Black6spdZ, 2023-05-13:
> If Armbian shows the temp somewhere, then it is obtained with a different mechanism.
> 
> Which SBC exactly do you use? And can you show the output of:
> 
> ```shell
> for i in /sys/devices/platform/coretemp*/hwmon/hwmon*/temp*; do [[ -e $i ]] && echo "$i : $(<$i)"; done
> ```

its a generic S905 1g/8g box "uses gxbb-p201 device tree"

no output with that command "no coretemp folder"

MichaIng, 2023-05-15:
Ah sorry, wrong command, please repeat with this one:
```sh
for i in /sys/class/thermal/thermal_zone[0-9]/temp /sys/class/hwmon/hwmon[0-9]/temp[0-9]_input; do [[ -e $i ]] && echo "$i : $(<$i)"; done
```
> generic S905 1g/8g box

It is using the Armbian meson64 "current" kernel, Linux 6.1?
```sh
uname -a
```

Black6spdZ, 2023-05-15:
/sys/class/thermal/thermal_zone0/temp : -1000
/sys/class/hwmon/hwmon0/temp1_input : 4294967295000

Linux DietPi 6.1.27-ophub #1 SMP Mon May  1 13:03:40 EDT 2023 aarch64 GNU/Linux

MichaIng, 2023-05-15:
```
6.1.27-ophub
```
This does not look like an Armbian kernel. Which base image did you use before running the DietPi installer? Or which kernel package did you install? In case you do not know, this should show it:
```sh
dpkg -l | grep linux
```
None of the two only available sysfs sensor nodes report something useful. `-1000` is of course rubbish, `4294967295000` might be just an unusual unit, I thought, but actually there is no power of 1000 giving a reasonable value:
```sh
4294967295000 / 1000 / 1000 / 1000 / 1000 ~= 4.29
```
42 °C would have been a value that makes sense, 4.2 °C however not, and 10 picoseconds won't be the unit. So if this did work on Armbian, then either with an older kernel version (probably Linux 5.15 which got upgraded to 6.1) or not using sysfs at all. Does this return some reasonable value?
```sh
apt install lm-sensors
sensors
```
This is not something we would implement but take it as broken sysfs hwmon API for this particular board, but at least you would have a tool to get the actual value then, or at least one where you can observe temperature changes, while I wouldn't trust the absolute value.

Black6spdZ, 2023-05-15:
this thread seems pertinent https://forum.khadas.com/t/thermal-reading-wrong/168/20

I tried to break into u-boot to issue that SARADC command but apears no USB keyboard support