MichaIng/DietPi #6316: Two USB drives, won't boot?

mernamer, 2023-04-15:
### Creating a bug report/issue

#### Required Information
- DietPi version | CORE=8; SUB=16; RC=2
- Distro version | bullseye 0
- Kernel version |  6.1.21-v8+ #1642 SMP PREEMPT Mon Apr 3 17:24:16 BST 2023 aarch64 GNU/Linux
- SBC model | RPi 3 Model B (aarch64)
- Power supply used | At least 5V 1A, bought a "Raspberry Pi" specific one, can check if needed though
- SD card - NA, using USB drive instead, have two plugged in via power hub.
#### Additional Information (if applicable)
- Software title | NA, boot issue, but running both Nextcloud and Emby on this server (probably need to split, just don't have a second Rpi3 at the moment)
- Was the software title installed freshly or updated/migrated? No, up to date, but recent updates don't seem correlated, but also I don't try to reboot the device that frequently, happens every time I reboot only.
- Can this issue be replicated on a fresh installation of DietPi? Yes, have run into this before, though with a slightly different setup.

- Bug report ID | 8e22bd4e-9ff4-4a3a-8995-20918a3b83f7

#### Steps to reproduce

1. Flash Dietpi to USB drive (connected via power hub)
2. Boot, setup DietPi
3. Add second drive to powered hub.
4. Reboot with both drives plugged in

#### Expected behaviour

- ...Boots, display visible via HDMI / monitor

#### Actual behaviour
- Appears to just not boot at all:
- Dark screen on monitor
- Not pingable
- All ports normally open appear closed
- No relevant logs I can find
- HOWEVER: I can complete the boot normally if I unplug the second harddrive, let it boot, then plug in the second harddrive once boot is complete. This is a pain, but Drive Manager has helped quite a bit, it at least connects to the right place consistently when I plug the second drive in.

#### Extra details

- Reported issue before when using an SD card as the primary drive and plugging in a single external harddrive, it resolved when I followed steps provided (looking for ticket, will update if I find own solution first)
- Previously resolved based on advice to flash dietpi directly to the drive, that worked, recurred when I plugged in a second drive.
- Have a HAT, have not used in this scenario before, but working on getting some additional logs / info that way.

Joulinar, 2023-04-15:
How are the external drives powered? Do they have an own PSU or just connected to USB ports?

mernamer, 2023-04-15:
Sorry, meant to share it's a PSU. Also, following steps previously provided on #5729, since the issue seems similar.

Output from `lsblk -o name,fstype,label,size,ro,type,mountpoint,partuuid,uuid` is this:
`NAME   FSTYPE LABEL   SIZE RO TYPE MOUNTPOINT   PARTUUID                             UUID
sda                 465.7G  0 disk
├─sda1 vfat           128M  0 part /boot        c26b8af2-01                          9041-5D00
└─sda2 ext4         465.6G  0 part /            c26b8af2-02                          14bb0923-da43-46db-919f-a1b300fdc73f
sdb                 465.8G  0 disk
└─sdb1 ext4         465.8G  0 part /mnt/piDrive 23bee6e2-9f23-4916-85bf-a06c9da6c272 5375d312-1184-440e-8589-0969adbaf986`

No errors from `dmesg -l emerg,alert,crit,err`

 Set `initial_turbo=0` via Performance options in dietpi-config, no luck. Already tried a different USB hub, they both work if you only plug in the primary drive, let it boot, then plug in drive 2. Generates some errors on boot, but issues are resolved when both drives are ultimately plugged in.

Followed steps from #5729 to hopefully update logs, digging through now, will update if I find something, also updated bug report:
`dietpi-software uninstall 103
mkdir /var/log/journal
poweroff`

> Attach both drives and let it trying to boot for a while. Then power cycle and unplug the second drive again to allow for boot. > Then check

`journalctl`

In case someone is trying, from last instance, can revert the RAM logging via:
`systemctl stop systemd-journald
rm -R /var/log/journal
systemctl start systemd-journald
dietpi-software install 103
reboot`

MichaIng, 2023-04-15:
Do you have a UART adapter to check early boot messages via serial console?

What kind of USB drives are those? Did you try it with a different (secondary) one and with a different USB cable?

I think we won't be able to definitely debug without a serial console. And it might still be a voltage issue, e.g. if the USB hub is not able to cover quick power peaks when both drives are powering up just when booting the RPi and are not spinning (of those are spinning HDDs) yet before.

The HDD docking station I use here is spinning up the drives as fast as it gets power, so I'm able to boot the RPi ones those are at full speed. If this is possible as well in your case, would be at least worth to test.

mernamer, 2023-04-22:
Recently got UART adapter, going to try serial console.

They're older drives, but did try with other secondary drives, issue seems to persist. Trying an alternate USB hub as well. Won't rule out the voltage issue if it doesn't work, but worth a shot.

Will circle back when have more info.

closed 2025-05-31 by MichaIng