MichaIng/DietPi #6319: git install fails on new dietpi install

thomasmockridge, 2023-04-17:
#### Details:
- Date           | Mon Apr 17 13:18:21 SAST 2023
- DietPi version | v8.16.2 (MichaIng/master)
- Image creator  |
- Pre-image      |
- Hardware       | RPi Zero W (armv6l) (ID=1)
- Kernel version | `Linux pi 6.1.21+ #1642 Mon Apr  3 17:19:14 BST 2023 armv6l GNU/Linux`
- Distro         | bullseye (ID=6,RASPBIAN=1)
- Command        | `apt-get -y --allow-change-held-packages install git`
- Exit code      | 100
- Software title | DietPi-Software
#### Steps to reproduce:

1. install git via dietpi-software
2. ...
#### Expected behaviour:

- git is installed
#### Actual behaviour:

- see below fails.
#### Extra details:

- ...
#### Additional logs:
```
Reading package lists...
Building dependency tree...
Reading state information...
The following additional packages will be installed:
  git-man libcurl3-gnutls liberror-perl libgdbm-compat4 libgdbm6 libperl5.32
  perl perl-modules-5.32
Suggested packages:
  gettext-base git-daemon-run | git-daemon-sysvinit git-doc git-el git-email
  git-gui gitk gitweb git-cvs git-mediawiki git-svn gdbm-l10n perl-doc
  libterm-readline-gnu-perl | libterm-readline-perl-perl make
  libtap-harness-archive-perl
Recommended packages:
  patch less ssh-client netbase
The following NEW packages will be installed:
  git git-man libcurl3-gnutls liberror-perl libgdbm-compat4 libgdbm6
  libperl5.32 perl perl-modules-5.32
0 upgraded, 9 newly installed, 0 to remove and 0 not upgraded.
Need to get 13.5 MB of archives.
After this operation, 74.1 MB of additional disk space will be used.
Get:1 http://raspbian.raspberrypi.org/raspbian bullseye/main armhf perl-modules-5.32 all 5.32.1-4+deb11u2 [2823 kB]
Get:2 http://mirror.liquidtelecom.com/raspbian/raspbian bullseye/main armhf libgdbm6 armhf 1.19-2 [61.7 kB]
Get:3 http://mirror.liquidtelecom.com/raspbian/raspbian bullseye/main armhf libgdbm-compat4 armhf 1.19-2 [44.5 kB]
Get:4 http://raspbian.raspberrypi.org/raspbian bullseye/main armhf libperl5.32 armhf 5.32.1-4+deb11u2 [3373 kB]
Get:5 http://raspbian.raspberrypi.org/raspbian bullseye/main armhf perl armhf 5.32.1-4+deb11u2 [293 kB]
Err:5 http://raspbian.raspberrypi.org/raspbian bullseye/main armhf perl armhf 5.32.1-4+deb11u2
  Error reading from server - read (104: Connection reset by peer) [IP: 197.155.77.1 80]
Get:6 http://raspbian.raspberrypi.org/raspbian bullseye/main armhf libcurl3-gnutls armhf 7.74.0-1.3+deb11u7 [306 kB]
Get:7 http://mirror.liquidtelecom.com/raspbian/raspbian bullseye/main armhf liberror-perl all 0.17029-1 [31.0 kB]
Get:8 http://raspbian.raspberrypi.org/raspbian bullseye/main armhf git-man all 1:2.30.2-1+deb11u2 [1828 kB]
Get:9 http://raspbian.raspberrypi.org/raspbian bullseye/main armhf git armhf 1:2.30.2-1+deb11u2 [4704 kB]
Fetched 13.2 MB in 13s (982 kB/s)
E: Failed to fetch http://mirror.liquidtelecom.com/raspbian/raspbian/pool/main/p/perl/perl_5.32.1-4+deb11u2_armhf.deb  Error reading from server - read (104: Connection reset by peer) [IP: 197.155.77.1 80]
E: Unable to fetch some archives, maybe run apt-get update or try with --fix-missing?

thomasmockridge, 2023-04-17:
Works from command line:
#  apt-get install git
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
The following additional packages will be installed:
  git-man libcurl3-gnutls liberror-perl libgdbm-compat4 libgdbm6 libperl5.32 perl perl-modules-5.32
Suggested packages:
  gettext-base git-daemon-run | git-daemon-sysvinit git-doc git-el git-email git-gui gitk gitweb git-cvs git-mediawiki git-svn
  gdbm-l10n perl-doc libterm-readline-gnu-perl | libterm-readline-perl-perl make libtap-harness-archive-perl
Recommended packages:
  patch less ssh-client netbase
The following NEW packages will be installed:
  git git-man libcurl3-gnutls liberror-perl libgdbm-compat4 libgdbm6 libperl5.32 perl perl-modules-5.32
0 upgraded, 9 newly installed, 0 to remove and 0 not upgraded.
Need to get 293 kB/13.5 MB of archives.
After this operation, 74.1 MB of additional disk space will be used.
Do you want to continue? [Y/n] y
Get:1 http://raspbian.raspberrypi.org/raspbian bullseye/main armhf perl armhf 5.32.1-4+deb11u2 [293 kB]
Fetched 199 kB in 1s (150 kB/s)
debconf: delaying package configuration, since apt-utils is not installed
Selecting previously unselected package perl-modules-5.32.
(Reading database ... 25091 files and directories currently installed.)
Preparing to unpack .../0-perl-modules-5.32_5.32.1-4+deb11u2_all.deb ...
Unpacking perl-modules-5.32 (5.32.1-4+deb11u2) ...
Selecting previously unselected package libgdbm6:armhf.
Preparing to unpack .../1-libgdbm6_1.19-2_armhf.deb ...
Unpacking libgdbm6:armhf (1.19-2) ...
Selecting previously unselected package libgdbm-compat4:armhf.
Preparing to unpack .../2-libgdbm-compat4_1.19-2_armhf.deb ...
Unpacking libgdbm-compat4:armhf (1.19-2) ...
Selecting previously unselected package libperl5.32:armhf.
Preparing to unpack .../3-libperl5.32_5.32.1-4+deb11u2_armhf.deb ...
Unpacking libperl5.32:armhf (5.32.1-4+deb11u2) ...
Selecting previously unselected package perl.
Preparing to unpack .../4-perl_5.32.1-4+deb11u2_armhf.deb ...
Unpacking perl (5.32.1-4+deb11u2) ...
Selecting previously unselected package libcurl3-gnutls:armhf.
Preparing to unpack .../5-libcurl3-gnutls_7.74.0-1.3+deb11u7_armhf.deb ...
Unpacking libcurl3-gnutls:armhf (7.74.0-1.3+deb11u7) ...
Selecting previously unselected package liberror-perl.
Preparing to unpack .../6-liberror-perl_0.17029-1_all.deb ...
Unpacking liberror-perl (0.17029-1) ...
Selecting previously unselected package git-man.
Preparing to unpack .../7-git-man_1%3a2.30.2-1+deb11u2_all.deb ...
Unpacking git-man (1:2.30.2-1+deb11u2) ...
Selecting previously unselected package git.
Preparing to unpack .../8-git_1%3a2.30.2-1+deb11u2_armhf.deb ...
Unpacking git (1:2.30.2-1+deb11u2) ...
Setting up perl-modules-5.32 (5.32.1-4+deb11u2) ...
Setting up libcurl3-gnutls:armhf (7.74.0-1.3+deb11u7) ...
Setting up git-man (1:2.30.2-1+deb11u2) ...
Setting up libgdbm6:armhf (1.19-2) ...
Setting up libgdbm-compat4:armhf (1.19-2) ...
Setting up libperl5.32:armhf (5.32.1-4+deb11u2) ...
Setting up perl (5.32.1-4+deb11u2) ...
Setting up liberror-perl (0.17029-1) ...
Setting up git (1:2.30.2-1+deb11u2) ...
Processing triggers for libc-bin (2.31-13+rpt2+rpi1+deb11u5) ...
root@pi:~#

MichaIng, 2023-04-17:
This was a temporary APT mirror issue, the package downloads fine now. In such cases, often a retry solves it, also since another mirror may be picked.

thomasmockridge, 2023-04-17:
Thanks for the prompt reply - please close issue.

closed 2023-04-17 by MichaIng