MichaIng/DietPi #6322: Update error

Zsombe, 2023-04-18:
Hello!

When i tried upgrade to Bookworm with the following command:
```sudo bash -c "$(curl -sSf 'https://raw.githubusercontent.com/MichaIng/DietPi/dev/.meta/dietpi-bookworm-upgrade')"```
i get a following error:
```[ INFO ] dietpi-bookworm-upgrade | APT update, please wait...
Hit:1 http://raspbian.raspberrypi.org/raspbian bullseye InRelease
Hit:2 https://download.docker.com/linux/raspbian bullseye InRelease
Hit:3 https://apt.sonarr.tv/debian buster InRelease
Hit:4 https://archive.raspberrypi.org/debian buster InRelease
Hit:5 https://download.mono-project.com/repo/debian raspbianbuster InRelease
Hit:6 https://repo.jellyfin.org/debian bullseye InRelease
Ign:7 https://quantum-mirror.hu/mirrors/pub/raspbian bullseye InRelease
Err:8 https://quantum-mirror.hu/mirrors/pub/raspbian bullseye Release
  404  Not Found [IP: 91.120.148.107 443]
Reading package lists...	
E: The repository 'https://quantum-mirror.hu/mirrors/pub/raspbian bullseye Release' does not have a Release file.
[FAILED] dietpi-bookworm-upgrade | APT update
 - Command: apt-get -y -eany update
```
What is the problem?

Joulinar, 2023-04-18:
Looks like a local Raspbian package mirror in Hungary is not available. Maybe it's temporary and you can try again later.

Zsombe, 2023-04-18:
> Looks like a local Raspbian package mirror in Hungary is not available. Maybe it's temporary and you can try again later.

True and thanks for your help! I removed the Hungarian mirror and it's okay now.

MichaIng, 2023-04-18:
What was that mirror for, actually? Since `raspbian.raspberrypi.org` is there as well, it seems to double it? `raspbian.raspberrypi.org` is a CDN/mirror director, so it automatically pulls packages from one of the mirrors close to your location. It may not contain/know all of them.

However, that mirror also seems to work again, at least here access works well and it has a Release file (now): https://quantum-mirror.hu/mirrors/pub/raspbian/raspbian/dists/bullseye/Release

Joulinar, 2023-04-20:
Anything left on this or can it be closed.

Zsombe, 2023-04-21:
Everything is fine, it can be closed. Thanks!

closed 2023-04-21 by Joulinar