MichaIng/DietPi #6329: Update-Upgrade dashboard like dashy

jimmyjohnnetcom, 2023-04-21:
### Creating a feature request

#### Is your feature request related to a problem? Please describe:

- ... NA

#### Describe the solution you'd like:

- ... Maybe add to the home page a like Dashy interface so that we can use DietPi as a dashboard as well that would be great and have features like Dashy .

#### Describe alternatives you've considered:

- ... Layout like Dashy with edit title name, right-click to open source and/or edit. Add your server address or link, add icon from a source that has good icons that you can change or add your own icon.

#### Additional context

- ... Maybe talk with Dashy and ask hey how can we work together to add your Dashy to our DietPi Dashboard.

Joulinar, 2023-04-21:
The request would need to be open at Dashboard own GitHub. But there is already something similar requested https://github.com/ravenclaw900/DietPi-Dashboard/issues/169

jimmyjohnnetcom, 2023-04-21:
I like your dashboard for terms and programs or other things, but yeah maybe that could be first then add Dietpi dashboard and click to go in. hiemdall looks great and it has Widgets, this may be the one I was looking for, like Dashy, has great Widgets. I want to use one of these and I don't know which one, but I was leaning more on Dashy because of the way it works, look and easy, also Widgets.

Joulinar, 2023-04-21:
For Dashy we already have a request open https://github.com/MichaIng/DietPi/discussions/5494