MichaIng/DietPi #6330: Rock Pi 4 - No HDMI output

97carmine, 2023-04-22:
### Creating a bug report/issue

#### Required Information
- DietPi version | 
```
G_DIETPI_VERSION_CORE=8
G_DIETPI_VERSION_SUB=16
G_DIETPI_VERSION_RC=2
G_GITBRANCH='master'
G_GITOWNER='MichaIng'
G_LIVE_PATCH_STATUS[0]='not applicable'
```
- Distro version | `bullseye`
- Kernel version | `Linux DietPi 6.1.11-rockchip64 #23.02.2 SMP PREEMPT Sat Feb 18 00:08:25 UTC 2023 aarch64 GNU/Linux`
- SBC model | `ROCK Pi 4 (aarch64)`
- Power supply used | (EG: 5V 1A RAVpower)
- SD card used | `internal eMMC`

When turning on the Rock Pi 4, there is no HDMI when connecting to a monitor.

But when using an external HDMI capture card, it is observed that the screen is displayed but in a horrible way.

https://user-images.githubusercontent.com/6619414/233780745-3efc14e1-41c9-488c-a811-06f15f6ab31e.mp4

MichaIng, 2023-04-22:
Thanks for reporting. Which model is it exactly, A, B or C, plus or non-plus? Did you try different HDMI screens and HDMI cables?

I see it is now Linux 6.1, which must have been an update from today or yesterday. Most like that is the culprit. So far there is only one older report, but sadly the Armbian SBC maintainers are mostly ignoring the forum: https://forum.armbian.com/topic/27790-rock-pi-4-c-no-hdmi-signal-after-ok-password/

Hmm, actually "current" is still on Linux 5.15: https://github.com/armbian/build/blob/main/config/sources/families/include/rockchip64_common.inc
Did you switch to "edge" kernel intentionally? Please try reverting to "current" if there is no strong reason to use edge.

97carmine, 2023-04-22:
The Model is 4B+, It was in "current" kernel, I switch to "edge" to test.

I tested various monitors and some TVs with different HDMI cables. In some screens show the horrible display and in others the screen don't detect the HDMI output. 

But if you try to start a program like sway or kodi. The screen freezes and if you switch tty, the screen stops recognizing the HDMI output.

MichaIng, 2023-04-22:
Does the kernel report any error?
```sh
dmesg -l 0,1,2,3
```
And do you have another stronger PSU to test with or peripheral/USB devices you can detach to see whether this makes a difference?

97carmine, 2023-04-23:
<details>
<summary>dmesg -l 0,1,2,3</summary>

```
[    2.022305] rockchip-pcie f8000000.pcie: PCIe link training gen1 timeout!
[    3.210825] rk_gmac-dwmac fe300000.ethernet: cannot get clock clk_mac_speed
[    6.420370] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43456-sdio for chip BCM4345/9
[    6.539711] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43456-sdio for chip BCM4345/9
[    6.546845] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM4345/9 wl0: Jun 16 2017 12:38:26 version 7.45.96.2 (66c4e21@sh-git) (r) FWID 01-1813af84
[    8.995130] hdmi-audio-codec hdmi-audio-codec.5.auto: Only one simultaneous stream supported!
[    8.995890] hdmi-audio-codec hdmi-audio-codec.5.auto: ASoC: error at snd_soc_dai_startup on i2s-hifi: -22
[    8.996732]  ff8a0000.i2s-i2s-hifi: soc_pcm_open() failed (-22)
[    8.997510] hdmi-audio-codec hdmi-audio-codec.5.auto: Only one simultaneous stream supported!
[    8.998286] hdmi-audio-codec hdmi-audio-codec.5.auto: ASoC: error at snd_soc_dai_startup on i2s-hifi: -22
[    8.999128]  ff8a0000.i2s-i2s-hifi: soc_pcm_open() failed (-22)
[    9.000506] hdmi-audio-codec hdmi-audio-codec.5.auto: Only one simultaneous stream supported!
[    9.001270] hdmi-audio-codec hdmi-audio-codec.5.auto: ASoC: error at snd_soc_dai_startup on i2s-hifi: -22
[    9.002117]  ff8a0000.i2s-i2s-hifi: soc_pcm_open() failed (-22)
[    9.003044] hdmi-audio-codec hdmi-audio-codec.5.auto: Only one simultaneous stream supported!
[    9.003801] hdmi-audio-codec hdmi-audio-codec.5.auto: ASoC: error at snd_soc_dai_startup on i2s-hifi: -22
[    9.004647]  ff8a0000.i2s-i2s-hifi: soc_pcm_open() failed (-22)
```
</details>

I don't thing it related with PSU because it worked before.

I can't say when it was broken because I use it mainly as server, but occasionally I use desktop.

MichaIng, 2023-04-23:
```
hdmi-audio-codec hdmi-audio-codec.5.auto: Only one simultaneous stream supported!
```
That error is strange, but indeed doesn't seem to be directly related.

> I don't thing it related with PSU because it worked before.

"Worked before" or "Works with Radxa's Debian image" is not a guarantee that the PSU is sufficiently stable. With each bootloader and/or kernel upgrade, the timing of hardware initialisation and device detection can change, and the way devices are configured/operated.

Do you have a GUI application installed or aim to do so, to be able to test whether a lower screen resolution stabilises the image?

The DRM resolution can be also enforced kernel-wise, but I'm not sure whether the text console uses DRM or a framebuffer device. At least it could be tested. First check for the name of the HDMI node:
```sh
for i in /sys/class/drm/*; do [[ -e $i/status ]] && echo "$i: $(<$i/status)"; done
```
It might be something like `HDMI-A-1`. Then add e.g. the following to the end (space separated) of the `extraargs=` line in `/boot/dietpiEnv.txt`:
```
video=HDMI-A-1:1024x768@60
```
to enforce resolution and frequency. The whole line might look like this:
```sh
extraargs=net.ifnames=0 video=HDMI-A-1:1024x768@60
```

97carmine, 2023-04-23:
Wow, it seems that I was a little wrong, yes it starts the program (in this case sway) but I have to disconnect and reconnect the HDMI cable to visualize the result.

```
dietpi@DietPi:~$ for i in /sys/class/drm/*; do [[ -e $i/status ]] && echo "$i: $(<$i/status)"; done
/sys/class/drm/card0-HDMI-A-1: connected
```

Setting the DRM resolution has had no effect. I've tried other, more powerful PSUs but the result is equal.

https://user-images.githubusercontent.com/6619414/233845003-a28d5f36-f70d-49ab-ac79-ebc19b5cabb0.mp4

MichaIng, 2023-04-23:
If DRM is not used for the console, the console size can be changed with `fbset`. But it might not change the screen resolution accordingly, so the console either be a smaller frame in the middle of the screen or larger and parts of it outside of the screen.
```sh
apt install fbset
fbset -xres 640 -yres 480
```

So you mean with an X11 session it does work, at last when re-attaching the HDMI cable? And it does work regardless which resolution you apply, e.g. with `xrandr` (which only works if an X11 session is running).

97carmine, 2023-04-23:
I have tried with sway (wayland), with Kodi I have not had luck.
```
dietpi@DietPi:~$ swaymsg -t get_outputs
[
  {
    "id": 3,
    "name": "HDMI-A-1",
    "rect": {
      "x": 0,
      "y": 0,
      "width": 1024,
      "height": 768
    },
    "focus": [
      4
    ],
    "border": "none",
    "current_border_width": 0,
    "layout": "output",
    "orientation": "none",
    "percent": 1.0,
    "window_rect": {
      "x": 0,
      "y": 0,
      "width": 0,
      "height": 0
    },
    "deco_rect": {
      "x": 0,
      "y": 0,
      "width": 0,
      "height": 0
    },
    "geometry": {
      "x": 0,
      "y": 0,
      "width": 0,
      "height": 0
    },
    "window": null,
    "urgent": false,
    "marks": [
    ],
    "fullscreen_mode": 0,
    "nodes": [
    ],
    "floating_nodes": [
    ],
    "sticky": false,
    "type": "output",
    "active": true,
    "dpms": true,
    "primary": false,
    "make": "<Unknown>",
    "model": "<Unknown>",
    "serial": "",
    "scale": 1.0,
    "scale_filter": "nearest",
    "transform": "normal",
    "adaptive_sync_status": "disabled",
    "current_workspace": "1",
    "modes": [
      {
        "width": 800,
        "height": 600,
        "refresh": 56250
      },
      {
        "width": 800,
        "height": 600,
        "refresh": 60317
      },
      {
        "width": 1024,
        "height": 768,
        "refresh": 60004
      }
    ],
    "current_mode": {
      "width": 1024,
      "height": 768,
      "refresh": 60004
    },
    "max_render_time": 0,
    "focused": true,
    "subpixel_hinting": "unknown"
  }
]
```

MichaIng, 2023-04-23:
Okay sway seems to use DRM and respects the resolution defined via `video=` command-line argument. Kodi should actually do the same when started from console or explicitly `kodi-gbm`. But it has own resolution settings, so probably it overrides it, causing the issue.

Does with work with `fbset` for the console?

97carmine, 2023-04-24:
I don't see `fbset` working

[image]

MichaIng, 2023-04-24:
This will only have an effect for the naked console, not any terminal emulator, X session or such.

97carmine, 2023-04-24:
The same thing happens, nothing changes

MichaIng, 2023-04-26:
For my clarification: It does still only work at all with the "external HDMI capture card" (USB-HDMI adapter, I assume?), not with the onboard HDMI port?

The flickering image might then be an issue with the adapter itself, or the USB capabilities of the ROCK 4.

Did this work earlier, or with a different OS?

Some things to test:
- `legacy` kernel
- Test with an official Armbian image: https://dl.armbian.com/rockpi-4b/archive/
   If this does not work either, it can be reported at the Armbian bug tracker: https://www.armbian.com/bugs/
   But as mentioned already, this seems to be mostly ignored.
- Test with an unmodified mainline kernel build. Linux 6.3 was just released: https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git/?h=linux-rolling-stable

97carmine, 2023-05-02:
After installing DietPi again, and installing other OS like Armbian and LibreELEC with the same result.

I conclude that it is a hardware problem. I'm closing because it's not a DietPi problem

closed 2023-05-02 by 97carmine