MichaIng/DietPi #6332: Odroid C1 hangs on reboot / shutdown now -r

gociii, 2023-04-23:
### Creating a bug report/issue

#### Required Information
- DietPi version |
G_DIETPI_VERSION_CORE=8
G_DIETPI_VERSION_SUB=16
G_DIETPI_VERSION_RC=2
G_GITBRANCH='master'
G_GITOWNER='MichaIng'
G_LIVE_PATCH_STATUS[0]='not applicable'

- Distro version | bullseye
- Kernel version | Linux ratchet 5.15.93-meson #23.02.2 SMP Fri Feb 17 22:38:47 UTC 2023 armv7l GNU/Linux
- SBC model | Odroid C1 (armv7l)
- Power supply used | 5V 2A 
- SD card used | Sandisk 

#### Additional Information (if applicable)
- Was the software title installed freshly or updated/migrated? No
- Can this issue be replicated on a fresh installation of DietPi? Yes

#### Steps to reproduce

1. Fresh install DietPi and update.
2. Run command 'reboot' or 'shutdown now -r' from CLI

#### Expected behaviour

- Should restart dietpi. LED red stays on and then ethernet LED lights up.

#### Actual behaviour

- Screen goes blank. LED red stays on. Ethernet LED no activity.

#### Extra details

- Possible kernel patch required? Can help with recompilation and testing if required.

MichaIng, 2023-04-23:
> Screen goes blank.

Does this mean you have HDMI output on the Odroid C1? That would be great news, so [this](https://github.com/armbian/build/commit/c6b9135) patch than seems to have fixed it after years.

Linux 6.1 should be shipped soon (change merged already but related kernel packages not shipped yet): https://github.com/armbian/build/commit/41709dd
If you want to test it earlier:
```sh
apt install linux-{image,dtb}-edge-meson
```

As the links indicate, we do not own kernel builds for most SBCs, but use the packages from Armbian. The reboot issue is a very old one with mainline kernel on all Odroid devices, and appears depending on used block device type (eMMC vs SD card) and exact model. Some are bugged with it, others not, there have been fixes for it on some Odroid's in some circumstances, but they broke again for others with a later kernel release... never ending story.

For the Odroid C1, the additional problem is that it is a very old SBC which is not really developed anymore on Linux (mainline) end and also has no official maintainer at Armbian end + since it has a very small user space, also our (my) capacities are better focused on other SBCs, especially when its about non-trivial issues (obviously) which most likely go beyond my abilities. So if someone has a C1 and some experience with Linux device trees and want to try fixing things, that's the only way to get things forward. Check out Odroid C1 related (closed) issues here and on the Armbian forum:
- https://github.com/MichaIng/DietPi/issues?q=is%3Aissue+label%3A"Odroid+C1"+is%3Aclosed
- https://forum.armbian.com/tags/odroidc1/

juanlufont, 2024-03-13:
> Does [this](https://github.com/armbian/build/commit/c6b9135) mean you have HDMI output on the Odroid C1? That would be great news, so this patch than seems to have fixed it after years.

I just checked (fresh DietPi bookworm install) and HDMI still not working for me. After playing a bit with `/boot/boot.ini`  (disabling HDMI autodetect and testing several screen modes, from vga to 1920x1200) I get a purple screen at boot but then the monitor warns me the video input is using an unsupported timing.

MichaIng, 2024-03-19:
Hmm, there is still a patch, stating to fix this: https://github.com/armbian/build/blob/main/patch/kernel/archive/meson-6.6/odroidc1-dts-Enable-HDMI.patch

juanlufont, 2024-05-18:
I just tested DietPi 9.4 on the Odroid C1 (the Bookworm-based image), and I finally get HDMI output :smiley: 

It totally ignores the USB keyboard though (no hotplugged, connected before booting). Not sure if it may be fixed from `boot.ini`.

MichaIng, 2024-05-18:
So far so great. I'll build a new kernel package, let's see whether this fixes the USB keyboard: https://github.com/MichaIng/DietPi/actions/runs/9142379516/job/25137864315
Do other USB devices work at this port? Hotplug should of course work on all USB ports.
Also, does `lsusb` list the keyboard?`

Build will take some hours, also since GitHub runners are busy with my Amiberry builds. But when done, the packages can be found here: https://dietpi.com/downloads/binaries/testing/
To install:
```sh
cd /tmp
wget https://dietpi.com/downloads/binaries/testing/linux-{image,dtb}-current-meson.deb
dpkg -i linux-{image,dtb}-current-meson.deb
reboot
```

juanlufont, 2024-05-19:
No USB devices were detected using the default DietPi image (I tried with keyboards and also some old USB memory sticks, no USB 3.0).

Not many USB-related entries in `dmesg`:
```
root@DietPi:~# dmesg | grep -i usb
[    0.000000] Kernel command line: root=UUID=c4481023-828a-4f18-b76d-3a01a6701243 rootfstype=ext4 rootwait rw console=tty1 console=ttyAML0,115200n8 consoleblank=0 net.ifnames=0 vdaccfg=0xa000 dmfc=3 cvbsmode=576cvbs hdmimode=1080p m_bpp=32 vout=hdmi disablehpd=true hdmitx=cecf monitor_onoff=true max_freq=1536 usbhid.quirks=0x0eef:0x0005:0x0004
[    1.561850] usbcore: registered new interface driver usbfs
[    1.561916] usbcore: registered new interface driver hub
[    1.561987] usbcore: registered new device driver usb
[    2.793525] usbcore: registered new interface driver usb-storage
[    2.818544] usbcore: registered new interface driver usbhid
[    2.823818] usbhid: USB HID core driver
[    3.301396] dwc2 c90c0000.usb: supply vusb_d not found, using dummy regulator
[    3.303050] dwc2 c90c0000.usb: supply vusb_a not found, using dummy regulator
[    3.377819] dwc2 c90c0000.usb: DWC OTG Controller
[    3.377996] dwc2 c90c0000.usb: new USB bus registered, assigned bus number 1
[    3.384067] dwc2 c90c0000.usb: irq 44, io mem 0xc90c0000
[    3.393748] hub 1-0:1.0: USB hub found
```

Command `lsusb` does not return much info neither:
```
root@DietPi:~# lsusb
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

Next I am going to try with the new kernel packages you just compiled, thanks!

juanlufont, 2024-05-19:
> ```shell
> cd /tmp
> wget https://dietpi.com/downloads/binaries/testing/linux-{image,dtb}-current-meson.deb
> dpkg -i linux-{image,dtb}-current-meson.deb
> reboot
> ```

I followed this instructions, installation and reboot went well but the USB situation still the same: same `dmesg` and `lsusb` output.

MichaIng, 2024-05-19:
```console
root@DietPi:~# lsusb
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```
Hmm, it shows a single USB port only. Weird is that there was a PR done at Armbian, which states to fix the "onboard USB hub": https://github.com/armbian/build/pull/6559

Though, at, this is mostly adding U-Boot patches. Wait, I'll do a U-Boot build then: https://github.com/MichaIng/DietPi/actions/runs/9147643884

juanlufont, 2024-05-19:
Same result after installing the updated uboot:

```
root@DietPi:~# lsusb
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

root@DietPi:~# dmesg | grep -i usb
[    0.000000] Kernel command line: root=UUID=c4481023-828a-4f18-b76d-3a01a6701243 rootfstype=ext4 rootwait rw console=tty1 console=ttyAML0,115200n8 consoleblank=0 net.ifnames=0 vdaccfg=0xa000 dmfc=3 cvbsmode=576cvbs hdmimode=1080p m_bpp=32 vout=hdmi hdmitx=cecf monitor_onoff=true max_freq=1536 usbhid.quirks=0x0eef:0x0005:0x0004
[    1.561848] usbcore: registered new interface driver usbfs
[    1.562036] usbcore: registered new interface driver hub
[    1.562124] usbcore: registered new device driver usb
[    2.805266] usbcore: registered new interface driver usb-storage
[    2.830144] usbcore: registered new interface driver usbhid
[    2.835526] usbhid: USB HID core driver
[    3.325753] dwc2 c90c0000.usb: supply vusb_d not found, using dummy regulator
[    3.327778] dwc2 c90c0000.usb: supply vusb_a not found, using dummy regulator
[    3.408566] dwc2 c90c0000.usb: DWC OTG Controller
[    3.408785] dwc2 c90c0000.usb: new USB bus registered, assigned bus number 1
[    3.414855] dwc2 c90c0000.usb: irq 44, io mem 0xc90c0000
[    3.423124] hub 1-0:1.0: USB hub found
```

MichaIng, 2024-05-19:
Btw, were USB ports working with the previous Linux 5.15.93?

And since one port shows up in `lsusb`, could you test all 4 USB ports and see whether one of them works?

juanlufont, 2024-05-21:
I have just tested the 4 USB ports, nothing is detected (`lsusb` always return the same output, no new entries in `dmesg`)

About the HDMI not working, I only tested it back in March after finding this ticket, before that I long gave up on HDMI output in the Odroid-C1(+)

MichaIng, 2024-05-22:
> I am not sure, but I think the only USB hub that is listed corresponds to the micro USB OTG. It seems the hub with 4 ports is not detected at all.

Makes sense.

juanlufont, 2024-05-22:
I have been testing some stuff out of DietPi, with some results. I have generated a couple of Armbian images for Odroid-C1:
- current : same problem, it has HDMI output but no USB Hub detected
- edge: HDMI output AND USB Hub detected. The USB works a bit spotty (it is picky for some keyboards, sometimes the keys get "stuck", but it mostly works

With the Armbian "edge" image, I see the following:
```
root@odroidc1:~# lsusb
Bus 001 Device 003: ID 0461:4002 Primax Electronics, Ltd Homoo Gaming Keyboard
Bus 001 Device 002: ID 05e3:0610 Genesys Logic, Inc. Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

and 

```
root@odroidc1:~# uname -a
Linux odroidc1 6.8.10-edge-meson #1 SMP Fri May 17 10:15:12 UTC 2024 armv7l GNU/Linux
```

MichaIng, 2024-05-22:
Weird: The device tree and all patches exactly the same. No idea how it can work on edge but not on current. Let me do an edge build as well, to test on DietPi. The ones from the Armbian APT repo are still Linux 6.7.

juanlufont, 2024-05-22:
To make sure I was not mixing up things, today I tested all images again, including the ones you generated some days ago. I can confirm that only Armbian `edge` works.

MichaIng, 2024-05-22:
Please test this on DietPi:
```sh
cd /tmp
wget https://dietpi.com/downloads/binaries/testing/linux-{image,dtb}-edge-meson.deb
dpkg -i linux-{image,dtb}-edge-meson.deb
reboot
```

juanlufont, 2024-05-22:
I started with a default DietPi installation (no USB Hub other than the OGT)
```
root@DietPi:~# lsusb
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

I installed the new kernel image you generated. The Odroid C1 struggle a bit and after a slow boot, the USB Hub was detected:
```
root@DietPi:~# lsusb
Bus 001 Device 002: ID 05e3:0610 Genesys Logic, Inc. Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

This is the MOTD:
```
 ─────────────────────────────────────────────────────
 DietPi v9.4.2 : 00:44 - Thu 05/23/24
 ─────────────────────────────────────────────────────
 - Device model : Odroid C1 (armv7l)
 - CPU temp : 34 °C / 93 °F : Cool runnings
 - LAN IP : 192.168.8.237 (eth0)
 - MOTD : DietPi v9.4 has been released. Check out all changes:
          https://dietpi.com/docs/releases/v9_4/
 ─────────────────────────────────────────────────────
```

And the info of the _edge_ kernel running:

```
root@DietPi:~# uname -a
Linux DietPi 6.8.10-edge-meson #1 SMP Fri May 17 10:15:12 UTC 2024 armv7l GNU/Linux
```

The situation after installing the _edge_ kernel:
- slow boot, with some snd module errors
- USB hot-plug does not work
- connecting an USB keyboard to the board before booting it up works, but the performance is so so (missed keystrokes, other times a key get stuck)
- HDMI works (but not hot-plug neither)

MichaIng, 2024-05-23:
Okay, thanks for testing. So then there seems to be a difference in the underlying driver or maybe timing at boot between Linux 6.6 and 6.8, which has an effect, despite device tree being the same.

But no hot-plug and bad scattered input does not sound great either.

... wait a second, the PR I mentioned enabled `CONFIG_USB_ONBOARD_HUB=y` for edge kernel only, while it is still unset in current. Let me try to enable this for current as well.

juanlufont, 2024-05-23:
USB hub is not detected :-)

```
 ─────────────────────────────────────────────────────
 DietPi v9.4.2 : 19:10 - Thu 05/23/24
 ─────────────────────────────────────────────────────
 - Device model : Odroid C1 (armv7l)
 - CPU temp : 31 °C / 87 °F : Cool runnings
 - LAN IP : 192.168.8.237 (eth0)
 - MOTD : DietPi v9.4 has been released. Check out all changes:
          https://dietpi.com/docs/releases/v9_4/
 ─────────────────────────────────────────────────────

root@DietPi:~# lsusb
Bus 001 Device 002: ID 05e3:0610 Genesys Logic, Inc. Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

root@DietPi:~# uname -a
Linux DietPi 6.6.31-current-meson #1 SMP Fri May 17 10:02:40 UTC 2024 armv7l GNU/Linux
root@DietPi:~# 
```

juanlufont, 2024-05-23:
The USB support is really spotty in the _current_ branch:
- it takes a lot of time to boot up
- it is very picky with the USB keyboards I connect
- boot process gets stuck at HW / USB device detection, unstuck when I disconnected the  keyboard
- I get some random errors from kernel _irq_threads_
- once I manage to log in, the system feels quite laggy

The Armbian _edge_ image (kernel 6.8), in comparison, booted up without so much trouble and it felt more responsive.

MichaIng, 2024-05-23:
Okay, so then there seem to be other differences in newer Linux which have an effect on this. The PR I linked did nothing else than setting this kernel config flag for edge kernel. All other changes in that PR were patches for legacy kernel. The question now is whether it is better to have such spotty USB support, than having none at all? I am not a great fan to use the edge kernel, until it reached a new LTS version, since it means regular major changes, including chances for breakage features.

Could you check `htop` whether you see a `udev-worker` or similar process? Probably udev device detection is in a loop, causing the laggy behaviour and possibly additional constant CPU load etc.

juanlufont, 2024-05-23:
I lost hope for decent Odroid C1 support long ago, these recent developments are a bit of a miracle to me :-D It makes sense to not adopt the _edge_ kernel to just gain buggy USB support.

About `htop`, when I connect an USB device several `udev-worker` processes show up. The load of the system also goes up despite there is little actual CPU usage (it goes up to 6, which sounds quite bad for an idle system).

gociii, 2024-05-24:
> ```
> root@DietPi:~# lsusb
> Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
> ```
> 
> Hmm, it shows a single USB port only. Weird is that there was a PR done at Armbian, which states to fix the "onboard USB hub": [armbian/build#6559](https://github.com/armbian/build/pull/6559)
> 
> Though, at, this is mostly adding U-Boot patches. Wait, I'll do a U-Boot build then: https://github.com/MichaIng/DietPi/actions/runs/9147643884
> 
> EDIT: Done:
> 
> ```shell
> cd /tmp
> wget https://dietpi.com/downloads/binaries/testing/linux-u-boot-odroidc1-current.deb
> dpkg -i linux-u-boot-odroidc1-current.deb
> /boot/dietpi/func/dietpi-set_hardware flash-u-boot-mmc
> reboot
> ```

Thanks @MichaIng  - the reboot / shutdown issue is resolved using the meson kernel. Haven't tested USB ports as I'm not using any on this C1 device.

MichaIng, 2024-05-26:
Okay, I'll upload a fresh `current` kernel build into our APT repo, to fix the reboot issue for everyone. Since this is what this issue was about, I'll close it. I opened a new issue to track the missing/troublesome USB support: #7089

closed 2024-05-26 by MichaIng

MichaIng, 2024-05-26:
Again unrelated, but the PR at Armbian fixed U-Boot builds as well, and indeed it compiled fine: https://github.com/MichaIng/DietPi/actions/runs/9243339257/job/25427343895

Not sure whether anything relevant changed, the U-Boot version itself remained the same old Hardkernel one, but I feel better to provide a fresh builds, rather than an old one pulled from Armbian's repo compiled in 2021. So if someone could test it, just to be sure, that would be great:
```sh
cd /tmp
wget https://dietpi.com/downloads/binaries/testing/linux-u-boot-odroidc1-current.deb
dpkg -i linux-u-boot-odroidc1-current.deb
/boot/dietpi/func/dietpi-set_hardware flash-u-boot-mmc
```
In the unexpected incidence that the device does not boot up again, I can provide steps how to flash the old U-Boot image from another Linux system.

gociii, 2024-05-27:
Ran the commands in bold.

*dietpi-update cd /tmp wget 
https://dietpi.com/downloads/binaries/testing/linux-u-boot-odroidc1-current.deb 
dpkg -i linux-u-boot-odroidc1-current.deb 
/boot/dietpi/func/dietpi-set_hardware flash-u-boot-mmc reboot shutdown 
now -r *

(Tested both with reboot & shutdown now -r)

*uname -a*
Linux ratchet 6.6.31-current-meson #1 SMP Fri May 17 10:02:40 UTC 2024 armv7l GNU/Linux

*apt update; apt full-upgrade -y; apt autoremove -y*

  "...upgraded:
   linux-dtb-current-meson linux-image-current-meson"

*reboot*

*uname -a *Linux ratchet 6.6.32-current-meson #1 SMP Sat May 25 14:22:56 UTC 2024 armv7l GNU/Linux**

Looks all good to me.. Thanks once again @MichaIng :)

**

th2j, 2024-06-12:
Hello,

i updated to dietpi 9.5.1 on Odroid C1+
but reboot doesn't work

root@DietPi:~# uname -a
Linux DietPi 6.1.11-meson #23.02.2 SMP Fri Feb 17 22:35:23 UTC 2023 armv7l GNU/Linux

Thanks
Thomas

th2j, 2024-06-12:
> Since you are still using the old kernel, I guess you did not reboot
> (power cycle) once yet, after the update? Please do so, afterwards, it
> should work.

yes i have restarted once, but the kernel have not been updated
it is still 6.1.11

th2j, 2024-06-12:
> Can you check this:
> dpkg -l | grep linux-
> ls -l /boot

***@***.***:~# uname -a
Linux DietPi 6.1.11-meson #23.02.2 SMP Fri Feb 17 22:35:23 UTC 2023 armv7l
GNU/Linux

***@***.***:~# dpkg -l | grep linux-
ii  linux-base                    4.9                            all
   Linux image base package
ii  linux-dtb-current-meson       23.02.2                        armhf
   Armbian Linux DTB, version 5.15.93-meson current
ii  linux-dtb-edge-meson          23.02.2                        armhf
   Armbian Linux DTB, version 6.1.11-meson edge
ii  linux-image-current-meson     23.02.2                        armhf
   Linux kernel, armbian version 5.15.93-meson current
ii  linux-image-edge-meson        23.02.2                        armhf
   Linux kernel, armbian version 6.1.11-meson edge
ii  linux-u-boot-odroidc1-current 24.8.0-trunk-dietpi1           armhf
   Das U-Boot for odroidc1
ii  util-linux-extra              2.38.1-5+deb12u1               armhf
   interactive login tools

***@***.***:~# ls -l /boot
total 42876
-rwxr-xr-x 1 root root     3274 22 févr. 21:39 boot.ini
-rwxr-xr-x 1 root root   195501 17 févr.  2023 config-6.1.11-meson
drwxr-xr-x 4 root root     4096  9 juin  19:54 dietpi
-rwxr-xr-x 1 root root    18092 23 sept.  2023 dietpi-LICENSE.txt
-rwxr-xr-x 1 root root    16059 23 sept.  2023 dietpi-README.md
-rwxr-xr-x 1 root root    17473 12 juin  13:54 dietpi.txt
-rwxr-xr-x 1 root root     3950 25 sept.  2023 dietpi-wifi.txt
drwxr-xr-x 2 root root     2048 10 janv. 20:17 dtb
-rwxr-xr-x 1 root root  6304133 31 déc.  11:26 initrd.img-5.15.93-meson
-rwxr-xr-x 1 root root  6373587 19 févr. 10:49 initrd.img-6.1.11-meson
-rwxr-xr-x 1 root root  4221311 17 févr.  2023 System.map-6.1.11-meson
-rwxr-xr-x 1 root root 10179136 10 janv. 20:18 uImage
-rwxr-xr-x 1 root root  6373651 19 févr. 10:49 uInitrd
-rwxr-xr-x 1 root root 10179072 17 févr.  2023 vmlinuz-6.1.11-meson

MichaIng, 2024-06-12:
Oh, I see, I forgot about an old problem with an earlier package upgrade shipped by Armbian, which rendered the C1 unbootable: https://dietpi.com/forum/t/solved-odroid-c1-not-booting-after-kernel-upgrade/17818

However, this has been solved at Armbian, and we ship our own packages now, hence this is not needed anymore. Please do this:
```sh
apt-mark unhold linux-{image,dtb,headers}-{current,edge}-meson
G_AGP linux-{image,dtb,headers}-edge-meson
apt upgrade
```
This should upgrade your kernel to Linux 6.6.

th2j, 2024-06-12:
ok reboot works now
but i have no usb

root@DietPi:~# lsusb
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

MichaIng, 2024-06-12:
Did USB work before?

It has been reported already: #7089

Uff, this old SBC is a constant source of issues 🙈.

th2j, 2024-06-12:
yes all worked fine
i use the odroid connected to a dac with usb

th2j, 2024-06-12:
is there a fix ?

if not, how can i go back to old kernel please ?
i do not not need reboot working but i need usb

Thanks 
Thomas