MichaIng/DietPi #6335: default /etc/sudoers doesn't allow use of dietpi-* tools

thinkpanzer, 2023-04-23:
### Creating a bug report/issue

#### Required Information
- DietPi version
G_DIETPI_VERSION_CORE=8
G_DIETPI_VERSION_SUB=16
G_DIETPI_VERSION_RC=2
G_GITBRANCH='beta'
G_GITOWNER='MichaIng'

- Distro version

returns null

- Kernel version

Linux 5.10.110 #1 SMP Tue Mar 14 21:59:07 CST 2023 aarch64 GNU/Linux

- SBC model

NanoPi R6S

#### Additional Information (if applicable)
- Can this issue be replicated on a fresh installation of DietPi?

Yes

#### Steps to reproduce
- Trying to run `dietpi-update` as an unprivileged user returns `[FAILED] DietPi-Update | Root privileges required. Please run the command with "sudo" or "G_SUDO".`
- By default, `/etc/sudoers` does not include `/boot/dietpi` in `secure_path`
- When trying to run `sudo dietpi-updater` manually, sudo returns `sudo: dietpi-updater: command not found`

#### Expected behaviour
- Error messages and default DietPi behavior should be consistent

#### Extra details
- I can imagine legitimate reasons for not including `/boot/dietpi` as a default part of `secure_path` in `/etc/sudoers` so I'm not going to propose simply adding it, mainly because I don't know if the DietPi developers consider `/boot/dietpi` to be a secure and trustable path. If not, then adding it to `secure_path` is probably not a good idea. If this is the case, the error messages should be updated.

Thanks!

MichaIng, 2023-04-23:
> When trying to run `sudo dietpi-updater` manually, sudo returns `sudo: dietpi-updater: command not found`

The following should work:
```sh
sudo dietpi-update
```

thinkpanzer, 2023-04-23:
Sorry, the original string was a transcription error on my part.

`sudo dietpi-update` generates the error, and the root cause/fix was because `/boot/dietpi` was not part of `secure_path` in `/etc/sudoers`. I **have** changed my default login shell to zsh and made a few other customizations, so I'll retry on a newly installed image to see if I can understand why it apparently works on a fresh system. Thanks.

MichaIng, 2023-04-23:
Ah, this is because zsh does not load the `dietpi-*` aliases from `/etc/bashrc.d/dietpi.bash`. The scripts are not added to `PATH` but added as alias. Furthermore, `sudo` itself cannot make use of aliases via `sudo <alias>`, but only if it is made an alias of itself.

If I remember right, it generally works wo `source` this script in zsh, but some of the `G_*` functions won't work correctly, e.g. since zsh counts array indices from 1 up instead of 0, right? But the aliases, login banner and autostart options should work that way.

MichaIng, 2025-05-31:
`fish` support has been implemented. But it does not allow to run aliases with sudo trivially either 😅. However, that is another issue I'll try to solve for upcoming release.

closed 2025-05-31 by MichaIng