MichaIng/DietPi #6345: hostapd cant start for hotspot

danilabi, 2023-04-27:
#### Details:
- Date           | Thu Apr 27 21:10:13 BST 2023
- Bug report     | df1c14cf-d37f-4a76-ad9f-679658c0c131
- DietPi version | v8.16.2 (MichaIng/master)
- Image creator  |
- Pre-image      |
- Hardware       | ROCK64 (aarch64) (ID=43)
- Kernel version | `Linux DietPi 5.15.93-rockchip64 #23.02.2 SMP PREEMPT Fri Feb 17 23:48:36 UTC 2023 aarch64 GNU/Linux`
- Distro         | bullseye (ID=6)
- Command        | `systemctl start hostapd`
- Exit code      | 1
- Software title | DietPi-Config
#### Steps to reproduce:

1. ...
2. ...
#### Expected behaviour:

- ...
#### Actual behaviour:

- ...
#### Extra details:

- ...
#### Additional logs:
```
Job for hostapd.service failed because the control process exited with error code.
See "systemctl status hostapd.service" and "journalctl -xe" for details.
```

Joulinar, 2023-04-27:
what is the output of 

```
journalctl -u hostapd.service
```

danilabi, 2023-04-28:
Apr 28 11:13:03 DietPi systemd[1]: hostapd.service: Scheduled restart job, restart counter is at 80.
Apr 28 11:13:03 DietPi systemd[1]: Stopped Access point and authentication server for Wi-Fi and Ethernet.
Apr 28 11:13:03 DietPi systemd[1]: Starting Access point and authentication server for Wi-Fi and Ethernet...
Apr 28 11:13:03 DietPi hostapd[1804]: Configuration file: /etc/hostapd/hostapd.conf
Apr 28 11:13:03 DietPi hostapd[1804]: nl80211: Could not configure driver mode
Apr 28 11:13:03 DietPi hostapd[1804]: nl80211: deinit ifname=wlan0 disabled_11b_rates=0
Apr 28 11:13:03 DietPi hostapd[1804]: nl80211 driver initialization failed.
Apr 28 11:13:03 DietPi hostapd[1804]: wlan0: interface state UNINITIALIZED->DISABLED
Apr 28 11:13:03 DietPi hostapd[1804]: wlan0: AP-DISABLED
Apr 28 11:13:03 DietPi hostapd[1804]: wlan0: CTRL-EVENT-TERMINATING
Apr 28 11:13:03 DietPi hostapd[1804]: hostapd_free_hapd_data: Interface wlan0 wasn't started
Apr 28 11:13:03 DietPi systemd[1]: hostapd.service: Control process exited, code=exited, status=1/FAILURE
Apr 28 11:13:03 DietPi systemd[1]: hostapd.service: Failed with result 'exit-code'.
Apr 28 11:13:03 DietPi systemd[1]: Failed to start Access point and authentication server for Wi-Fi and Ethernet.

danilabi, 2023-04-28:
already uninstalled and installed again - doesn't work

Joulinar, 2023-04-28:
your Wifi Adapter doesn't seems to be working / available.

can you show:

```
ip a
```

Joulinar, 2023-04-28:
interface seems to be down. Can you try to bring it up

```
ifup wlan0
```

danilabi, 2023-04-28:
ifup: interface wlan0 already configured

MichaIng, 2023-04-28:
Which WiFi chip is it?
```sh
lsmod
```

danilabi, 2023-04-28:
Module                  Size  Used by
binfmt_misc            24576  1
nft_chain_nat          16384  8
xt_MASQUERADE          20480  2
nf_nat                 49152  2 nft_chain_nat,xt_MASQUERADE
nft_counter            16384  6
xt_state               16384  0
xt_conntrack           16384  2
nf_conntrack          163840  4 xt_conntrack,nf_nat,xt_state,xt_MASQUERADE
nf_defrag_ipv6         24576  1 nf_conntrack
nf_defrag_ipv4         16384  1 nf_conntrack
nft_compat             20480  4
nf_tables             212992  29 nft_compat,nft_counter,nft_chain_nat
nfnetlink              20480  2 nft_compat,nf_tables
8192eu               1396736  0
rtl8xxxu              122880  0
mac80211              913408  1 rtl8xxxu
snd_soc_hdmi_codec     24576  1
snd_soc_spdif_tx       16384  0
libarc4                16384  1 mac80211
hantro_vpu            118784  0
rockchip_vdec          32768  0
gpio_ir_recv           16384  0
snd_soc_rockchip_i2s    24576  4
snd_soc_rk3328         16384  1
rc_core                57344  2 gpio_ir_recv
snd_soc_simple_card    24576  0
rockchip_iep           20480  0
snd_soc_rockchip_spdif    16384  0
snd_soc_simple_card_utils    28672  1 snd_soc_simple_card
rockchip_rga           28672  0
snd_soc_rockchip_pcm    16384  1 snd_soc_rockchip_i2s
v4l2_h264              16384  2 rockchip_vdec,hantro_vpu
videobuf2_dma_contig    24576  3 rockchip_vdec,hantro_vpu,rockchip_iep
videobuf2_dma_sg       24576  1 rockchip_rga
snd_soc_core          262144  8 snd_soc_spdif_tx,snd_soc_rockchip_pcm,snd_soc_hdmi_codec,snd_soc_rockchip_spdif,snd_soc_rk3328,snd_soc_simple_card_utils,snd_soc_rockchip_i2s,snd_soc_simple_card
dw_hdmi_cec            16384  0
v4l2_mem2mem           40960  4 rockchip_vdec,hantro_vpu,rockchip_rga,rockchip_iep
cfg80211              876544  3 8192eu,mac80211,rtl8xxxu
snd_pcm_dmaengine      20480  1 snd_soc_core
videobuf2_vmalloc      20480  1 hantro_vpu
videobuf2_memops       20480  3 videobuf2_vmalloc,videobuf2_dma_contig,videobuf2_dma_sg
dw_hdmi_i2s_audio      16384  0
videobuf2_v4l2         32768  5 rockchip_vdec,hantro_vpu,rockchip_rga,rockchip_iep,v4l2_mem2mem
snd_pcm               131072  3 snd_soc_hdmi_codec,snd_soc_core,snd_pcm_dmaengine
videobuf2_common       65536  10 rockchip_vdec,videobuf2_vmalloc,videobuf2_dma_contig,videobuf2_v4l2,hantro_vpu,rockchip_rga,videobuf2_dma_sg,rockchip_iep,v4l2_mem2mem,videobuf2_memops
lima                   61440  0
gpu_sched              40960  1 lima
rfkill                 36864  2 cfg80211
videodev              262144  7 rockchip_vdec,videobuf2_v4l2,hantro_vpu,rockchip_rga,videobuf2_common,rockchip_iep,v4l2_mem2mem
snd_timer              40960  1 snd_pcm
mc                     61440  6 rockchip_vdec,videodev,videobuf2_v4l2,hantro_vpu,videobuf2_common,v4l2_mem2mem
snd                    94208  4 snd_soc_hdmi_codec,snd_timer,snd_soc_core,snd_pcm
soundcore              16384  1 snd
cpufreq_dt             20480  0
ip_tables              32768  0
x_tables               49152  5 xt_conntrack,nft_compat,xt_state,ip_tables,xt_MASQUERADE
autofs4                49152  2
realtek                32768  1
dwmac_rk               28672  0
stmmac_platform        24576  1 dwmac_rk
stmmac                237568  2 stmmac_platform,dwmac_rk
pcs_xpcs               24576  1 stmmac
gpio_syscon            20480  1

MichaIng, 2023-04-28:
Ah sorry, wrong command, this one I meant:
```sh
lsusb
```
I see `8192eu` and `rtl8xxxu` modules, but would be still good to know which chip exactly it is. Because the RTL8188CUS simply does not support hostapd, respectively the default driver used by it, and all existing forks to implement an alternative driver are outdated in the meantime: https://github.com/pritambaral/hostapd-rtl871xdrv

danilabi, 2023-04-28:
Bus 005 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 004 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 003 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 001 Device 002: ID 0bda:818b Realtek Semiconductor Corp. RTL8192EU 802.11b/g/n WLAN Adapter
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

danilabi, 2023-04-28:
If my wifi module doesn't support hostapd could you recommend me another one i could use?

MichaIng, 2023-04-29:
I don't know any other access point software for Linux, but that doesn't mean they do not exist.

Hmm, it does not match 100%, which WiFi adapter model is it exactly?

There are also other possible reasons why it fails: https://askubuntu.com/questions/472794/hostapd-error-nl80211-could-not-configure-driver-mode
```sh
apt install rfkill
ifdown wlan0
rfkill unblock wlan
ifup wlan0
systemctl restart isc-dhcp-server
systemctl restart hostapd
```
Also, are there probably other networking services running?
```sh
systemctl status systemd-networkd NetworkManager wpa_supplicant dhcpcd
```

danilabi, 2023-04-29:
root@DietPi:~# apt install rfkill
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
The following NEW packages will be installed:
  rfkill
0 upgraded, 1 newly installed, 0 to remove and 0 not upgraded.
Need to get 87.7 kB of archives.
After this operation, 146 kB of additional disk space will be used.
Get:1 https://deb.debian.org/debian bullseye/main arm64 rfkill arm64 2.36.1-8+deb11u1 [87.7 kB]
Fetched 87.7 kB in 0s (308 kB/s)
debconf: delaying package configuration, since apt-utils is not installed
Selecting previously unselected package rfkill.
(Reading database ... 19432 files and directories currently installed.)
Preparing to unpack .../rfkill_2.36.1-8+deb11u1_arm64.deb ...
Unpacking rfkill (2.36.1-8+deb11u1) ...
Setting up rfkill (2.36.1-8+deb11u1) ...
root@DietPi:~# ifdown wlan0
ifdown: interface wlan0 not configured
root@DietPi:~# rfkill unblock wlan
root@DietPi:~# ifup wlan0
Error for wireless request "Set Power Management" (8B2C) :
    SET failed on device wlan0 ; Operation not supported.
root@DietPi:~# systemctl restart isc-dhcp-server
root@DietPi:~# systemctl restart hostapd
Job for hostapd.service failed because the control process exited with error code.
See "systemctl status hostapd.service" and "journalctl -xe" for details.
root@DietPi:~# systemctl status systemd-networkd NetworkManager wpa_supplicant dhcpcd
Unit NetworkManager.service could not be found.
Unit dhcpcd.service could not be found.
● systemd-networkd.service - Network Service
     Loaded: loaded (/lib/systemd/system/systemd-networkd.service; disabled; vendor preset: enabled)
     Active: inactive (dead)
TriggeredBy: ● systemd-networkd.socket
       Docs: man:systemd-networkd.service(8)

● wpa_supplicant.service - WPA supplicant
     Loaded: loaded (/lib/systemd/system/wpa_supplicant.service; disabled; vendor preset: enabled)
     Active: inactive (dead)

MichaIng, 2023-04-29:
Okay that looks all good. If you know the exact WiFi adapter model, that would help to assure it is really the chip being unsupported. In this case there are some very old but existing `hostapd-realtek` builds from Armbian, which use mentioned alternative driver: https://mirrors.xtom.de/armbian/pool/bullseye-utils/w/wpa/

danilabi, 2023-04-29:
RTL8192EU is the chipset

MichaIng, 2023-04-29:
If you take this from `lsusb` output, this may not be accurate, especially since some of these Realtek chips name themselves differently, have aliases etc. The adapter model may give clarification.

danilabi, 2023-04-29:
root@DietPi:~# lsusb
Bus 005 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 004 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 003 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 001 Device 003: ID 0bda:818b Realtek Semiconductor Corp. RTL8192EU 802.11b/g/n WLAN Adapter
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

MichaIng, 2023-04-29:
You posted it above already. As said, this information may be not accurate/wrong. However, if you do not know the name of the WiFi adapter (the whole brand + product, not the chip only), then we won't find out.

Try this:
```sh
cd /tmp
curl -Lo package.deb 'https://apt.armbian.com/pool/bullseye-utils/w/wpa/hostapd-realtek_2.5-4~armbian22.02.3+1_arm64.deb'
dpkg -i package.deb
G_CONFIG_INJECT 'driver=' 'driver=rtl871xdrv' /etc/hostapd/hostapd.conf
systemctl restart hostapd
```

danilabi, 2023-04-29:
now hostapd is working but i still cant find the hotspot

danilabi, 2023-04-29:
there is a problem with the dhcp server: 
root@DietPi:~# sudo systemctl restart isc-dhcp-server.service
Job for isc-dhcp-server.service failed because the control process exited with error code.
See "systemctl status isc-dhcp-server.service" and "journalctl -xe" for details.
root@DietPi:~# sudo systemctl status isc-dhcp-server.service
● isc-dhcp-server.service - LSB: DHCP server
     Loaded: loaded (/etc/init.d/isc-dhcp-server; generated)
     Active: failed (Result: exit-code) since Sat 2023-04-29 23:23:04 BST; 4s ago
       Docs: man:systemd-sysv-generator(8)
    Process: 14417 ExecStart=/etc/init.d/isc-dhcp-server start (code=exited, status=1/FAILURE)
        CPU: 130ms

Apr 29 23:23:02 DietPi dhcpd[14432]: before submitting a bug.  These pages explain the proper
Apr 29 23:23:02 DietPi dhcpd[14432]: process and the information we find helpful for debugging.
Apr 29 23:23:02 DietPi dhcpd[14432]:
Apr 29 23:23:02 DietPi dhcpd[14432]: exiting.
Apr 29 23:23:04 DietPi isc-dhcp-server[14417]: Starting ISC DHCPv4 server: dhcpdcheck syslog for diagnostics. ...
Apr 29 23:23:04 DietPi isc-dhcp-server[14437]:  failed!
Apr 29 23:23:04 DietPi isc-dhcp-server[14438]:  failed!
Apr 29 23:23:04 DietPi systemd[1]: isc-dhcp-server.service: Control process exited, code=exited, status=1/FAILURE
Apr 29 23:23:04 DietPi systemd[1]: isc-dhcp-server.service: Failed with result 'exit-code'.
Apr 29 23:23:04 DietPi systemd[1]: Failed to start LSB: DHCP server.

Joulinar, 2023-04-30:
can you restart your system and check following afterwards

```
journalctl -u isc-dhcp-server.service
```

danilabi, 2023-04-30:
root@DietPi:~# journalctl -u isc-dhcp-server.service
-- Journal begins at Sun 2023-04-30 11:51:20 BST, ends at Sun 2023-04-30 11:51:38 BST. --
Apr 30 11:51:32 DietPi systemd[1]: Starting LSB: DHCP server...
Apr 30 11:51:32 DietPi isc-dhcp-server[645]: Launching IPv4 server only.
Apr 30 11:51:32 DietPi dhcpd[654]: Internet Systems Consortium DHCP Server 4.4.1
Apr 30 11:51:32 DietPi dhcpd[654]: Copyright 2004-2018 Internet Systems Consortium.
Apr 30 11:51:32 DietPi dhcpd[654]: All rights reserved.
Apr 30 11:51:32 DietPi dhcpd[654]: For info, please visit https://www.isc.org/software/dhcp/
Apr 30 11:51:32 DietPi dhcpd[660]: Internet Systems Consortium DHCP Server 4.4.1
Apr 30 11:51:32 DietPi dhcpd[660]: Copyright 2004-2018 Internet Systems Consortium.
Apr 30 11:51:32 DietPi dhcpd[660]: All rights reserved.
Apr 30 11:51:32 DietPi dhcpd[660]: For info, please visit https://www.isc.org/software/dhcp/
Apr 30 11:51:32 DietPi dhcpd[660]: Wrote 0 deleted host decls to leases file.
Apr 30 11:51:32 DietPi dhcpd[660]: Wrote 0 new dynamic host decls to leases file.
Apr 30 11:51:32 DietPi dhcpd[660]: Wrote 0 leases to leases file.
Apr 30 11:51:33 DietPi dhcpd[660]: Can't bind to dhcp address: Address already in use
Apr 30 11:51:33 DietPi dhcpd[660]: Please make sure there is no other dhcp server
Apr 30 11:51:33 DietPi dhcpd[660]: running and that there's no entry for dhcp or
Apr 30 11:51:33 DietPi dhcpd[660]: bootp in /etc/inetd.conf.   Also make sure you
Apr 30 11:51:33 DietPi dhcpd[660]: are not running HP JetAdmin software, which
Apr 30 11:51:33 DietPi dhcpd[660]: includes a bootp server.
Apr 30 11:51:33 DietPi dhcpd[660]:
Apr 30 11:51:33 DietPi dhcpd[660]: If you think you have received this message due to a bug rather
Apr 30 11:51:33 DietPi dhcpd[660]: than a configuration issue please read the section on submitting
Apr 30 11:51:33 DietPi dhcpd[660]: bugs on either our web page at www.isc.org or in the README file
Apr 30 11:51:33 DietPi dhcpd[660]: before submitting a bug.  These pages explain the proper
Apr 30 11:51:33 DietPi dhcpd[660]: process and the information we find helpful for debugging.
Apr 30 11:51:33 DietPi dhcpd[660]:
Apr 30 11:51:33 DietPi dhcpd[660]: exiting.
Apr 30 11:51:35 DietPi isc-dhcp-server[645]: Starting ISC DHCPv4 server: dhcpdcheck syslog for diagnostics. ...
Apr 30 11:51:35 DietPi isc-dhcp-server[672]:  failed!
Apr 30 11:51:35 DietPi isc-dhcp-server[673]:  failed!
Apr 30 11:51:35 DietPi systemd[1]: isc-dhcp-server.service: Control process exited, code=exited, status=1/FAILURE
Apr 30 11:51:35 DietPi systemd[1]: isc-dhcp-server.service: Failed with result 'exit-code'.
Apr 30 11:51:35 DietPi systemd[1]: Failed to start LSB: DHCP server.

Joulinar, 2023-04-30:
Looks like DHCP server port is alreda in use by another application. Can you check listen ports.

```
ss -tulpn
```

danilabi, 2023-04-30:
root@DietPi:~# ss -tulpn
Netid       State        Recv-Q       Send-Q             Local Address:Port               Peer Address:Port       Process
udp         UNCONN       0            0                        0.0.0.0:53                      0.0.0.0:*           users:(("dnsmasq",pid=631,fd=6))
udp         UNCONN       0            0                        0.0.0.0:67                      0.0.0.0:*           users:(("dnsmasq",pid=631,fd=4))
udp         UNCONN       0            0                        0.0.0.0:68                      0.0.0.0:*           users:(("dhclient",pid=474,fd=9))
udp         UNCONN       0            0                           [::]:53                         [::]:*           users:(("dnsmasq",pid=631,fd=8))
tcp         LISTEN       0            32                       0.0.0.0:53                      0.0.0.0:*           users:(("dnsmasq",pid=631,fd=7))
tcp         LISTEN       0            1000                     0.0.0.0:22                      0.0.0.0:*           users:(("dropbear",pid=419,fd=4))
tcp         LISTEN       0            511                      0.0.0.0:1880                    0.0.0.0:*           users:(("node-red",pid=617,fd=18))
tcp         LISTEN       0            32                          [::]:53                         [::]:*           users:(("dnsmasq",pid=631,fd=9))
tcp         LISTEN       0            1000                        [::]:22                         [::]:*           users:(("dropbear",pid=419,fd=5))

Joulinar, 2023-04-30:
There is `dnsmasq` running as DHCP server. Any purpose for this?

danilabi, 2023-04-30:
no

Joulinar, 2023-04-30:
It is running as DNS server as well? Are you aware on this functionality? If not, you could purge the `dnsmasq` package .

danilabi, 2023-04-30:
i'll purge it

danilabi, 2023-04-30:
ok dhcp server is now working

danilabi, 2023-04-30:
now it says no subnet decleration for wlan0

MichaIng, 2023-04-30:
Did you install the WiFi hospot via `dietpi-software`? `/etc/dhcp/dhcpd.conf` should look like this, containing the subnet declaration:
```
ddns-update-style none;
default-lease-time 600;
max-lease-time 7200;
authoritative;
log-facility local7;

subnet 192.168.42.0 netmask 255.255.255.0 {
        range 192.168.42.10 192.168.42.50;
        option broadcast-address 192.168.42.255;
        option routers 192.168.42.1;
        option domain-name "local";
        option domain-name-servers 9.9.9.9, 149.112.112.112;
```

danilabi, 2023-04-30:
did it over the dietpi-software and the file is the same but it still says that there is no subnet decleration

Joulinar, 2023-04-30:
Can you share where you see the error message. Best would be the whole log

danilabi, 2023-04-30:
Apr 30 16:29:30 DietPi systemd[1]: Starting LSB: DHCP server...
Apr 30 16:29:30 DietPi isc-dhcp-server[1559]: Launching IPv4 server only.
Apr 30 16:29:30 DietPi dhcpd[1570]: Internet Systems Consortium DHCP Server 4.4.1
Apr 30 16:29:30 DietPi dhcpd[1570]: Copyright 2004-2018 Internet Systems Consortium.
Apr 30 16:29:30 DietPi dhcpd[1570]: All rights reserved.
Apr 30 16:29:30 DietPi dhcpd[1570]: For info, please visit https://www.isc.org/software/dhcp/
Apr 30 16:29:30 DietPi dhcpd[1577]: Internet Systems Consortium DHCP Server 4.4.1
Apr 30 16:29:30 DietPi dhcpd[1577]: Copyright 2004-2018 Internet Systems Consortium.
Apr 30 16:29:30 DietPi dhcpd[1577]: All rights reserved.
Apr 30 16:29:30 DietPi dhcpd[1577]: For info, please visit https://www.isc.org/software/dhcp/
Apr 30 16:29:30 DietPi dhcpd[1577]: Wrote 0 leases to leases file.
Apr 30 16:29:30 DietPi dhcpd[1577]:
Apr 30 16:29:30 DietPi dhcpd[1577]: No subnet declaration for wlan0 (no IPv4 addresses).
Apr 30 16:29:30 DietPi dhcpd[1577]: ** Ignoring requests on wlan0.  If this is not what
Apr 30 16:29:30 DietPi dhcpd[1577]:    you want, please write a subnet declaration
Apr 30 16:29:30 DietPi dhcpd[1577]:    in your dhcpd.conf file for the network segment
Apr 30 16:29:30 DietPi dhcpd[1577]:    to which interface wlan0 is attached. **
Apr 30 16:29:30 DietPi dhcpd[1577]:
Apr 30 16:29:30 DietPi dhcpd[1577]:
Apr 30 16:29:30 DietPi dhcpd[1577]: Not configured to listen on any interfaces!
Apr 30 16:29:30 DietPi dhcpd[1577]:
Apr 30 16:29:30 DietPi dhcpd[1577]: If you think you have received this message due to a bug rather
Apr 30 16:29:30 DietPi dhcpd[1577]: than a configuration issue please read the section on submitting
Apr 30 16:29:30 DietPi dhcpd[1577]: bugs on either our web page at www.isc.org or in the README file
Apr 30 16:29:30 DietPi dhcpd[1577]: before submitting a bug.  These pages explain the proper
Apr 30 16:29:30 DietPi dhcpd[1577]: process and the information we find helpful for debugging.
Apr 30 16:29:30 DietPi dhcpd[1577]:
Apr 30 16:29:30 DietPi dhcpd[1577]: exiting.
Apr 30 16:29:32 DietPi isc-dhcp-server[1559]: Starting ISC DHCPv4 server: dhcpdcheck syslog for diagnostics. ...
Apr 30 16:29:32 DietPi isc-dhcp-server[1582]:  failed!
Apr 30 16:29:32 DietPi isc-dhcp-server[1583]:  failed!
Apr 30 16:29:32 DietPi systemd[1]: isc-dhcp-server.service: Control process exited, code=exited, status=1/FAILURE
Apr 30 16:29:32 DietPi systemd[1]: isc-dhcp-server.service: Failed with result 'exit-code'.
Apr 30 16:29:32 DietPi systemd[1]: Failed to start LSB: DHCP server.

MichaIng, 2023-04-30:
And the content of both config files?
```sh
cat /etc/dhcp/dhcpd.conf
cat /etc/default/isc-dhcp-server
```

danilabi, 2023-04-30:
root@DietPi:~# cat /etc/dhcp/dhcpd.conf
ddns-update-style none;
default-lease-time 600;
max-lease-time 7200;
authoritative;
log-facility local7;

subnet 192.168.42.0 netmask 255.255.255.0 {
        range 192.168.42.10 192.168.42.50;
        option broadcast-address 192.168.42.255;
        option routers 192.168.42.1;
        option domain-name "local";
        option domain-name-servers 9.9.9.9, 149.112.112.112;
}

root@DietPi:~# cat /etc/default/isc-dhcp-server
INTERFACESv4="wlan0"

MichaIng, 2023-04-30:
Ah, and is the interface up with `192.168.42.1` statically assigned?
```sh
ip a
cat /etc/network/interfaces
```

danilabi, 2023-04-30:
root@DietPi:~# ip a
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host
       valid_lft forever preferred_lft forever
2: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc mq state UP group default qlen 1000
    link/ether 62:ab:c1:c1:2d:9a brd ff:ff:ff:ff:ff:ff
    inet 10.0.0.44/24 brd 10.0.0.255 scope global dynamic eth0
       valid_lft 84762sec preferred_lft 84762sec
    inet6 fe80::60ab:c1ff:fec1:2d9a/64 scope link
       valid_lft forever preferred_lft forever
3: wlan0: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc mq state DOWN group default qlen 1000
    link/ether 00:13:ef:40:e0:b1 brd ff:ff:ff:ff:ff:ff
root@DietPi:~# cat /etc/network/interfaces
# Location: /etc/network/interfaces
# Please modify network settings via: dietpi-config
# Or create your own drop-ins in: /etc/network/interfaces.d/

# Drop-in configs
source interfaces.d/*

# Ethernet
allow-hotplug eth0
iface eth0 inet dhcp
address 192.168.0.100
netmask 255.255.255.0
gateway 192.168.0.1
#dns-nameservers 10.0.0.138 10.0.0.138

# WiFi
allow-hotplug wlan0
iface wlan0 inet static
address 192.168.42.1
netmask 255.255.255.0
#gateway 192.168.0.1
#dns-nameservers 9.9.9.9 149.112.112.112

# iptables NAT rules
up iptables-restore < /etc/iptables.ipv4.nat
up ip6tables-restore < /etc/iptables.ipv6.nat
root@DietPi:~# systemctl stop dnsmasq
Failed to stop dnsmasq.service: Unit dnsmasq.service not loaded.
root@DietPi:~# killall dnsmasq
dnsmasq: no process found

MichaIng, 2023-04-30:
The interface is shown as up but somehow no IP address is applied. Probably `dnsmasq` messed with it. Try:
```sh
ifdown wlan0
ifup wlan0
systemctl restart isc-dhcp-server
```

danilabi, 2023-04-30:
root@DietPi:~# ifdown wlan0
root@DietPi:~# ifup wlan0
root@DietPi:~# systemctl restart isc-dhcp-server
root@DietPi:~# ip a
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host
       valid_lft forever preferred_lft forever
2: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc mq state UP group default qlen 1000
    link/ether 62:ab:c1:c1:2d:9a brd ff:ff:ff:ff:ff:ff
    inet 10.0.0.44/24 brd 10.0.0.255 scope global dynamic eth0
       valid_lft 82783sec preferred_lft 82783sec
    inet6 fe80::60ab:c1ff:fec1:2d9a/64 scope link
       valid_lft forever preferred_lft forever
3: wlan0: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc mq state DOWN group default qlen 1000
    link/ether 00:13:ef:40:e0:b1 brd ff:ff:ff:ff:ff:ff
    inet 192.168.42.1/24 brd 192.168.42.255 scope global wlan0
       valid_lft forever preferred_lft forever

Joulinar, 2023-04-30:
Ok that looks good now.

danilabi, 2023-04-30:
yes but i still cant find the hotspot

MichaIng, 2023-04-30:
```sh
systemctl restart hostapd
sleep 3
journalctl -u hostapd
```

danilabi, 2023-04-30:
Apr 30 18:30:51 DietPi systemd[1]: Started LSB: Access point and authentication server for Wi-Fi and Ethernet.
Apr 30 18:31:18 DietPi systemd[1]: Stopping LSB: Access point and authentication server for Wi-Fi and Ethernet...
Apr 30 18:31:18 DietPi hostapd[2835]: Stopping advanced IEEE 802.11 management: hostapd.
Apr 30 18:31:18 DietPi systemd[1]: hostapd.service: Succeeded.
Apr 30 18:31:18 DietPi systemd[1]: Stopped LSB: Access point and authentication server for Wi-Fi and Ethernet.
Apr 30 18:31:18 DietPi systemd[1]: Starting LSB: Access point and authentication server for Wi-Fi and Ethernet...
Apr 30 18:31:19 DietPi hostapd[2839]: Starting advanced IEEE 802.11 management: hostapd
Apr 30 18:31:19 DietPi hostapd[2842]: ioctl[SIOCSIWMODE]: Invalid argument
Apr 30 18:31:19 DietPi hostapd[2843]:  failed!
Apr 30 18:31:19 DietPi systemd[1]: Started LSB: Access point and authentication server for Wi-Fi and Ethernet.

MichaIng, 2023-04-30:
This is after installing the Realtek variant of `hostapd` from here, right? https://github.com/MichaIng/DietPi/issues/6345#issuecomment-1528874650

If so, then it seems that indeed the chip is not one of the affected ones. To revert:
```sh
apt install hostapd
G_CONFIG_INJECT 'driver=' 'driver=nl80211' /etc/hostapd/hostapd.conf
systemctl restart hostapd
sleep 3
journalctl -u hostapd
```
If this fails [as before](https://github.com/MichaIng/DietPi/issues/6345#issuecomment-1527325969), then I am out of ideas. Note that it is perfectly possible that this dongle does not support `hostapd` (for other reasons than the RTL8188C* ones) or cannot be used as access point at all. Not all do. Of course you can test it on other OSes like Windows, or check for other Linux access point software which does not use `hostapd`. AFAIK NetworkManager can do it as well, but be aware that it may conflict with the `ifupdown` setup currently in use, so you'd need to tinker to get this working.

danilabi, 2023-04-30:
Ok so i simply cant use my wlan stick for an acces point. Do you know one, that I can use?

MichaIng, 2023-04-30:
I don't like linking this platform, but there is no other international one. I just tested this adapter and it works well providing a hostspot even with WiFi 5 (802.11ac) : https://www.amazon.de/gp/product/B08SCQC4W6
But this particular model has no external antenna, and commonly those tiny ones which fit into such dongle have a very long range. I'm using it only up to 3 meters from the router for testing, so it doesn't matter here, but for larger ranges I'd recommend one with external antennas or one where tests/reviews show that they have a good range. Generally the RTL8811CU (used by this dongle) supports it.

One downside is that it does not have a driver in mainline Linux. Armbian adds the available 3rd party drivers to their kernel builds (which we benefit from), but e.g. on RPi and other Linux distros, one might need to compile the driver. There are quite some GitHub sources which make it comfortable via DKMS: https://duckduckgo.com/?q=RTL8811CU+Linux
But the problem is that they are not perfectly maintained and often break for new kernel releases. See the open PRs on each repo for fixing build with recent Linux versions. But this all doesn't matter with Armbian kernel 🙂.

danilabi, 2023-04-30:
Ok thank you very much!

MichaIng, 2023-08-10:
I'm closing this issue. There are not questions open, right?

closed 2023-08-10 by MichaIng