MichaIng/DietPi #6347: dietpi-update shows NO_PUBKEY for Grafana error

n0valis, 2023-04-28:
I wanted to update DietPi to latest then this happened. 
I got this error before when I tried to update Grafana. I manually installed the new key and then updated Grafana successfully. DietPi doesn't seem to find the new key. 

#### Details:
- Date           | Fri Apr 28 16:42:57 CEST 2023
- DietPi version | v8.7.1 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | From scratch
- Hardware       | RPi 4 Model B (aarch64) (ID=4)
- Kernel version | `Linux THORIN 6.1.19-v8+ #1637 SMP PREEMPT Tue Mar 14 11:11:47 GMT 2023 aarch64 GNU/Linux`
- Distro         | bullseye (ID=6,RASPBIAN=0)
- Command        | `apt-get -y update`
- Exit code      | 255
- Software title | DietPi-Update
#### Steps to reproduce:

Grafana key was rotated. DietPi can't handle it. 

#### Additional logs:
```
Hit:1 http://ftp.de.debian.org/debian bullseye InRelease
Hit:2 http://ftp.de.debian.org/debian bullseye-updates InRelease
Hit:3 https://repos.influxdata.com/debian stable InRelease
Hit:4 https://deb.debian.org/debian-security bullseye-security InRelease
Hit:5 http://ftp.de.debian.org/debian bullseye-backports InRelease
Hit:6 https://deb.nodesource.com/node_12.x bullseye InRelease
Hit:7 https://archive.raspberrypi.org/debian bullseye InRelease
Hit:8 https://deb.nodesource.com/node_14.x bullseye InRelease
Hit:9 https://apt.grafana.com stable InRelease
Err:9 https://apt.grafana.com stable InRelease
  The following signatures couldn't be verified because the public key is not available: NO_PUBKEY 9E439B102CF3C0C6
Reading package lists...
W: An error occurred during the signature verification. The repository is not updated and the previous index files will be used. GPG error: https://apt.grafana.com stable InRelease: The following signatures couldn't be verified because the public key is not available: NO_PUBKEY 9E439B102CF3C0C6
W: Failed to fetch https://apt.grafana.com/dists/stable/InRelease  The following signatures couldn't be verified because the public key is not available: NO_PUBKEY 9E439B102CF3C0C6W: Some index files failed to download. They have been ignored, or old ones used instead.
```
---------------------------------------------------------------------
[FAILED] DietPi-Update | Unable to continue, DietPi-Update will now terminate.

MichaIng, 2023-04-28:
The key should be updated via pre-patch. Is there not some message prior to this step
```
[ INFO ] Updating Grafana APT repo key
```
If not, can you show the output of this command:
```sh
date -u '+%Y%m%d' -r '/etc/apt/trusted.gpg.d/dietpi-grafana.gpg'
```

n0valis, 2023-04-28:
The output is: 20230428

But I managed to do the update now. 
I removed grafana.list from /etc/apt/sources.list.d
Content was
deb https://apt.grafana.com/ stable main 

Then it went through.

MichaIng, 2023-04-28:
But then you do not get updates for the Grafana server anymore.

> The output is: 20230428

That should work. I just checked on our server + redownloaded the key and they are identical + work. Probably something went wrong with the import of this key. Can be redone via:
```sh
curl -sSfL 'https://apt.grafana.com/gpg.key' | gpg --dearmor -o /etc/apt/trusted.gpg.d/dietpi-grafana.gpg --yes
```

Joulinar, 2023-04-28:
And Grafana source list would need to be added again.

n0valis, 2023-04-29:
I added grafana.list to /etc/apt/sources.list.d again.
So far I don't know how to test it now as everything is up to date.

Joulinar, 2023-04-29:
you should see source liste configuration working if you just run

```
apt update
```

n0valis, 2023-04-29:
I see no errors and everything up to date for apt update.

MichaIng, 2023-04-29:
Okay great. Not sure what the issue was in the first place, but as it works now, I'll mark this as closed.

closed 2023-04-29 by MichaIng