MichaIng/DietPi #6349: Add firewall setting for make a dietpi a gateway VPN ( ex. nordvpn, protonvpn)

black00019, 2023-04-29:
### Creating a feature request

#### Is your feature request related to a problem? Please describe:

- Having a dietpi with pihole (dhcp), vpn server (pivpn - wireguard) and a connection with vpn service like nordvpn or protonvpn ( via dietpi-vpn), I would like my dietpi to become a gateway to my local network so that i connected clients navigate with the nordvpn vpn connection or other, at the same time I would like that the external clients connected with pivpn to the dietpi could always use the nordvpn or proton connection of the dietpi as a gateway, all this should be able to be enabled as desired by the user, without fiddling with difficult firewall configurations.

#### Describe the solution you'd like:

-  i have a partial solution posted in forum:
[]([https://dietpi.com/forum/t/dietpi-vpn-gateway-with-pivpn-server-and-pihole/16620/3])

MichaIng, 2023-04-29:
Many thanks for your request. So basically VPN server + `dietpi-vpn` and then the forwarding rules adjusted to not forward incoming requests from the VPN server to the internet facing network interface, but to the VPN client interface.

Our OpenVPN server implementation has no forwarding rules added OOTB, and I'm not even sure whether this server works well with `dietpi-vpn` or whether they both try to use the `tun0` interface 🤔. With the WireGuard server it should be however pretty simple: In `/etc/wireguard/wg0.conf` replace all `eth0` or `wlan0` with `tun0`. And the killswitch cannot be used, or it would need to be modified (rules added) to allow packets/connections through `wg0`.

Joulinar, 2023-04-30:
Yes you can have both working together. It's a matter if `iptable `rules to implement. But I'm not a network export. usually @trendy is the Master of Network. We have a couple of these request in our forum. Not sure if this will be possible for all scenario we offer/support.

black00019, 2023-04-30:
Thanks for the reply
As a first proposal I would make the connection via dietpi-vpn work, as I noticed that once connected with protonvpn, from the dietpi console you can no longer navigate in any way.
As a second request instead I would make sure that the dietpi-vpn connection becomes an exit gateway for all clients connected to both the lan and wireless network, and that it is settable (my dietpi has also installed pi-hole which acts as a dhcp server) .

MichaIng, 2023-04-30:
> As a first proposal I would make the connection via dietpi-vpn work, as I noticed that once connected with protonvpn, from the dietpi console you can no longer navigate in any way.

Would be great if you could open a dedicated issue about this. This of course shouldn't happen, not even with SSH when enabling the killswitch. But the killswitch is expected to break incoming VPN connections. Basically for every incoming connection (port), an additional rule needs to be applied if the killswitch is active.

> As a second request instead I would make sure that the dietpi-vpn connection becomes an exit gateway for all clients connected

The VPN is already made the default gateway, the only missing thing indeed are the forwarding rules like the ones applied in the WireGuard settings. Without them, incoming requests would land in the ether.