MichaIng/DietPi #6351: Odroid HC4 with microSD to emmc adapter does not boot

cozybear-dev, 2023-04-29:
Hi,

I'm aware this is not officially supported, but wanted to open this issue, so perhaps other people might be able to help. It should work? Sadly the HC4 does not have eMMC port exposed, hence going this route to at least have some better performance.

My setup;
- Odroid HC4
- eMMC module 32GB (orange square) using microSD eMMC Reader (v0.2)
- the latest dietpi image

To be quick, it does not work. No HDMI output, nothing works. Only red LED stable, no blue light whatsoever. 

A regular microSD does work. 

Note that for the above two scenario's the exact same steps are used as documented (balanaetcher etc).

The preinstalled OS on the eMMC does interestingly work, making me believe it would be possible. I just don't know where to look to figure this out.

cozybear-dev, 2023-04-29:
Hm, so my HC4 shipped with uboot. Just installed the latest petitboot. According to petitboot, a valid partition table is missing. The disk itself is recognized just fine.

cozybear-dev, 2023-04-29:
To remove petitboot, go to shell;

# flash_eraseall /dev/mtd0
# flash_eraseall /dev/mtd1
# flash_eraseall /dev/mtd2
# flash_eraseall /dev/mtd3

To install petitboot;
https://minskio.co.uk/restoring-petitboot-on-an-odroid-hc4/

cozybear-dev, 2023-04-29:
Few of the key differences between the mmc module that works, and dietpi that doesn't;

Working module;
- uses explicit `load mmc` in boot.ini whilst dietpi does not
- boot partition is fat16 readable, dietpi is one single ext4 partition nowadays

This is the whole boot.ini from the working module;
```
ODROIDC4-UBOOT-CONFIG

setenv bootlabel "Hardkernel Ubuntu 20.04"

setenv board "odroidc4"
setenv display_autodetect "true"
setenv hdmimode "1080p60hz"
setenv monitor_onoff "false" # true or false
setenv overscan "100"
setenv sdrmode "auto" 
setenv voutmode "hdmi"
setenv disablehpd "false"
setenv cec "true"
setenv disable_vu7 "true"
setenv max_freq_a55 "1908"    # 1.908 GHz, default value
setenv maxcpus "4"
setenv enable_wol "0"

# Set load addresses
setenv dtb_loadaddr "0x10000000"
setenv dtbo_addr_r "0x11000000"
setenv k_addr "0x1100000"
setenv loadaddr "0x1B00000"
setenv initrd_loadaddr "0x3700000"

if test "${variant}" = "hc4"; then
       setenv max_freq_a55 "1800"
fi

load mmc ${devno}:1 ${loadaddr} config.ini \
    && ini generic ${loadaddr}
if test "x${overlay_profile}" != "x"; then
    ini overlay_${overlay_profile} ${loadaddr}
fi

setenv condev "console=ttyS0,115200n8"   # on both

### Normal HDMI Monitors
if test "${display_autodetect}" = "true"; then hdmitx edid; fi
if test "${hdmimode}" = "custombuilt"; then setenv cmode "modeline=${modeline}"; fi
if test "${cec}" = "true"; then setenv cec_enable "hdmitx=cec3f"; fi
if test "${disable_vu7}" = "false"; then setenv hid_quirks "usbhid.quirks=0x0eef:0x0005:0x0004"; fi

# Boot Args
setenv bootargs "root=UUID=e139ce78-9841-40fe-8823-96a304a09859 rootwait rw ${condev} ${amlogic} no_console_suspend fsck.repair=yes net.ifnames=0 elevator=noop hdmimode=${hdmimode} cvbsmode=576cvbs max_freq_a55=${max_freq_a55} maxcpus=${maxcpus} voutmode=${voutmode} ${cmode} disablehpd=${disablehpd} cvbscable=${cvbscable} overscan=${overscan} ${hid_quirks} monitor_onoff=${monitor_onoff} logo=osd0,loaded ${cec_enable} sdrmode=${sdrmode} enable_wol=${enable_wol}"

# Load kernel, dtb and initrd
load mmc ${devno}:1 ${k_addr} Image.gz
load mmc ${devno}:1 ${dtb_loadaddr} amlogic/meson64_odroid${variant}.dtb
load mmc ${devno}:1 ${initrd_loadaddr} uInitrd
fdt addr ${dtb_loadaddr}

if test "x{overlays}" != "x"; then
    fdt resize ${overlay_resize}
    for overlay in ${overlays}; do
        load mmc ${devno}:1 ${dtbo_addr_r} amlogic/overlays/${board}/${overlay}.dtbo \
            && fdt apply ${dtbo_addr_r}
    done
fi

# unzip the kernel
unzip ${k_addr} ${loadaddr}

# boot
booti ${loadaddr} ${initrd_loadaddr} ${dtb_loadaddr}
```

cozybear-dev, 2023-04-29:
Files inside the boot partition;

[image]

cozybear-dev, 2023-04-29:
I've tried a fuck ton by now. Tried various images, from various sources, using both the latest uboot + petitboot. What I think, is that the issue lies in that it needs to load as an emmc device - which dietpi (nor any other OS) is made for with the HC4. HC4 does not officially support emmc, so that would make sense as well.

The fact the preinstalled image works with the adapter, proves it's possible. If anyone has any suggestions to test with dietpi, please let me know! Eager to find the solution here.

cozybear-dev, 2023-04-29:
https://github.com/MichaIng/DietPi/blob/9c701dd8e3349694d7d83a6aa8c62cb9fb303b6a/.build/images/OdroidC1/boot.ini#L85
https://github.com/MichaIng/DietPi/blob/master/.build/images/OdroidXU4/boot.ini#L179

Some builds do seem to do something with it?

cozybear-dev, 2023-04-29:
What's interesting as well, is that the official ubuntu images - that the emmc modules are shipped with, also don't work after flashing with etcher. It seems hardkernel does some unknown magic, that makes it work.

MichaIng, 2023-04-29:
> Few of the key differences between the mmc module that works, and dietpi that doesn't;

What do you mean by module? `boot.ini` is deprecated and used by legacy vendor U-Boot versions only (which we indeed still use for Odroid C1 and XU4 due to lack of proper upstream support). The boot partition content shown looks like form one of the Hardkernel vendor images, isn't it? Nowadays one uses either `boot.scr`, extlinux or EFI. Hardkernel's petitboot basically embeds an ancient vendor U-Boot build. Recently I had quite some efforts to add petitboot support, and it works often, but not very reliably, without any further way or support from Hardkernel do debug and make it stable. Since there also seem to be no efforts done by Hardkernel to to a major upgrade in this regards and support + embed modern U-Boot + the `boot.scr` syntax, extlinux or any such, this likely won't change and is basically doomed to fail to ever work well with modern distros. AFAIK, DietPi is the only one which supports petitboot at all.

> It seems hardkernel does some unknown magic, that makes it work.

Jep, and any attempt of me to get some more info about this was ignored, or help debugging, e.g. why on my Odroid N2+ petitboot is able to boot from USB and SD card but not from eMMC, quite similar than our issue. Forther others however it worked with eMMC while booting from certain USB drives instead failed, all very inconsistent.

But, booting from neither SD card nor eMMC should require an SPI bootloader.

> uses explicit `load mmc` in boot.ini whilst dietpi does not

The `${devtype}` we use equals `mmc` for SD cards and eMMC storages, for e.g. `usb` for USB drives etc. Using this variable is standard for generic boot scripts with support for all storage types. This is also correctly passed by petitboot or the embedded U-Boot.

> boot partition is fat16 readable, dietpi is one single ext4 partition nowadays

Generally also petitboot and even Hardkernel's `boot.ini` support ext4 as boot filesystem.

> According to petitboot, a valid partition table is missing.

There is an option in the menu to scan for drives. Does it find and show the boot partition correctly after doing so? In case of my Odroid N2+, it does, but still fails to boot from it.

Do do not have a UART adapter to check via serial console the issue when booting with mainline U-Boot from the eMMC directly, do you?

cozybear-dev, 2023-04-29:
Thank you for the extensive reply!

Let me clarify. I have two 32GB emmc modules with linux preinstalled (https://www.antratek.nl/32gb-emmc-module-with-linux-for-c4). Both worked out of the box (as in they booted succesfully), with the adapter (https://www.antratek.nl/emmc-module-reader).

I do come further with the images listed here (but still minor issues blocking a full boot); http://docs.linuxfactory.or.kr/install/odroidhc4/image.html

I have some generic UART adapters, but haven't been able to get them working with my HC4. For now I'm just trying other things, so I don't have to.

cozybear-dev, 2023-04-29:
This is the furthest I've gotten with my own flashing;

Using https://ppa.linuxfactory.or.kr/images/raw/arm64/jammy/ubuntu-22.04-server-odroidhc4-20220531.img.xz

I now boot partially, but at least have a shell (initramfs). The error that is currently being show every x seconds;

"mmc0: error -84 whilst initialising mmc"

MichaIng, 2023-04-29:
Do I understand it correctly that currently this SD-to-eMMC adapter only works with Hardkernel's Ubuntu + petitboot, or did they also boot before you flashed petitboot?

cozybear-dev, 2023-04-29:
The only successful boot I ever had, was with the preinstalled (so untouched by me) "Linux" installation using the SD-to-eMMC adapter.

cozybear-dev, 2023-04-29:
The HC4 was also completely untouched, I'd done nothing to the SPI storage (bootloader).

MichaIng, 2023-04-29:
> The HC4 was also completely untouched, I'd done nothing to the SPI storage (bootloader).

Okay, then it was with petitboot. I guess these Ubuntu images will work as well: https://wiki.odroid.com/odroid-hc4/getting_started/os_installation_guide#tab__odroid-c4hc4

cozybear-dev, 2023-04-29:
I'm not so sure that petitboot was installed tho. When no sd card was inserted, there was no boot occurring. Once I did had petitboot installed, I was able to boot into petitboot. Before that, HC4 would do litterally nothing.

The behaviour of my HC4 matches the behaviour of no petitboot installed.

I've already tested the images (both mate and minimal) at that URL for several ubuntu versions and none of them worked. Thanks for the suggestion though.

MichaIng, 2023-04-30:
> When no sd card was inserted, there was no boot occurring. Once I did had petitboot installed, I was able to boot into petitboot. Before that, HC4 would do litterally nothing.

Ah okay, probably you erased it earlier.

> I've already tested the images (both mate and minimal) at that URL for several ubuntu versions and none of them worked.

Hmm strange, as these should mostly be the same than what the eMMC modules are shipped with. The images are dated Juli 2022, so not very new.

Does the adapter work well when you e.g. boot from USB or SATA via petitboot and then attach the adapter?