MichaIng/DietPi #6352: Open Beta v8.17 | Please help testing and hardening our upcoming release

MichaIng, 2023-04-30:
RC version | v8.17.0
----------- | -------
v8.17.0 PR | #6343
Raw changelog | https://github.com/MichaIng/DietPi/blob/beta/CHANGELOG.txt
Code changes | https://github.com/MichaIng/DietPi/compare/master...beta
Release date | **2023-05-06**

### How to apply: https://github.com/MichaIng/DietPi/blob/master/BRANCH_SYSTEM.md

#### Related/solved issues: https://github.com/MichaIng/DietPi/issues?q=is%3Aissue+milestone%3Av8.17
____
### Beta v8.17.0
_(2023-04-29)_

#### New software
- openHAB | This long requested vendor and technology agnostic FLOSS home automation software has been finally added to DietPi. Many thanks to @just-jason and many others for requesting it and @MDAR for providing install instructions and valuable information: https://github.com/MichaIng/DietPi/issues/3857
- Moonlight | This game streaming client for Sunshine and NVIDIA GameStream has been added to our software catalogue. Many thanks to @mtekman for implementing it: https://github.com/MichaIng/DietPi/pull/6303
- Moonlight Qt | This GUI game streaming client for Sunshine and NVIDIA GameStream has been added to our software catalogue. Many thanks to @mtekman for implementing it: https://github.com/MichaIng/DietPi/pull/6339

#### Enhancements
- NanoPi R series | Updated udev rules for the Ethernet LEDs to not lid the LEDs of disabled Ethernet devices. If an Ethernet device has been detected by the kernel/udev already and an LED is configured to light on link (connected cable), it lights until the interface is set up and no link is detected. So for disabled interfaces, some LEDs remained lit. The udev rules have now been changed to quickly set up and down the interfaces for the LEDs to remain off until an interface has been successfully configured.
- NanoPi R6S | Since our image runs fine on NanoPi R6C as well, the device name in DietPi is now shown as "NanoPi R6S/R6C" to indicate this fact.
- ROCK Pi 4 | We followed Radxa and renamed the device to "ROCK 4", i.e. removed the "Pi" from its name.
- DietPi-Banner | Added a new option to show the Let's Encrypt certificate status (expiry date), when installed via dietpi-letsencrypt or Certbot. Many thanks to @gary2002 for implementing this option: https://github.com/MichaIng/DietPi/pull/6314

#### Bug fixes
- General | Resolved an issue where our recent images had the cron service masked accidentally after first run setup. Many thanks to `@Johannes` for reporting this issue: https://dietpi.com/forum/t/cron-service-is-masked/16544
- Raspberry Pi | Resolved an issue where enabling the 3.5mm analogue onboard audio jack did not work if a HDMI device was attached, since due to a change with Linux 6.1 HDMI audio was not disabled correctly. Many thanks to `@Gale` for reporting this issue: https://dietpi.com/forum/t/no-audio-from-rpi-4b-headphone-jack/16538
- DietPi-Globals | Resolved a DietPi v8.16 regression where wrong CPU temperatures were shown on some devices. Many thanks to @duledxb for reporting this issue: https://github.com/MichaIng/DietPi/issues/6315
- DietPi-Config | Audio: 'firmware-sof-signed' required for functionality on some Intel devices, is now installed when selecting 'intel-sst-dsp': https://github.com/MichaIng/DietPi/issues/6281#issuecomment-1500990841
- DietPi-Software | Resolved an issue where it was possible to install software titles via AUTO_SETUP_INSTALL_SOFTWARE_ID dietpi.txt entries on unsupported platforms.
- DietPi-Software | Home Assistant: Resolved an issue where to installation on 32-bit ARM systems failed since Python cryptography source builds do now require pky-config. Many thanks to `@retrofame` for reporting this issue: https://dietpi.com/forum/t/unable-to-upgrade-home-assistant-to-2021-10-x/5823/7
- DietPi-Software | HTPC Manager/Synapse: Resolved an issue where to installation on 32-bit ARM Bookworm systems failed since Python cryptography source builds do now require pky-config.
- DietPi-Software | microblog.pub: Resolved an issue where the installation failed on 32-bit ARM devices since a Rust compiler is required for the cryptography Python module build. Many thanks to @kinoushe for reporting this issue: https://github.com/MichaIng/DietPi/issues/6304
- DietPi-Software | Home Assistant/HTPC Manager/microblog-pub/Synapse: Resolved an issue where the install on RPi 4 with 32-bit image failed since rustup/cargo tried to compile for 64-bit. The issue occurred since the 64-bit kernel is used on RPi 4 by default since Linux 6.1, even if the OS/userland is 32-bit. Many thanks to @josh3003 for reporting this issue: https://github.com/MichaIng/DietPi/issues/6306
- DietPi-Software | Home Assistant: Resolved an issue where the installation failed on 32-bit ARM systems due to newly required C++ compiler and FFmpeg libraries for compiling all required Python modules.
- DietPi-Software | Google AIY: Resolved an issue where the install failed on ARMv7 systems due to conflicting Python module dependencies.

kinoushe, 2023-04-30:
I'm guessing this isn't live yet, as the beta branch is still reporting v8.16.2?

MichaIng, 2023-04-30:
Now it is 🙂.

viseshrp, 2023-05-05:
Hi, I am new here and would like to help beta-test. Are there any instructions to upgrade my existing installation to the beta version?

kinoushe, 2023-05-05:
Hi @viseshrp, yes, just follow the instructions in the "How to apply" link at the top of the post.

MichaIng, 2023-05-05:
### Beta v8.17.1 has been merged: #6359

MichaIng, 2023-05-06:
### Beta v8.17.2 has been merged: #6366

MichaIng, 2023-05-06:
### DietPi v8.17 has been released: #6361

Many thanks to all testers ❤️ !

closed 2023-05-06 by MichaIng