MichaIng/DietPi #6368: TimeSync issue when updating to 8.16.2

nzbaxterman, 2023-05-06:
#### Details:
- Date           | Sun May  7 08:16:20 NZST 2023
- DietPi version | v8.15.2 (MichaIng/master)
- Image creator  |
- Pre-image      |
- Hardware       | RPi B+ (armv6l) (ID=1)
- Kernel version | `Linux Tarrant 4.14.24+ #1097 Mon Mar 5 16:16:18 GMT 2018 armv6l GNU/Linux`
- Distro         | bullseye (ID=6,RASPBIAN=1)
- Command        | `systemctl start systemd-timesyncd`
- Exit code      | 1
- Software title | DietPi-TimeSync
#### Steps to reproduce:

1. dietpi-update

#### Expected behaviour:

dietpi update should run without error
#### Actual behaviour:

Failed to start systemd-timesyncd.service: Unit systemd-timesyncd.service is masked.
I tried the retry option a few times - no change
Then selected the option to update locale option.  Set my locale and timezone from the lists several times, but it just kept looping updating the boot.txt with the locale, but then asking to set again.
#### Extra details:

- ...
#### Additional logs:
```
Failed to start systemd-timesyncd.service: Unit systemd-timesyncd.service is masked.
```

MichaIng, 2023-05-06:
> Linux Tarrant 4.14.24+

How is it possible that you run such an outdated kernel on Raspberry Pi? Can you show:
```sh
dpkg -l raspberrypi-kernel
ls -l /boot /lib/modules
apt install raspberrypi-kernel
```

The mask however is caused as you upgraded to Bullseye but obviously did not follow all steps in [our guide](https://dietpi.com/blog/?p=811) 😉:
```sh
apt install systemd-timesyncd
systemctl unmask systemd-timesyncd
/boot/dietpi/func/run_ntpd 1
```

nzbaxterman, 2023-05-07:
Yes - I am running a very old kernel.  For no particular reason other than
this Pi has been so stable I barely touch it!

Thanks for the pointers - RESOLVED

***@***.***:~$ dpkg -l raspberrypi-kernel
Desired=Unknown/Install/Remove/Purge/Hold
|
Status=Not/Inst/Conf-files/Unpacked/halF-conf/Half-inst/trig-aWait/Trig-pend
|/ Err?=(none)/Reinst-required (Status,Err: uppercase=bad)
||/ Name               Version      Architecture Description
+++-==================-============-============-=================================
hi  raspberrypi-kernel 1.20171029-1 armhf        Raspberry Pi bootloader
***@***.***:~$
***@***.***:~$ ls -l /boot /lib/modules
/boot:
total 21620
-rwxr-xr-x 1 root root   22284 Mar 12  2018  bcm2708-rpi-0-w.dtb
-rwxr-xr-x 1 root root   21805 Mar 12  2018  bcm2708-rpi-b.dtb
-rwxr-xr-x 1 root root   22064 Mar 12  2018  bcm2708-rpi-b-plus.dtb
-rwxr-xr-x 1 root root   21518 Mar 12  2018  bcm2708-rpi-cm.dtb
-rwxr-xr-x 1 root root   23068 Mar 12  2018  bcm2709-rpi-2-b.dtb
-rwxr-xr-x 1 root root   24276 Mar 12  2018  bcm2710-rpi-3-b.dtb
-rwxr-xr-x 1 root root   22996 Mar 12  2018  bcm2710-rpi-cm3.dtb
-rwxr-xr-x 1 root root   52064 Mar 12  2018  bootcode.bin
-rwxr-xr-x 1 root root     171 Jan 22 12:39  cmdline.txt
-rwxr-xr-x 1 root root    1984 Jan 23 11:38  config.txt
-rwxr-xr-x 1 root root   18693 Feb 19  2018  COPYING.linux
drwxr-xr-x 4 root root    4608 May  7 08:56  dietpi
-rwxr-xr-x 1 root root    6092 Feb 19  2018  dietpi-README.md
-rwxr-xr-x 1 root root   10641 May  7 08:49  dietpi.txt
-rwxr-xr-x 1 root root    2592 Mar 12  2018  fixup_cd.dat
-rwxr-xr-x 1 root root    6570 Mar 12  2018  fixup.dat
-rwxr-xr-x 1 root root    9725 Mar 12  2018  fixup_db.dat
-rwxr-xr-x 1 root root    9723 Mar 12  2018  fixup_x.dat
-rwxr-xr-x 1 root root     145 Nov 29  2017  issue.txt
-rwxr-xr-x 1 root root 4827200 Mar 12  2018  kernel7.img
-rwxr-xr-x 1 root root 4621952 Mar 12  2018  kernel.img
-rwxr-xr-x 1 root root    1494 Feb 19  2018  LICENCE.broadcom
-rwxr-xr-x 1 root root   18974 Nov 29  2017  LICENSE.oracle
drwxr-xr-x 2 root root   11264 Mar 12  2018  overlays
-rwxr-xr-x 1 root root  671236 Mar 12  2018  start_cd.elf
-rwxr-xr-x 1 root root 4965508 Mar 12  2018  start_db.elf
-rwxr-xr-x 1 root root 2822660 Mar 12  2018  start.elf
-rwxr-xr-x 1 root root 3909604 Mar 12  2018  start_x.elf
drwxr-xr-x 2 root root     512 Jan 11  2018 'System Volume Information'

/lib/modules:
total 16
drwxr-xr-x 3 root root 4096 Mar 12  2018 4.14.24+
drwxr-xr-x 3 root root 4096 Mar 12  2018 4.14.24-v7+
drwxr-xr-x 3 root root 4096 Feb 19  2018 4.9.59+
drwxr-xr-x 3 root root 4096 Feb 19  2018 4.9.59-v7+

closed 2023-05-07 by Joulinar

MichaIng, 2023-05-07:
> Yes - I am running a very old kernel.  For no particular reason other than this Pi has been so stable I barely touch it!

As kernel upgrades also include security updates, I highly recommend to keep it updated. With the APT packages it is quite simple:
```sh
apt-mark unhold raspberrypi-linux raspberrypi-bootloader
apt upgrade
```

nzbaxterman, 2023-05-08:
Thanks Michalng.  I'm getting a:
E: Unable to locate package raspberrypi-linux

on the above...

Joulinar, 2023-05-08:
> E: Unable to locate package raspberrypi-linux

I guess it should be `raspberrypi-kernel` instead