MichaIng/DietPi #6369: Domoticz issue to start and bug report for re-install

mcnahum, 2023-05-08:
### Creating a bug report/issue

#### Required Information

DietPi version: v8.17.2 (MichaIng/master) | HW_MODEL: 4 | HW_ARCH: 2 | DISTRO: 6   

#### Additional Information (if applicable)
- Software title | (EG: Nextcloud) Domoticz
- Was the software title installed freshly or updated/migrated?  Trying to re-install to fix connection issue
- Can this issue be replicated on a fresh installation of DietPi?

- Bug report ID | `echo $G_HW_UUID`.   --> - Reference code: e771dc04-94ef-48f4-b13d-66e98ffb33d3

#### Steps to reproduce

I guess there is 2 issues:

Issues 1: When there is cut on the wifi or a power outage ( as I'm doing some house upgrade who need power to be turn-down)
Domotictz doesn't reconnect and not working
To fix it, I tried to re-install it with: dietpi-software reinstall 140

issue 2: the re-install failed by saying there is no more space after falling a backup

root@DietPiAB:~# df -h
Filesystem           Size  Used Avail Use% Mounted on
/dev/root             59G   58G     0 100% /
devtmpfs             916M     0  916M   0% /dev
tmpfs                950M  4.0K  950M   1% /dev/shm
tmpfs                380M   39M  342M  11% /run
tmpfs                5.0M     0  5.0M   0% /run/lock
tmpfs                1.0G     0  1.0G   0% /tmp
tmpfs                 50M  8.0K   50M   1% /var/log
/dev/mmcblk0p1       253M   55M  198M  22% /boot

Joulinar, 2023-05-08:
yes, you SD card is fully used. Let's try to find out where your space is gone

```sh
du -a / | sort -n -r | head -n 20
```

mcnahum, 2023-05-08:
Thanks, Seams to be a software I tried to install to read comics who never worked ... 

So issue 2, is to re-install Domoticz (sorry for the long log):

DietPi-Software
─────────────────────────────────────────────────────
 Step: Installing Domoticz: Open source home automation platform

[  OK  ] DietPi-Software | cd /tmp/DietPi-Software
[  OK  ] DietPi-Software | Checking URL: https://releases.domoticz.com/releases/release/domoticz_linux_armv7l.tgz
[ INFO ] DietPi-Software | G_THREAD_START_0 | curl -sSfL https://releases.domoticz.com/releases/release/domoticz_linux_armv7l.tgz -o domoticz_linux_armv7l.tgz
[ INFO ] DietPi-Software | APT install libusb-0.1-4, please wait...
Reading package lists...
Building dependency tree...
Reading state information...
libusb-0.1-4 is already the newest version (2:0.1.12-32).
0 upgraded, 0 newly installed, 0 to remove and 2 not upgraded.
[  OK  ] DietPi-Software | APT install libusb-0.1-4
[  OK  ] DietPi-Software | G_THREAD: All threads finished
[FAILED] DietPi-Software | tar xf domoticz_linux_armv7l.tgz --one-top-level=domoticz

---------------------------------------------------------------------
- DietPi has encountered an error                                   -
- Please create a ticket: https://github.com/MichaIng/DietPi/issues -
- Copy and paste only the BLUE lines below into the ticket          -
---------------------------------------------------------------------
#### Details:
- Date           | Mon May  8 12:33:56 BST 2023
- DietPi version | v8.17.2 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Raspbian Lite
- Hardware       | RPi 4 Model B (aarch64) (ID=4)
- Kernel version | `Linux DietPiAppleBox 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023 aarch64 GNU/Linux`
- Distro         | bullseye (ID=6,RASPBIAN=1)
- Command        | `tar xf domoticz_linux_armv7l.tgz --one-top-level=domoticz`
- Exit code      | 2
- Software title | DietPi-Software
#### Steps to reproduce:

1. ...
2. ...
#### Expected behaviour:

- ...
#### Actual behaviour:

- ...
#### Extra details:

- ...
#### Additional logs:
```
tar: domoticz/History.txt: Cannot open: Not a directory
tar: domoticz/License.txt: Cannot open: Not a directory
tar: domoticz/domoticz.sh: Cannot open: Not a directory
tar: domoticz/server_cert.pem: Cannot open: Not a directory
tar: domoticz/updatebeta: Cannot open: Not a directory
tar: domoticz/updaterelease: Cannot open: Not a directory
tar: domoticz/www: Cannot mkdir: Not a directory
tar: domoticz/www/switch_icons.txt: Cannot open: Not a directory
tar: domoticz/www/js: Cannot mkdir: Not a directory
tar: domoticz/www/js/angular-scrollglue.js.gz: Cannot open: Not a directory
tar: domoticz/www/js/base64.min.js.gz: Cannot open: Not a directory
tar: domoticz/www/js/jquery-3.6.0.min.js.gz: Cannot open: Not a directory
tar: domoticz/www/js/jquery.ui.touch-punch.min.js.gz: Cannot open: Not a directory
tar: domoticz/www/js/date.format.js.gz: Cannot open: Not a directory
tar: domoticz/www/js/colpick.js.gz: Cannot open: Not a directory
tar: domoticz/www/js/i18next-1.8.0.min.js.gz: Cannot open: Not a directory
tar: domoticz/www/js/angularAMD.min.js.gz: Cannot open: Not a directory
tar: domoticz/www/js/jquery.wheelcolorpicker.js.gz: Cannot open: Not a directory
[... 2190 lines cut ...]
tar: domoticz/dzVents/runtime/device-adapters/ampere_3_phase_device.lua: Cannot open: Not a directory
tar: domoticz/dzVents/runtime/device-adapters/electric_usage_device.lua: Cannot open: Not a directory
tar: domoticz/dzVents/runtime/device-adapters/zone_heating_device.lua: Cannot open: Not a directory
tar: domoticz/dzVents/runtime/device-adapters/evohome_device.lua: Cannot open: Not a directory
tar: domoticz/dzVents/runtime/device-adapters/barometer_device.lua: Cannot open: Not a directory
tar: domoticz/dzVents/runtime/device-adapters/onkyo_device.lua: Cannot open: Not a directory
tar: domoticz/dzVents/runtime/device-adapters/thermostat_type_3_device.lua: Cannot open: Not a directory
tar: domoticz/dzVents/runtime/persistence.lua: Cannot open: Not a directory
tar: domoticz/dzVents/runtime/EventHelpers.lua: Cannot open: Not a directory
tar: domoticz/dzVents/runtime/Utils.lua: Cannot open: Not a directory
tar: domoticz/dzVents/runtime/dzVents.lua: Cannot open: Not a directory
tar: domoticz/dzVents/runtime/TimedCommand.lua: Cannot open: Not a directory
tar: domoticz/dzVents/runtime/JSON.lua: Cannot open: Not a directory
tar: domoticz/dzVents/.gitignore: Cannot open: Not a directory
tar: domoticz/dzVents/documentation: Cannot mkdir: Not a directory
tar: domoticz/dzVents/documentation/history.md: Cannot open: Not a directory
tar: domoticz/dzVents/documentation/pandoc.md: Cannot open: Not a directory
tar: domoticz/dzVents/documentation/README.wiki: Cannot open: Not a directory
tar: domoticz/dzVents/documentation/dzvents-smoothing.png: Cannot open: Not a directory
tar: domoticz/dzVents/documentation/history.wiki: Cannot open: Not a directory
tar: domoticz/dzVents/documentation/README.md: Cannot open: Not a directory
tar: Exiting with failure status due to previous errors
```
---------------------------------------------------------------------
[FAILED] DietPi-Software | Unable to continue, DietPi-Software will now terminate.

Joulinar, 2023-05-08:
I see. Actually, I'm able to replicate the behaviour. We would need to check our install procedure.

MichaIng, 2023-05-10:
Isn't your SD card still full? Both issues are most likely only related to this fact. So cleanup the SD card or migrate everything to a larger one, or move DietPi userdata to a USB dongle or drive, then repeat the reinstall.

Joulinar, 2023-05-10:
@MichaIng I can replicate the issue. https://github.com/MichaIng/DietPi/actions/runs/4918655815/jobs/8785273757#step:2:1037
It has nothing to do with full SD card.

MichaIng, 2023-05-10:
Ah right sorry, I just saw the PR. The problem seems to be that there is the executable named `domoticz` within the archive, and `tar` seems to somehow misinterpret the given directory name as file to extract. A bug as the used option has a clear meaning:
```console
root@micha:/tmp# tar xf domoticz_linux_armv7l.tgz --one-top-level=domoticz
tar: domoticz/History.txt: Cannot open: Not a directory
...
tar: Exiting with failure status due to previous errors
2023-05-10 21:42:46 root@micha:/tmp# l
-rwxr-xr-x 1 1000 1000 14M Feb 14 15:07 domoticz
-rw-r--r-- 1 root root 15M May 10 21:41 domoticz_linux_armv7l.tgz
root@micha:/tmp# tar xf domoticz_linux_armv7l.tgz --one-top-level=domoticz2
root@micha:/tmp# l
drwxr-xr-x 7 root root 280 May 10 21:44 domoticz2
-rw-r--r-- 1 root root 15M May 10 21:41 domoticz_linux_armv7l.tgz
```
Probably this archive previously contained a parent directory, in which case this issue does not happen, and now it contains the files right in the archives root. Good to know this behaviour 😄.

mcnahum, 2023-05-14:
Thanks, tested the solution and it's ok

closed 2023-05-14 by mcnahum