MichaIng/DietPi #6372: Radxa Zero LED permanently on

Jargendas, 2023-05-09:
### Creating a bug report/issue

#### Required Information
DietPi 8.17.2 | bulleye | 6.1.11-meson64 | Radxa Zero (arch64) (1 GB memory, V1.51)

#### Additional Information (if applicable)
- Can this issue be replicated on a fresh installation of DietPi? 
Yes

#### Steps to reproduce
Boot up DietPi on Radxa Zero.

#### Expected behaviour
The power LED next to the power USB should be showing some activity, e.g. when the SD card is active, like on the Raspberry Pi.

#### Actual behaviour
The LED is permanently on.

#### Extra details
The problem might only exist for hardware revision 1.51, as the pin of the power LED was [changed](https://wiki.radxa.com/Zero/hardware/revisions). Also, I found a [post](https://forum.radxa.com/t/led-meaning-of-radxa-zero/9047/4) explaining how to modify the device tree in Radxa's OS version to match the correct pin. However, the device tree used by DietPi seems to be different, at least I could not find the referenced line.

Jargendas, 2023-05-10:
This probably means the LED is not present in the device tree altogether, so all versions of the Zero have this problem.

MichaIng, 2023-05-10:
Just to verify:
```sh
ls -l /sys/class/leds/
```
Possibly another regression with Linux 6.1.

Jargendas, 2023-05-11:
Yeah, that's empty. Does this have to be fixed upstream in the Linux Kernel?

MichaIng, 2023-05-11:
Not sure. At least we should have a look into the Armbian kernel patches: https://github.com/armbian/build/tree/main/patch/kernel/archive/meson64-6.1

Here the upstream device tree: https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git/tree/arch/arm64/boot/dts/amlogic/meson-g12a-radxa-zero.dts?h=linux-6.1.y