MichaIng/DietPi #6379: drives regularly unmount without user interaction

TDuffinNTU, 2023-05-13:
#### Required Information
- DietPi version | `cat /boot/dietpi/.version`
```cat /boot/dietpi/.version 
G_DIETPI_VERSION_CORE=8
G_DIETPI_VERSION_SUB=16
G_DIETPI_VERSION_RC=2
G_GITBRANCH='master'
G_GITOWNER='MichaIng'
G_LIVE_PATCH_STATUS[0]='not applicable'
```

- Distro version | `echo $G_DISTRO_NAME $G_RASPBIAN`
```
echo $G_DISTRO_NAME $G_RASPBIAN
bullseye 0
```
- Kernel version | `uname -a`
```
uname -a
Linux DietPi 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023 aarch64 GNU/Linux
```
- SBC model | `echo $G_HW_MODEL_NAME` or (EG: RPi3)
RPi 4B
- Power supply used | (EG: 5V 1A RAVpower)
Included 5V power
- SD card used | (EG: SanDisk ultra)
Included sandisk SD.

#### Additional Information (if applicable)
- Software title | (EG: Nextcloud)
dietpi-drive_manager (possibly?)
- Was the software title installed freshly or updated/migrated?
No
- Can this issue be replicated on a fresh installation of DietPi?
Unlikely. This might be HW specific but mentioning here for posterity.

- Bug report ID | `echo $G_HW_UUID`
```6073ebdb-5364-493d-922d-28d37a2b78d6```

#### Steps to reproduce

1. Boot system
2. Wait

#### Expected behaviour

- Drives remain mounted to their defined locations

#### Actual behaviour

- Drives unmount, sometimes creating 'ghost' block devices until rebooted
- User_data drive becomes readonly and won't revert (even with fschk) until rebooted.

#### Extra details

- I Realise this might not be a dietpi specific issue so apologies if this isn't the right place to ask!
- Maybe a potential enhancement would be to store a few logs before/after a drive enders RW mode/auto-unmounts for bughunting?

Joulinar, 2023-05-13:
How is the drive powered? Does it have an own PSU or just connected to a USB port of your RPI4? We see such thinks if drives don't have own power unit. A downside of the RPI4 as their USB ports are not designed to power HDD or SDD. Especially on hight load situations drives could become unavailable leading to data corruption. From our side we recommend a powered USB hub or own PSU.

Once a drive become r/o, you could check kernel error messages

```
dmesg -l 0,1,2,3
```

TDuffinNTU, 2023-05-13:
Ah that could be it! I'm very careful with ensuring my setup gets uninterrupted power but I've always just used the stock power cable supplied by Pimroni. I figured two SSDs wouldn't pull _too_ much power but under load (with nightly backups from one drive to another) this could be causing it for sure.

I'll look into getting an externally-powered hub or moving to a NAS. Thanks!

MichaIng, 2023-05-13:
Even a single SSD is often too much. Note that RPi's provide an overall shared 1.2A/6W on all USB ports, and even a single SSD can peak higher.

closed 2023-10-11 by TDuffinNTU