MichaIng/DietPi #6381: [FAILED] DietPi-TimeSync | systemctl start systemd-timesyncd

GotAnAccount, 2023-05-17:
#### Details:
- Date           | Wed May 17 06:16:18 EDT 2023
- DietPi version | v8.17.2 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Raspberry Pi OS (32-bit) Lite
- Hardware       | RPi 3 Model B+ (armv7l) (ID=3)
- Kernel version | `Linux PrusaPi 5.10.103-v7+ #1529 SMP Tue Mar 8 12:21:37 GMT 2022 armv7l GNU/Linux`
- Distro         | bullseye (ID=6,RASPBIAN=1)
- Command        | `systemctl start systemd-timesyncd`
- Exit code      | 5
- Software title | DietPi-TimeSync
#### Steps to reproduce:

1. ...
2. ...
#### Expected behaviour:

- ...
#### Actual behaviour:

- ...
#### Extra details:

- ...
#### Additional logs:
```
Failed to start systemd-timesyncd.service: Unit systemd-timesyncd.service not found.
```

Joulinar, 2023-05-17:
Did you update to Debian Bullseye before? 

```
apt install systemd-timesyncd
systemctl unmask systemd-timesyncd
/boot/dietpi/func/run_ntpd 1
```

GotAnAccount, 2023-05-17:
Yes I did and I have resolved the issue issue by  simply reinstalling the
service. I just followed the instructions to send the message to Github.
It's all good now. Thanks.

GotAnAccount, 2023-05-17:
> did you followed our guide to update your system to upgrade to Bullseye?

Yes, but the last instruction set was not applied correctly I think. Also,
my octoprint module was not reinstalled correctly after the upgrade. I had
to manually reinstall it using its number (153) if memory serves me right.

Joulinar, 2023-05-17:
pl going to close the issue as reinstall of `systemd-timesyncd` is already part of our guide.

closed 2023-05-17 by Joulinar