MichaIng/DietPi #6382: RTL8811cu Not Found

actraiser30, 2023-05-17:
Hello there im Using a Raspberry Pi and trying to make a hotspot using the latest dietpi armv7
but my usb wifi adaptor with (RTL8811cu) cannot be found,probably there are not drivers

i search the internet and found this

https://github.com/fastoe/RTL8811CU_for_Raspbian

https://github.com/whitebatman2/rtl8821CU

can you include the drivers for rtl8821CU in a future update??

thanks a lot

Joulinar, 2023-05-17:
Did you tried to install drivers already?

actraiser30, 2023-05-17:
no as i saw from posts they work at least in raspbian

Joulinar, 2023-05-17:
They should work same way usually.

actraiser30, 2023-05-17:
most probably yes

but will you include them in a future update to have support for it?

MichaIng, 2023-05-18:
I have some dongles with the same chip here and are annoyed the same way that they do not work OOTB on RPi. Generally, RPi and Debian do not add 3rd party drivers into their kernel builds, so what definitely works are only the ones in the mainline Linux tree: https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git/tree/drivers/net/wireless?h=linux-6.1.y

Although, now I see in the meantime there are drivers for Realtek WiFi 5 chips: https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git/tree/drivers/net/wireless/realtek/rtw88?h=linux-6.1.y
The 8811 chip files are not explicitly present, but 8821 which may be compatible.

Those need to be enabled via kernel build config, which RPi obviously do not set:
```console
root@micha:~# ls -l /lib/modules/6.1.21-v7+/kernel/drivers/net/wireless/realtek/
total 12
drwxr-xr-x 3 root root 4096 Apr  6 12:42 rtl818x
drwxr-xr-x 2 root root 4096 Apr  6 12:42 rtl8xxxu
drwxr-xr-x 4 root root 4096 Apr  6 12:42 rtlwifi
```
We could ask at the RPi repo to have those Realtek WiFi 5 drivers added: https://github.com/raspberrypi/linux/issues
This is the needed kernel config option `CONFIG_RTW88_CORE=m`, which is indeed not present at all:
```console
root@micha:~# zgrep CONFIG_RTW88_CORE /proc/config.gz
root@micha:~#
```