MichaIng/DietPi #6383: DietPi-Update / DietPi-Services should only restart services installed by DietPi-Software (openhab)

ProtoFoo, 2023-05-18:
#### Required Information
- DietPi version | `G_DIETPI_VERSION_CORE=8
G_DIETPI_VERSION_SUB=17
G_DIETPI_VERSION_RC=2
G_GITBRANCH='master'
G_GITOWNER='MichaIng'
`
- Distro version | `bullseye 1`
- Kernel version | `Linux dietpi 6.1.21-v7+ #1642 SMP Mon Apr  3 17:20:52 BST 2023 armv7l GNU/Linux`
- SBC model | `RPi 3 Model B (armv7l)`
- Power supply used | `MeanWell HDR-30-12 12V 2A stepped down to 5V`
- SD card used | `SanDisk Extreme 64 GB`

#### Steps to reproduce

1. DietPi was installed in 2019 (running perfectly fine without any issues until now)
2. Openhab was installed manually on this DietPi installation
3. The update to DietPi with Openhab support restarted the openhab service

#### Expected behaviour
Openhab is part of a building infrastructure with access control, security system and shutters / lights management. The openhab service must never be restarted without a very good reason to achieve high availability for residents.

#### Actual behaviour
`dietpi-update` seems to blindly restart all services it knows of, without checking if they were installed by `dietpie-software`. Openhab is not in the `# Non-controlled services:` section of `dietpi-services` and for some people it might be fine to restart Openhab.

In my experience Openhab is fragile, needs manual intervention to work after a restart, and is an unmaintainable complexity nightmare. I prevent touching Openhab at all costs. As long as I leave it alone, it just works.

#### Suggestions
My workaround would be to put Openhab in `/boot/dietpi/.dietpi-services_include_exclude` for now.
A better design would be for `dietpi-services` to check `/boot/dietpi/.installed` and ignore services that were **not** installed by `dietpi-software`. I understand that this would require quite some effort as you need to map software IDs to service names somewhere.

I am interested in the opinions of others in this regard. Pi-hole for example is also treated as "never restart" as far as I understand this. An easy solution would be to apply the same rule to Openhab as it was done with Pi-hole.

#### Extra details

<details>
  <summary>Installed software:</summary>

```
# cat /boot/dietpi/.installed 
aSOFTWARE_INSTALL_STATE[17]=2
aSOFTWARE_INSTALL_STATE[69]=2
aSOFTWARE_INSTALL_STATE[81]=2
aSOFTWARE_INSTALL_STATE[84]=2
aSOFTWARE_INSTALL_STATE[87]=2
aSOFTWARE_INSTALL_STATE[89]=2
aSOFTWARE_INSTALL_STATE[93]=2
aSOFTWARE_INSTALL_STATE[103]=2
aSOFTWARE_INSTALL_STATE[104]=2
INDEX_LOGGING=-1
```

</details>

<details>
  <summary markdown="span">Output of dietpi-update:</summary>

```
 ─────────────────────────────────────────────────────
 DietPi v8.16.2 : Update available
 ─────────────────────────────────────────────────────
 - Device model : RPi 3 Model B (armv7l)
 - Uptime : up 5 weeks, 4 days, 17 minutes
 - CPU temp : 55 °C / 131 °F : Running warm, but safe
 - Freespace (RootFS) : 3.7G
 - MOTD : Check out the release notes of DietPi v8.17:
          https://dietpi.com/docs/releases/v8_17/
 ─────────────────────────────────────────────────────

 dietpi-update   : Run now to update DietPi from v8.16.2 to v8.17.2
[... 218 lines cut ...]
Building dependency tree...
Reading state information...
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
[  OK  ] DietPi-Update | APT autopurge
[  OK  ] DietPi-Update | systemctl daemon-reload
[  OK  ] DietPi-Update | Incremental patching to v8.17.2 completed
[ INFO ] DietPi-Update | Checking for new available live patches

 DietPi-Update
─────────────────────────────────────────────────────
 Phase: Completed

[ INFO ] DietPi-Update | Current version : v8.17.2
[ INFO ] DietPi-Update | Latest version  : v8.17.2
[  OK  ] DietPi-Survey | Purging survey data
[  OK  ] DietPi-Update | sync
[ SUB1 ] DietPi-Services > restart
[  OK  ] DietPi-Services | restart : php7.4-fpm
[  OK  ] DietPi-Services | restart : lighttpd
[  OK  ] DietPi-Services | restart : minidlna
[  OK  ] DietPi-Services | restart : openhab
[  OK  ] DietPi-Services | restart : cron
``` 

</details>

MichaIng, 2023-05-18:
This behaviour is actually expected. If a service must stay active during maintenance tasks (updates, software installs, backups), exclude it via `/boot/dietpi/.dietpi-services_include_exclude` or `dietpi-services` menu.

Pi-hole is a different case since the host itself may use it as DNS resolver. In this case, stopping it during updates and software installs would break network and hence make both: updates and software installs impossible.

ProtoFoo, 2023-05-27:
Nobody cares or is affected by this. Thanks for your answer @MichaIng and thank you very much for maintaining DietPi. I will use the workaround as described. Not ideal but it'll do for now.

closed 2023-05-27 by ProtoFoo