MichaIng/DietPi #6392: OpenVoiceOS

JarbasAl, 2023-05-25:
### Creating a software request

#### Formal software information
- Software title | OpenVoiceOS
- Short description | The continuation of the mycroft project by community members after[ it's collapse](https://community.mycroft.ai/t/faq-ovos-neon-and-the-future-of-the-mycroft-voice-assistant)
- Official URL (if available) | https://openvoiceos.com/
- GitHub URL (if available) | https://github.com/OpenVoiceOS/ovos-core
- (Official) install documentation | https://openvoiceos.github.io/ovos-technical-manual/

#### Are there similar/alternative software titles available with DietPi-Software?

the deprecated mycroft code base is included 

#### What makes your requested software better than the above solutions, if available?

a whole ecosystem of maintained software with a thriving community

#### How can DietPi make the installation easier or compatible, than following the install instructions or do APT installation, if available?

each service is just a regular python package, pip install + create a systemd service file for automatic launch on boot

each skill is also a regular python package

some services have alternative implementations, offering users the choice to select those instead of using default reference implementations is an option

#### Can you provide the installation steps that you would suggest DietPi-Software to do?

everything is on pypi, `pip install ovos-core[mycroft] --pre` should install all dependencies for the equivalent of a mycroft experience  (minus skills)

our automations publish the dev branch to pypi as an alpha version, `--pre` flag above is equivalent to installing from git

#### Are you willing to help maintaining the software installation, e.g. in case of needed setup changes due to updates etc.? 

I am the lead developer of OpenVoiceOS and am happy to provide any guidance needed and fix issues OVOS side

MichaIng, 2023-05-25:
Many thanks for your request, offer for guidance and the info about Mycroft AI. Sad to hear that its development has been discontinued.

I heard about OVOS but thought it would be more like a wrapping OS with additional features around the Mycroft AI core, not suitable for a multi-purpose OS like DietPi. However, I see now that OVOS is modular as well. Great that it continues the open source voice assistant development as a fork!

As it does not use `home.mycroft.ai`, is TTS is all done offline then? While I generally like that, at least I guess it requires more hardware resources and hence cannot be run too weak SBCs which were still able to run Mycroft AI with online TTS? I should probably just read the docs 🙂.

JarbasAl, 2023-05-25:
all components can be run offline or self hosted, the self hosted plugins (for STT/TTS) often include a default list of public servers. We also have a couple projects to automatically turn any plugin into a self hosted rest api

you are right that OVOS started as companion software and a reference OS, but over time we had no choice but fork basically every mycroft repo. the buildroot image where this all started still exists and aims to be a full reference OS with all OVOS components pre installed

being modular is a big part of ovos as we consider every component essentially optional or usable standalone/as a lib for other projects.  for example in a server you probably only want skills service, dont need any of the rest