MichaIng/DietPi #6404: DietPi domoticz[4816]: /opt/domoticz/domoticz: error while loading shared libraries: │ libcurl-gnutls.so.4: cannot open shared object file: No such file or directory

IgrekLg, 2023-06-01:
### Creating a bug report/issue

#### Required Information
- DietPi version | G_DIETPI_VERSION_CORE=8
G_DIETPI_VERSION_SUB=18
G_DIETPI_VERSION_RC=0
G_GITBRANCH='beta'
G_GITOWNER='MichaIng'
- Distro version | bullseye
- Kernel version | Linux DietPi 5.15.93-rockchip64 #23.02.2 SMP PREEMPT Fri Feb 17 23:48:36 UTC 2023 aarch64 GNU/Linux
- SBC model | ROCK Pi S (aarch64)
- Power supply used | 5V 2A 
- SD card used | SanDisk High Endurance Card

#### Additional Information (if applicable)
- Software title | Domoticz
- Can this issue be replicated on a fresh installation of DietPi? Yes

#### Steps to reproduce
Install the programm Domoticz from dietpi-software
Domticz not started, DietPi-Services shows "State: failed"
Status                  : Display systemd status log 

domoticz.service - Domoticz (DietPi)                                                                               │
            │      Loaded: loaded (/etc/systemd/system/domoticz.service; enabled; vendor preset: enabled)                          │
            │      Active: failed (Result: exit-code) since Thu 2023-06-01 23:39:50 MSK; 34min ago                                 │
            │     Process: 4816 ExecStart=/opt/domoticz/domoticz -f /mnt/dietpi_userdata/domoticz/domoticz.conf (code=exited,      │
            │ status=127)                                                                                                          │
            │    Main PID: 4816 (code=exited, status=127)                                                                          │
            │         CPU: 18ms                                                                                                    │
            │                                                                                                                      │
            │ Jun 01 23:39:50 DietPi systemd[1]: Started Domoticz (DietPi).                                                        │
            │ Jun 01 23:39:50 DietPi domoticz[4816]: /opt/domoticz/domoticz: error while loading shared libraries:                 │
            │ libcurl-gnutls.so.4: cannot open shared object file: No such file or directory                                       │
            │ Jun 01 23:39:50 DietPi systemd[1]: domoticz.service: Main process exited, code=exited, status=127/n/a                │
            │ Jun 01 23:39:50 DietPi systemd[1]: domoticz.service: Failed with result 'exit-code'.

MichaIng, 2023-06-01:
Many thanks for your report. Seems like the new release explicitly depends on the GnuTLS variant of libcurl instead of the one which uses OpenSSL. Does this work?
```sh
apt install libcurl3-gnutls
systemctl restart domoticz
```

MichaIng, 2023-06-01:
Ah, the reason may be the problem with libssl1.1 vs libssl3 (the OpenSSL library). It can only depend on either one or the other version, which makes it impossible to build one Domoticz binary that works on Debian up to Bullseye as well as on Debian Bookworm (and some for other distros which moved to libssl3). And Domoticz has no infrastructure to ship different binaries for different distro versions. Switching to GnuTLS might have been their solution. If so, this means we can re-enable it for Bookworm, which is great.

MichaIng, 2023-06-01:
PR up: #6402
And related tests: https://github.com/MichaIng/DietPi/actions/runs/5149388369

closed 2023-06-02 by IgrekLg