MichaIng/DietPi #6407: [FAILED] DietPi-Installer | /var/lib/dpkg/info/base-files.postinst configure

DigiDr, 2023-06-03:
### Creating a bug report/issue

#### Required Information

I tried the installer on a Rock Pi E (A Radxa device - strangely missing from OOB support), using https://dietpi.com/docs/hardware/#make-your-own-distribution.

It fails at: **[FAILED] DietPi-Installer | /var/lib/dpkg/info/base-files.postinst configure** in two places:

1. it tries to mkdir /root/.profile - but there is no root directly (easily fixed in sub shell)
2. it tries to delete /var/lock/ - but does not do this recursively and there are a couple of files in there (easily fixed in sub shell)

MichaIng, 2023-06-03:
Many thanks for reporting.
> mkdir /root/.profile

This is a file, not a directory. I checked the package, and it ships `/root` as directory natively, and `postinst` runs afterwards, copies the `.profile` (file!) into this directory. So if this did not happen in your case, then something else failed, e.g. an issue with the drive or filesystem. `/var/lock` should never be a directory but a symlink. Indeed the `base-files`' `postinst` script tries to remove it via `rmdir` if it is not a symlink yet which intentionally fails if it is not empty, as only the admin will know whether the content is still needed or not. However, if whichever base image you used ships `/var/lock` as directory, then it is a pretty broken image and I would use a different one.

So, where did you get the image from? If you used one from Radxa, I highly recommend the one from Armbian instead, which won't cause these issues: https://www.armbian.com/rockpie/

DigiDr, 2023-06-03:
Ah, that was written in a rush, and yes you are right on both points.

This was indeed done directly on a fresh install of Armbian…

steeevvo, 2023-06-10:
Had the same issue on a clean install of Armbian / OrangePi Zero 2. Created root folder and it carried on.

MichaIng, 2023-06-10:
Did you guys use Bullseye or Bookworm images?

DigiDr, 2023-06-10:
Bookworm

MichaIng, 2023-06-10:
Hmm, `/var/lock` is indeed shipped as directory by their images. I have no idea how this is even possible, as they use `debootstrap` as well, if I'm not mistaken, which definitely creates it as symlink to `/run/lock`. This is common since, I don't know, dozens of years, I don't remember this being ever different, so impossible that any common tools create this any different, other than Armbian's own tools, obviously having a bug regarding this.

However, the package ships the symlink (as well as the `/root` directory) regularly as package content, so whatever other things exist, those should be overwritten.

Just tested it and indeed if `/var/lock` is a directory, it is not overwritten, if it is a file, it is. Strange behaviour, but then the `postinst` migration step makes sense. The migration however is only done if the package is not installed before, which is the reason we actively call `/var/lib/dpkg/info/base-files.postinst configure` in our installer to force it. And this indeed fails if that directory has content:
```console
root@VM-Bookworm:/tmp# rm /var/lock
root@VM-Bookworm:/tmp# mkdir /var/lock
root@VM-Bookworm:/tmp# > /var/lock/test
root@VM-Bookworm:/tmp# /var/lib/dpkg/info/base-files.postinst configure
rmdir: failed to remove '/var/lock': Directory not empty
```
Easy to fix by checking after the package install whether it is a symlink, and if not, just removing it before calling the `postinst`.

Now about `/root`. This works well here:
```console
root@VM-Bookworm:/tmp# rm -R /root/
root@VM-Bookworm:/tmp# dpkg -i base-files_12.4_amd64.deb
(Reading database ... 15843 files and directories currently installed.)
Preparing to unpack base-files_12.4_amd64.deb ...
Unpacking base-files (12.4) over (12.4) ...
Setting up base-files (12.4) ...
root@VM-Bookworm:/tmp# ls -Al /root/
total 0
root@VM-Bookworm:/tmp# /var/lib/dpkg/info/base-files.postinst configure
root@VM-Bookworm:/tmp# ls -Al /root/
total 8.0K
-rw-r--r-- 1 root root 571 Apr 10  2021 .bashrc
-rw-r--r-- 1 root root 161 Jul  9  2019 .profile
```
I will test with an actual Armbian image.

MichaIng, 2023-06-10:
Found it:
```console
root@rockpi-e:/tmp# apt policy base-files
base-files:
  Installed: 23.05.1--bookworm-1armbian1-B9e4a
  Candidate: 23.05.1--bookworm-1armbian1-B9e4a
  Version table:
 *** 23.05.1--bookworm-1armbian1-B9e4a 100
        100 /var/lib/dpkg/status
     12.4 500
        500 http://deb.debian.org/debian bookworm/main arm64 Packages
root@rockpi-e:/tmp# apt install --reinstall base-files
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Reinstallation of base-files is not possible, it cannot be downloaded.
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
```
Armbian ships their images with an own `base-files` package, but their Bookworm repository does not contain this package (yet), so it cannot be reinstalled. Our installer even removes the related suites, to only leave the Armbian kernel/bootloader suites in place, so it wouldn't be possible to reinstall it even if they added it to their repo (as it is a downgrade due to their year-based version string). Not so trivial to solve:
```sh
echo -e '\''Package: base-files\nPin: origin *\nPin-Priority: 1000'\'' > /etc/apt/preferences.d/99dietpi-base-files
apt --reinstall install base-files
rm /etc/apt/preferences.d/99dietpi-base-files
```

MichaIng, 2023-06-10:
Both issues fixed in `dev` branch: https://github.com/MichaIng/DietPi/commit/bfcf773

closed 2023-07-01 by MichaIng