MichaIng/DietPi #6408: Deluge doesn't download/upload

shlatchz, 2023-06-03:
### Creating a bug report/issue

#### Required Information
- DietPi version | `cat /boot/dietpi/.version`

```
root@DietPi:~# cat /boot/dietpi/.version
G_DIETPI_VERSION_CORE=8
G_DIETPI_VERSION_SUB=17
G_DIETPI_VERSION_RC=2
G_GITBRANCH='master'
G_GITOWNER='MichaIng'
G_LIVE_PATCH_STATUS[0]='applied'
G_LIVE_PATCH_STATUS[1]='not applicable'
```

- Distro version | `echo $G_DISTRO_NAME $G_RASPBIAN`

```
root@DietPi:~# echo $G_DISTRO_NAME $G_RASPBIAN
bookworm
```

- Kernel version | `uname -a`

```
root@DietPi:~# uname -a
Linux DietPi 6.1.0-9-amd64 #1 SMP PREEMPT_DYNAMIC Debian 6.1.27-1 (2023-05-08) x86_64 GNU/Linux
```

- SBC model | `echo $G_HW_MODEL_NAME` or (EG: RPi3)

```
root@DietPi:~# echo $G_HW_MODEL_NAME
Native PC (x86_64)
```

- Power supply used | (EG: 5V 1A RAVpower)

:shrug:

- SD card used | (EG: SanDisk ultra)

exFAT

```
root@DietPi:~# ls -l /mnt
total 2072
drwxr-xr-x  3 root   root      4096 Apr 26 11:19 dietpi-backup
drwxrwxr-x 10 dietpi dietpi    4096 Oct 26  2018 dietpi_userdata
drwxrwxr-x  2 root   root      4096 Oct 26  2018 ftp_client
drwxrwxr-x  2 root   root      4096 Oct 26  2018 nfs_client
drwxr-xr-x  8 dietpi dietpi 2097152 Jun  3 15:40 PIHDD
drwxrwxr-x  2 root   root      4096 Oct 26  2018 samba
drwxr-xr-x  2 root   root      4096 Nov 24  2018 USB
```

#### Additional Information (if applicable)
- Software title | (EG: Nextcloud)

Deluge

```
root@DietPi:~# deluged -v
deluged 2.0.3
libtorrent: 2.0.8.0
Python: 3.11.2
OS: Linux 6.1.0-9-amd64
```

- Was the software title installed freshly or updated/migrated?
- Can this issue be replicated on a fresh installation of DietPi?

- Bug report ID | `echo $G_HW_UUID`

`3b0cd5c5-8000-40c6-ac83-14972cc6877b`

#### Steps to reproduce
#### Expected behaviour
#### Actual behaviour

Deluge client shows "Downloading" on a turrent but nothing happens.. it won't download / upload.

[image]

Tried to look at the log and got an error:

```
root@DietPi:~# journalctl -u deluged -u deluge-web
Journal file /var/log/journal/455a81b1b9734a369117844dfd7e4a58/system.journal is truncated, ignoring file.
Jun 03 15:40:57 DietPi systemd[1]: Started deluged.service - Deluge Daemon (DietPi).
Jun 03 15:40:57 DietPi systemd[1]: Started deluge-web.service - Deluge Web UI (DietPi).
Jun 03 15:40:58 DietPi deluge-web[654]: Unable to initialize gettext/locale!
Jun 03 15:40:58 DietPi deluge-web[654]: module 'gettext' has no attribute 'bind_textdomain_codeset'
Jun 03 15:40:58 DietPi deluge-web[654]: Traceback (most recent call last):
Jun 03 15:40:58 DietPi deluge-web[654]:   File "/usr/lib/python3/dist-packages/deluge/i18n/util.py", line 111, in setup_translation
Jun 03 15:40:58 DietPi deluge-web[654]:     gettext.bind_textdomain_codeset(I18N_DOMAIN, 'UTF-8')
Jun 03 15:40:58 DietPi deluge-web[654]:     ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
Jun 03 15:40:58 DietPi deluge-web[654]: AttributeError: module 'gettext' has no attribute 'bind_textdomain_codeset'
Jun 03 15:40:59 DietPi deluge-web[654]: 15:40:59 [ERROR   ][deluge.i18n.util           :146 ] Unable to initialize gettext/locale!
Jun 03 15:40:59 DietPi deluge-web[654]: 15:40:59 [ERROR   ][deluge.i18n.util           :147 ] module 'gettext' has no attribute 'bind_textdomain_codeset'
Jun 03 15:40:59 DietPi deluge-web[654]: Traceback (most recent call last):
Jun 03 15:40:59 DietPi deluge-web[654]:   File "/usr/lib/python3/dist-packages/deluge/i18n/util.py", line 111, in setup_translation
Jun 03 15:40:59 DietPi deluge-web[654]:     gettext.bind_textdomain_codeset(I18N_DOMAIN, 'UTF-8')
Jun 03 15:40:59 DietPi deluge-web[654]:     ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
Jun 03 15:40:59 DietPi deluge-web[654]: AttributeError: module 'gettext' has no attribute 'bind_textdomain_codeset'
```

#### Extra details

MichaIng, 2023-06-03:
Bug in Deluge itself due to a change in recent Python versions (3.10/3.11 => Bookworm): https://github.com/jendrikseipp/rednotebook/issues/632

Reported and fixed on Debian: https://bugs.debian.org/1031593
Due to repo freeze that short before Bookworm release, it will be merged after the Bookworm release (scheduled for June 10th). So be a little patient.

However, this is the web interface only and should not affect downloads/uploads, especially since you were still able to add torrents. You forwarded/opened port 6882? The torrent in your example does not show any peers. Did you test with a torrent which definitely does have peers which do currently seed it?

shlatchz, 2023-06-23:
@MichaIng I still receive the error when I run `journalctl -u deluged -u deluge-web`
I updated dietpi to the latest version and ran `dietpi-software reinstall 45`
Everything worked for me before.. I don't know what changed.
None of the torrents are downloading, no matter which one I pick.

shlatchz, 2023-06-23:
I don't see 6882 port

```
root@DietPi:~# ss -l | grep 6882
root@DietPi:~#
```

Joulinar, 2023-06-23:
> I don't see 6882 port

Just did a quick test on my OPi 5 (Bookworm). Ports showing up right after installation.

```
root@DietPiOPi5:~# echo $G_DISTRO_NAME
bookworm
root@DietPiOPi5:~# ss -l | grep 6882
udp   UNCONN 0      0                               192.168.0.46%eth0:6882              0.0.0.0:*
udp   UNCONN 0      0                                    127.0.0.1%lo:6882              0.0.0.0:*
tcp   LISTEN 0      5                               192.168.0.46%eth0:6882              0.0.0.0:*
tcp   LISTEN 0      5                                    127.0.0.1%lo:6882              0.0.0.0:*
root@DietPiOPi5:~#
```

MichaIng, 2023-06-24:
I just realised that the fix came with Deluge v2.1.1 while Bookworm ships v2.0.3. Since there are usually no such package version updates done within a stable Debian release, I just asked at the bug tracker whether a patch is planned to fix this bug specifically for Bookworm.

Arghh, 2025-06-05:
I also keep running into this issue. Any chance of a fix?

MichaIng, 2025-06-07:
I don't think the fix will be backported in Debian anymore, due to Trixie being released soon. I also did not get any reply when asking about it.

Maybe we can patch this manually, I mean it is just Python. Checking upstream changes, the problematic method calls have just been removed: https://github.com/deluge-torrent/deluge/commit/f331b6c
```sh
sed -i '/bind_textdomain_codeset/d' /usr/lib/python3/dist-packages/deluge/i18n/util.py
systemctl restart deluged
```
Error messages are gone. Can you test whether this solves it for you, and does not have any other side effects?

Arghh, 2025-06-11:
yes, that solved the problem 🙏

closed 2025-06-11 by MichaIng