MichaIng/DietPi #6409: Official Rock 5B Wifi M.2 card (RTL8852BE) not detected by the network config scripts

vanfanel, 2023-06-03:
### Creating a bug report/issue

#### Required Information
- DietPi version | `cat /boot/dietpi/.version`
root@DietPi:~# cat /boot/dietpi/.version
G_DIETPI_VERSION_CORE=8
G_DIETPI_VERSION_SUB=17
G_DIETPI_VERSION_RC=2
G_GITBRANCH='master'
G_GITOWNER='MichaIng'
G_LIVE_PATCH_STATUS[0]='applied'
G_LIVE_PATCH_STATUS[1]='not applicable'

- Distro version | `echo $G_DISTRO_NAME $G_RASPBIAN`
root@DietPi:~# echo $G_DISTRO_NAME $G_RASPBIAN
bullseye

- Kernel version | `uname -a`
root@DietPi:~# uname -a
Linux DietPi 5.10.110-rockchip-rk3588 #23.02.2 SMP Fri Feb 17 23:59:20 UTC 2023 aarch64 GNU/Linux

- SBC model | `echo $G_HW_MODEL_NAME` or (EG: RPi3)
root@DietPi:~# echo $G_HW_MODEL_NAME
ROCK 5B (aarch64)

- Power supply used: Official Raspberry Pi 4 USB-C power supply
- SD card used: SanDisk ultra

#### Additional Information (if applicable)
- Can this issue be replicated on a fresh installation of DietPi?
Yes

#### Steps to reproduce
1. Run dietpi-config and go to Network
2. See that the official Rock 5b adapter is not detected

#### Expected behaviour
- The official Rock 5b adapter should be detected and used

#### Actual behaviour
-- The official Rock 5b adapter is not detected

#### Extra details
The official Rock 5b adapter is supposed to be an RTL8852BE, according to this:
https://www.okdo.com/p/m-2-wireless-module-a8-for-rock-3a-rock-5b-wifi-6-bt5-2/

MichaIng, 2023-06-03:
Many thanks for your report. This does not list it?
```sh
ip l
```
Are there related kernel errors?
```sh
dmesg -l 0,1,2,3
```
We use the Armbian kernel which use the Rockchip kernel sources, AFAIK even exactly the repo from Radxa, so if the Radxa Debian images do support it, the kernel generally ships drivers. But of course the flag to have the driver module compiled may be missing, or matching firmware.

No related topic on Armbian forum: https://forum.armbian.com/forum/188-rock-5b/

vanfanel, 2023-06-03:
Nope, `ip l` doesn't list it, and I see no dmesg errors.

MichaIng, 2023-06-03:
Are there any kernel messages regarding WiFi, WLAN or an M.2 device in `dmesg`? Is there any Realtek/RTL or WiFi/WLAN/cfg80211 kernel module loaded?
```sh
lsmod
```

Just to rule that out: The card is detected when you use the Radxa Debian CLI image? https://github.com/radxa-build/rock-5b/releases

vanfanel, 2023-06-04:
There's no module related to the card detected in lsmod.
Nothing on dmesg, either.

I haven't yet tried the Radxa Debian image (will try to get a new MicroSD for trying it), but in Radxa's official Android image, the card works flawlessly.

vanfanel, 2023-06-04:
Ok, tried the Radxa Debian image (CLI) and WiFi card works fine.
I can easily connect to my WiFi using nmcli as instructed in https://wiki.radxa.com/Rock5/guide/radxa-debian
lsmod shows r8135 being loaded, etc

MichaIng, 2023-09-03:
Sorry for the late reply. Is this still an issue with the Linux 5.10.160 provided by DietPi v8.21?

Rikpat, 2024-02-28:
Hello, sorry for hijacking this thread, but it seems to not have been fixed, I have Orange Pi 5 plus and the [official wifi+bt adapter](http://www.orangepi.org/html/hardWare/computerAndMicrocontrollers/details/Orange-Pi-R6.html) is literally the same as linked one for Rock 5B. 
I'm running dietpi 9.1.1 and I can't get the adapter working, neither wifi or bluetooth.

MichaIng, 2025-05-31:
Closing this issue. Feel free to reopen if the problem persists with recent kernel.

closed 2025-05-31 by MichaIng