MichaIng/DietPi #6418: Spotify user needed: Please help me test Spotify Connect Web!

MichaIng, 2023-06-06:
Hey guys, someone out there with a Spotify account and an ARMv7 SBC to test whether Spotify Connect Web is actually still working?
```sh
dietpi-software install 141
journalctl -u spotify-connect-web
```
There was no GitHub + Spotify user involved when this software title was added, so I have nobody to ping: https://github.com/MichaIng/DietPi/issues/863

Its development stalled in 2018, the latest "release" has "alpha" in its name and the Spotify Connect library used is deprecated since at least 2018 as well and seems to not work anymore: https://github.com/Fornoth/spotify-connect-web/issues/146#issuecomment-422124995

It has basically decided that it will be removed, but as there are still some hundred reported installs, if it does still work, we may wait for the removal until about end of the year when Debian Buster support is dropped in DietPi.

There are alternatives, Raspotify and Spotifyd, depending on architecture, but both do not have a handy web interface, but do work with Spotify Connect apps.

MichaIng, 2023-06-06:
That would be great. Also RPi makes sense in case it is RPi-only.

MichaIng, 2023-06-06:
Did you reboot once, i.e. is there an audio device available?
```sh
aplay -l
speaker-test -c2
```
It seems to have issue to find the ALSA device, but "default" should actually work with the `/etc/asound.conf` where we define/override the "default" device explicitly. Probably the issues are on a different level. I mean it uses some Python 2.7 and other extremely outdated libraries shipped all with the archive. It aims to be self-contained, but probably it is not 100% and does not work on Bullseye or even not on Buster.

MichaIng, 2023-06-11:
@Joulinar ping regarding above.

However, as your error message is the same as on my NanoPi M1 with sound device definitely available, it seems to just not work on Bookworm and we should just remove it from `dietpi-software`.

Joulinar, 2023-06-11:
I was using a RPI3B+ 32bit and soud was available. Yeah I guess we would need to remove it on next release.

closed 2023-06-12 by MichaIng