MichaIng/DietPi #6419: spotifyd does not start as needs glibc 2.32 or newer, distro version is 2.31-13

nandrasg, 2023-06-06:
### Creating a bug report/issue

#### Required Information
- DietPi version | 8.18.2
- Distro version | Native PC x64
- Kernel version | 5.10.0-23-amd64
- SBC model | Native PC

MichaIng, 2023-06-06:
Many thanks for your report. You are right, the current binaries are compiled with too high glibc version requirement:
```condole
root@VM-Bullseye:~# journalctl -u spotifyd
-- Journal begins at Tue 2023-06-06 21:25:23 CEST, ends at Tue 2023-06-06 21:26:40 CEST. --
Jun 06 21:26:40 VM-Bullseye systemd[1]: Started Spotifyd (DietPi).
Jun 06 21:26:40 VM-Bullseye spotifyd[1501]: /opt/spotifyd/spotifyd: /lib/x86_64-linux-gnu/libc.so.6: version `GLIBC_2.32' not found (required by /opt/spotifyd/spotifyd)
Jun 06 21:26:40 VM-Bullseye spotifyd[1501]: /opt/spotifyd/spotifyd: /lib/x86_64-linux-gnu/libc.so.6: version `GLIBC_2.33' not found (required by /opt/spotifyd/spotifyd)
Jun 06 21:26:40 VM-Bullseye spotifyd[1501]: /opt/spotifyd/spotifyd: /lib/x86_64-linux-gnu/libc.so.6: version `GLIBC_2.34' not found (required by /opt/spotifyd/spotifyd)
Jun 06 21:26:40 VM-Bullseye systemd[1]: spotifyd.service: Main process exited, code=exited, status=1/FAILURE
Jun 06 21:26:40 VM-Bullseye systemd[1]: spotifyd.service: Failed with result 'exit-code'.
```
We take them from here: https://github.com/Spotifyd/spotifyd/releases
Same with the "default" and "slim" variants (we use the "full" one).

Joulinar, 2023-06-06:
This one sounds similar https://github.com/Spotifyd/spotifyd/issues/1147

MichaIng, 2023-06-06:
Okay, it was actually intended, true for last release already: https://github.com/Spotifyd/spotifyd/issues/1147
Last working release is v0.3.3: https://github.com/Spotifyd/spotifyd/releases/tag/v0.3.3
Try this:
```sh
cd /tmp
curl -LO 'https://github.com/Spotifyd/spotifyd/releases/download/v0.3.3/spotifyd-linux-full.tar.gz'
tar xf spotifyd-linux-full.tar.gz
rm spotifyd-linux-full.tar.gz
mv spotifyd /opt/spotifyd/spotifyd
systemctl restart spotifyd
```

Joulinar, 2023-06-06:
ok we would need to set 0.3.3 for Buster and Bullseye, right?

nandrasg, 2023-06-06:
> Okay, it was actually intended, true for last release already: [Spotifyd/spotifyd#1147](https://github.com/Spotifyd/spotifyd/issues/1147) Last working release is v0.3.3: https://github.com/Spotifyd/spotifyd/releases/tag/v0.3.3 Try this:
> 
> ```shell
> cd /tmp
> curl -LO 'https://github.com/Spotifyd/spotifyd/releases/download/v0.3.3/spotifyd-linux-full.tar.gz'
> tar xf spotifyd-linux-full.tar.gz
> rm spotifyd-linux-full.tar.gz
> mv spotifyd /opt/spotifyd/spotifyd
> systemctl restart spotifyd
> ```

Thank you very much for such a speed of light reply and fix. I have now:

root@J4105:/tmp# systemctl status spotifyd
● spotifyd.service - Spotifyd (DietPi)
     Loaded: loaded (/etc/systemd/system/spotifyd.service; enabled; vendor preset: enabled)
     Active: active (running) since Tue 2023-06-06 20:36:46 BST; 11s ago
   Main PID: 6255 (spotifyd)
      Tasks: 6 (limit: 9062)
     Memory: 2.3M
        CPU: 32ms
     CGroup: /system.slice/spotifyd.service
             └─6255 /opt/spotifyd/spotifyd --no-daemon --config-path=/mnt/dietpi_userdata/spotifyd/spotifyd.conf

MichaIng, 2023-06-06:
> ok we would need to set 0.3.3 for Buster and Bullseye, right?

Probably even an older version on Buster.

Lol I just tested it on Bookworm, and there it does not work because it depends on libssl1.1 while Bookworm ships libssl3.

At least the current version does work on ARMv7 Bullseye, only the x86 one is compiled with this high glibc requirement.

Pretty inconsistent. Let's get an overview and fix one by one where possible: https://github.com/MichaIng/DietPi/actions/runs/5192682619

Joulinar, 2023-06-06:
> Pretty inconsistent. Let's get an overview and fix one by one where possible: https://github.com/MichaIng/DietPi/actions/runs/5192682619

it's failing on our container ARMv7 Bullseye. However, it's working in RPi3B+ 32bit

MichaIng, 2023-06-06:
Very strange, the exact same binary on Buster shows that it supports glibc 2.29 as well:
```
Jun 06 20:44:05 DietPi spotifyd[2108]: /opt/spotifyd/spotifyd: /lib/x86_64-linux-gnu/libm.so.6: version `GLIBC_2.29' not found (required by /opt/spotifyd/spotifyd)
Jun 06 20:44:05 DietPi spotifyd[2108]: /opt/spotifyd/spotifyd: /lib/x86_64-linux-gnu/libc.so.6: version `GLIBC_2.32' not found (required by /opt/spotifyd/spotifyd)
Jun 06 20:44:05 DietPi spotifyd[2108]: /opt/spotifyd/spotifyd: /lib/x86_64-linux-gnu/libc.so.6: version `GLIBC_2.29' not found (required by /opt/spotifyd/spotifyd)
Jun 06 20:44:05 DietPi spotifyd[2108]: /opt/spotifyd/spotifyd: /lib/x86_64-linux-gnu/libc.so.6: version `GLIBC_2.33' not found (required by /opt/spotifyd/spotifyd)
Jun 06 20:44:05 DietPi spotifyd[2108]: /opt/spotifyd/spotifyd: /lib/x86_64-linux-gnu/libc.so.6: version `GLIBC_2.34' not found (required by /opt/spotifyd/spotifyd)
```
Buster however ships v2.28. So it does neither support libc6 from Buster, nor from Bullseye but supports one single version between them. Very strange toolchain used have such a glibc support list 😄.

On ARMv7:
```sh
Jun 06 20:47:29 DietPi spotifyd[5552]: /opt/spotifyd/spotifyd: error while loading shared libraries: libdbus-1.so.3: cannot open shared object file: No such file or directory
```
This was needed on x86 only, now on ARMv7 as well.

Joulinar, 2023-06-06:
> This was needed on x86 only, now on ARMv7 as well. It is a dependency of `wpasupplicant`, so usually installed, but not in containers.

Yes, it is for container only as `spotifyd` is working fine on my 32bit RPI

closed 2023-07-01 by MichaIng