MichaIng/DietPi #6421: Update from Armbian to DietPi is failing with 404

Gulbrath, 2023-06-10:
### Bug report regarding PREP_SYSTEM_FOR_DIETPI.sh
Following script was used to update Armbian to DietPi:
https://github.com/MichaIng/DietPi/blob/pre-v6.14/PREP_SYSTEM_FOR_DIETPI.sh

In the script line 1315 the 'Boot Logo' points to:
https://github.com/$G_GITOWNER/DietPi/raw/$G_GITBRANCH/.meta/images/dietpi-logo_boot.bmp
which gives a 404 error. To solve this I used:
https://github.com/$G_GITOWNER/DietPi/blob/pre-v6.14/.meta/images/dietpi-logo_boot.bmp

I tested it twice. Once with providing the link when error occurred and again on fresh install while editing the script before running it.
Worked flawlessly both times with the corrected link. 

#### Steps to reproduce

1. Install Armbian Bullseye Minimal
2. Do basic config to make system work (drivers, networking etc.)
3. Follow this tutorial to convert Armbian to DietPi: https://i12bretro.github.io/tutorials/0457.html

Joulinar, 2023-06-11:
Why are you using such an old outdated script?

MichaIng, 2023-06-11:
The `pre-v6.14` branch is only meant as intermediate upgrade branch: DietPi systems older than v6.14 are migrated to this branch automatically, and migrated back to master again automatically after having upgraded to the version provided by this branch (v7.9). It is not meant to be used for anything else, especially not for the PREP script, which offers master, beta and dev branch only, which are more than a year ahead of `pre-v6.14` and hence naturally incompatible.

Gulbrath, 2023-07-21:
Reporting back after my board had a power rail failure.
Thanks for the replies, much appreciated. Yes agreed Joulinar, I reviewed the code before just copy-paste some instruction.
Anyways your suggestion works perfect and thanks to MichaIng for the explanation. I'm learning more every day I use the system or reinstall from scratch.
Thanks again for your help and DietPi as a whole.
Kind regards, Gulbrath

closed 2023-07-21 by Gulbrath