MichaIng/DietPi #6422: dietpi-software | Docker: Install fails on RPi B

Nouhelgod, 2023-06-11:
#### Details:
- Date           | Sun Jun 11 13:20:08 BST 2023
- DietPi version | v8.18.2 (MichaIng/master)
- Image creator  |
- Pre-image      |
- Hardware       | RPi B+ (armv6l) (ID=1)
- Kernel version | `Linux RPi-Home 6.1.21+ #1642 Mon Apr  3 17:19:14 BST 2023 armv6l GNU/Linux`
- Distro         | bullseye (ID=6,RASPBIAN=1)
- Command        | `apt-get -y -eany update`
- Exit code      | 100
- Software title | DietPi-Software
#### Steps to reproduce:
1. dietpi-software
2. Search Software
3. "docker"
4. Check docker and docker-compose
5. Install
#### Expected behaviour:
- Docker and docker-compose installation.
#### Actual behaviour:
- Dietpi-software throws error.
```
Ign:1 https://download.docker.com/linux/ubuntu bullseye InRelease
Err:2 https://download.docker.com/linux/ubuntu bullseye Release
  404  Not Found [IP: 13.32.99.8 443]
Hit:3 https://archive.raspberrypi.org/debian bullseye InRelease
Hit:4 http://raspbian.raspberrypi.org/raspbian bullseye InRelease
Reading package lists...
E: The repository 'https://download.docker.com/linux/ubuntu bullseye Release' does not have a Release file.
```

Joulinar, 2023-06-11:
Did you tried to install Docker manually? Because there is an Ubuntu package server defined. But DietPi is based on Debian. Therefore this source needs to be removed.

Nouhelgod, 2023-06-11:
Yep. That was my fault.

closed 2023-06-11 by Nouhelgod