MichaIng/DietPi #6423: Change DNS without change dynamic IP to static

krishnasen2401, 2023-06-11:
### Creating a feature request

#### Is your feature request related to a problem? Please describe:

- ...
Some times I keep changing router when using ethernet connection some have different gateway address its always a issue to set the ip to static and back when changing in between just to use static dns
#### Describe the solution you'd like:

- ...
a Static dns option like the ones we in android, windows and Mac so that we can DNS without changing the IP as well
#### Describe alternatives you've considered:

- ...
can be changed to using cmd but it always creates problems
#### Additional context

- ...

JappeHallunken, 2023-06-12:
I think the problem here is, whenever you config your network with `dietpi-config`, your settings are written to `/etc/network/interfaces` and, if available, the DNS settings are written to `/etc/resolv.conf`.
You would need to avoid using `dietpi-config` and edit these two files directly. If you use DHCP you would only need to change your DNS settings in `/etc/resolv.conf`.

MichaIng, 2023-06-12:
I'm not sure whether this is natively possible with `ifupdown` resp. `/etc/network/interfaces`:
- If one configures an interface with DHCP, the DHCP client sets everything, the IP address as well as the DNS resolver, if provided by the DHCP server.
- If one sets an interface to "static", everything is set statically once.

Searching a bit around, this can be tested: https://askubuntu.com/a/623726
```sh
G_CONFIG_INJECT 'supersede[[:blank:]]+domain-name-servers[[:blank:]]' 'supersede domain-name-servers 1.1.1.1;' /etc/dhcp/dhclient.conf
```

Jep, that works here, just tested in a VM.