MichaIng/DietPi #6433: External Temperature Sensors

mdr92, 2023-06-15:
I currently use RPi-Monitor but it's well out of support having not been updated since 2017 by the looks of it, one great feature is the ability to use external sensors and have them plotted on the dashboard, graphs etc (along with a custom name), adding support for cheap temperature/humidty sensors like the DHT-11/DHT-22 would be great, a long with an easy way to setup and use them.

Thanks

jalsco, 2023-06-17:
Maybe I misunderstood ? But it seems already supported 😊

I use the super cheap BME280 sensor.  Which reminds me I need to buy a new one 😁 They cost under $2 .  They measure TEMP, AIR-PRESSURE and HUMIDITY.  

I attach to Rockpi 4  to the pins (same as raspi)  then cable it to outside balcony.  The only problem I get with BME280  (a known problem), caused by me using it outside/external  in the Tropics,  is that after a while there's some corrosion and the HUMIDITY sensor just reads at 100%. The other 2 continue working...  They say you can bake it in an oven at 120 C for 20 mins and fix that 🙄... but I prefer just buy a new one. 

I read the sensor every 3 mins (cron job) and store the data in a simple n SQlite db. Then I plot it to a webpage with some fairly simple Dash/Plotly  python code. I wrote it for myself  a few weeks ago, just to learn a bit about Dash/Plotly.  I even added some code for the Zambretti forecasts (which uses air-pressure and air pressure change, and optional wind direction ) .    

Happy to share the code if u are interested...  It was just a bit of a hobby project. Here is my site:  https://hot.starkchange.com/    I did not optimize appearance for mobile device , as I use mostly laptop. 

[image]

[image]

mdr92, 2023-06-17:
I posted this on the wrong git sorry, meant to put it on the DietPi-Dashboard one instead.

closed 2023-06-17 by mdr92