MichaIng/DietPi #6435: [FAILED] DietPi-Update | APT upgrade

fparodo, 2023-06-15:
#### Details:
- Date           | Thu Jun 15 17:40:01 BST 2023
- DietPi version | v8.15.2 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Raspbian Lite
- Hardware       | RPi 3 Model B+ (armv7l) (ID=3)
- Kernel version | `Linux Trinity 6.1.21-v7+ #1642 SMP Mon Apr  3 17:20:52 BST 2023 armv7l GNU/Linux`
- Distro         | bullseye (ID=6,RASPBIAN=1)
- Command        | `apt-get -y --with-new-pkgs upgrade`
- Exit code      | 100
- Software title | DietPi-Update
#### Steps to reproduce:

1. ...
2. ...
#### Expected behaviour:

- ...
#### Actual behaviour:

- ...
#### Extra details:

- ...
#### Additional logs:
```
Reading package lists...
Building dependency tree...
Reading state information...
Calculating upgrade...
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
2 not fully installed or removed.
After this operation, 0 B of additional disk space will be used.
Setting up mariadb-common (1:10.5.19-0+deb11u2) ...
update-alternatives: error: alternative path /etc/mysql/mariadb.cnf doesn't exist
dpkg: error processing package mariadb-common (--configure):
 installed mariadb-common package post-installation script subprocess returned error exit status 2
dpkg: dependency problems prevent configuration of libmariadb3:armhf:
 libmariadb3:armhf depends on mariadb-common; however:
  Package mariadb-common is not configured yet.

dpkg: error processing package libmariadb3:armhf (--configure):
 dependency problems - leaving unconfigured
Errors were encountered while processing:
 mariadb-common
 libmariadb3:armhf
E: Sub-process /usr/bin/dpkg returned an error code (1)
```
---------------------------------------------------------------------
[FAILED] DietPi-Update | Unable to continue, DietPi-Update will now terminate.

fparodo, 2023-06-15:
I uninstalled the mysql client package and managed to update

closed 2023-06-15 by fparodo

MichaIng, 2023-06-15:
It seems that `/etc/mysql/mariadb.cnf` was missing for some reason. This file is part of the package, but probably due to some error it was not installed as intended or got removed. Good that you were able to solve it.