MichaIng/DietPi #6439: Poll dietpi-wifi.txt on each boot

mdr92, 2023-06-15:
Sometimes I take my Pi to a new location, to connect to wifi at he new location I need to get an ethernet connection or a monitor/tv with keyboard/mouse which is quite a hassle.

Any chance for it to check dietpi-wifi.txt or similar on each boot? That would mean I can could just put the micro sd card in a reader and edit the text file on my phone with the wifi details for the new location, put the card back in and boot with the network connection.

It would make it much easier to move Pi's around, Thanks

JappeHallunken, 2023-06-16:
Couldn't this be done by editing/modifying the `wpa_supplicant.conf`?

mdr92, 2023-06-16:
> 

Isn't that stored on the Linux partitions so not really editable from Windows/Android and an sd card reader.

MichaIng, 2023-06-26:
I think it would be best to remove the file by default, since it may contain sensitive information on a world-readable and easy accessible storage device.

Probably we can do it like that:
- Add an additional variable to `dietpi-wifi.txt`, which indicates whether to leave that file in place or not.
- Run `/boot/dietpi/func/dietpi-wifidb 1` as part of [dietpi-preboot](https://github.com/MichaIng/DietPi/blob/master/dietpi/preboot) if `/boot/dietpi-wifi.txt` exists.

To not make it too complicated, it would run regardless whether WiFi has been actually enabled or not and it means that `/etc/wpa_supplicant/wpa_supplicant,conf` is rewritten on every boot.

On very first boot, `/boot/dietpi/func/dietpi-wifidb 1` currently runs in the dedicated [dietpi-firstboot](https://github.com/MichaIng/DietPi/blob/add1e3f/rootfs/var/lib/dietpi/services/dietpi-firstboot.bash#L263C21-L263C37) script. But I see no reason why this couldn't run in preboot, which run first. `dietpi-wifidb` does nothing more than translating `dietpi-wifi.txt` into `wpa_supplicant,conf`, so is pretty independent of any other setup steps.

LiminalD, 2024-02-21:
Hello Michal,

I've found that if I remove the micro SD card and insert it into a card reader, then edit the file located at /boot/dietpi/preboot by adding the line /boot/dietpi/func/dietpi-wifidb 1, and also create a file named dietpi-wifi.txt in the boot directory with my updated Wi-Fi credentials:

aWIFI_SSID[0]='YourSSID'
aWIFI_KEY[0]='YourPassword'

This method consistently works for me. It appears that this feature may already exist?