MichaIng/DietPi #6444: Problem with HA 2023.6.X and FFmpeg library

maury77, 2023-06-25:
### Creating a bug report/issue

#### Required Information
- DietPi version | `cat /boot/dietpi/.version`
- Distro version | `echo $G_DISTRO_NAME $G_RASPBIAN`
- Kernel version | `uname -a`
- SBC model | `echo $G_HW_MODEL_NAME` or (EG: RPi3)
- Power supply used | (EG: 5V 1A RAVpower)
- SD card used | (EG: SanDisk ultra)

#### Additional Information (if applicable)
- Software title | (EG: Nextcloud)
- Was the software title installed freshly or updated/migrated?
- Can this issue be replicated on a fresh installation of DietPi?

- Bug report ID | `echo $G_HW_UUID`

#### Steps to reproduce
I try install new version of HA 2023.6X but it have problem with ha-av-10.1.0 and FFmpeg
I try use this guide to install FFmpeg 4.4 https://pimylifeup.com/compiling-ffmpeg-raspberry-pi/

but not compile for mmal module missing

Manualy install pip install ha-av --upgrade
I try to use the 

#### Expected behaviour
ooking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple/
Requirement already satisfied: ha-av in ./.pyenv/versions/3.10.9/lib/python3.10/site-packages (10.0.0)
Collecting ha-av
  Using cached ha-av-10.1.0.tar.gz (2.5 MB)
  Installing build dependencies ... done
  Getting requirements to build wheel ... error
  error: subprocess-exited-with-error
  
  × Getting requirements to build wheel did not run successfully.
  │ exit code: 1
  ╰─> [2 lines of output]
      ****pkg-config returned flags we don't understand: -pthread -pthread -pthread
      Building PyAV against static FFmpeg libraries is not supported.****
      [end of output]
  

Now I don't know how to remove the static library, any Idea?

Joulinar, 2023-06-25:
related to https://dietpi.com/forum/t/problem-after-upgrade-ha-3-6-1-with-8-18-ha-av-library/17055

MichaIng, 2023-06-26:
I'm currently retesting, as only ARMv6, ARMv7 and RISC-V should be affected by the issue. PyPI provides `x86_64` and `aarch64` wheels for the `ha-av` module: https://pypi.org/project/ha-av/#files
EDIT: Yes it is.

If you compile the full set of FFmpeg libraries and headers, and install of of them, so that those appear below `/usr/local/` + headers in `/usr/local/include/`, it should work. Those should override the DEB package from Debian, but you can try to remove it:
```sh
apt autopurge libavdevice-dev
pip3 install -U ha-av
```
Note that while this is a good test, the module is otherwise built by the Home Assistant service internally during service start, into its isolated `pyenv`. The above command installs it system-wide, which also requires a system-wide Python instance.

I am just not keen to implement compilation of FFmpeg from source only for HA on Buster and Bullseye. Especially those architectures (32-bit and RISC-V) are usually significantly slower than x86 and aarch64, and FFmpeg, AFAIK, is huge, which makes the already very long install/compile process of HA again much longer. I think it would be much better to just upgrade to Bookworm or switch to an ARMv8 image where possible.

Hence "solution" currently is to just disable the HA install option on Buster and Bullseye ARMv6/7/RISC-V systems.

maury77, 2023-06-30:
root@sdomotica:~# apt autopurge libavdevice-dev
Lettura elenco dei pacchetti... Fatto
Generazione albero delle dipendenze... Fatto
Lettura informazioni sullo stato... Fatto   
I seguenti pacchetti saranno RIMOSSI:
  libavdevice-dev*
0 aggiornati, 0 installati, 1 da rimuovere e 0 non aggiornati.
Dopo quest'operazione, verranno liberati 718 kB di spazio su disco.
Continuare? [S/n] S    
(Lettura del database... 58558 file e directory attualmente installati.)
Rimozione di libavdevice-dev:armhf (7:4.3.6-0+deb11u1+rpt1)...

root@sdomotica:~# pip3 install -U ha-av
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting ha-av
  Downloading ha-av-10.1.0.tar.gz (2.5 MB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 2.5/2.5 MB 2.5 MB/s eta 0:00:00
  Installing build dependencies ... done
  Getting requirements to build wheel ... error
  error: subprocess-exited-with-error
  
  × Getting requirements to build wheel did not run successfully.
  │ exit code: 1
  ╰─> [2 lines of output]
      pkg-config returned flags we don't understand: -pthread -pthread -pthread
      Building PyAV against static FFmpeg libraries is not supported.
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
error: subprocess-exited-with-error

× Getting requirements to build wheel did not run successfully.
│ exit code: 1
╰─> See above for output.

note: This error originates from a subprocess, and is likely not a problem with pip.

MichaIng, 2023-06-30:
And your own FFmpeg build is definitely installed, along with the development headers in `/usr/local/include/`?

maury77, 2023-06-30:
root@sdomotica:~# ls /usr/local/include/
aom  dav1d  fdk-aac  kvazaar.h	libavutil  node  openssl  vpx

MichaIng, 2023-07-01:
@maury77 
This looks like only a subset of the required libraries have been installed. I guess the error message would be clearer about it when you remove all FFmpeg APT packages.

I could imagine that there are some build or install flags needed to have really all of them compiled/installed from source.

lrprawira, 2023-07-10:
> related to https://dietpi.com/forum/t/problem-after-upgrade-ha-3-6-1-with-8-18-ha-av-library/17055

Hi, I am having the exact same error to this forum post and I am on HA 2023.7.1, DietPi 8.19.1, updated to bookworm, RPi2B, and homeassistant is already on pyenv Python 3.11.4.  

Is there no other options than having to compile ffmpeg by myself?

MichaIng, 2023-07-10:
After upgrading to Bookworm, this should not be an issue anymore. Please first try to solve it with a simple service restart (`systemctl restart home-assistant`). If this does not help, verify that you see the exact same error message as posted in OP (`journalctl -u home-assistant`), otherwise please open a new issue with out error message.

It is not impossible that this happens as well on Bookworm when using an RPi, due to the recent RPi repo update after which the FFmpeg packages from the RPi repo have higher priority than those from Debian/Raspbian, even that they are Bullseye packages, i.e. with the older FFmpeg version which causes this issue: https://dietpi.com/forum/t/the-following-packages-have-been-kept-back-ffmpeg/17234

lrprawira, 2023-07-10:
@MichaIng  

My error is the same as this [forum post](https://dietpi.com/forum/t/problem-after-upgrade-ha-3-6-1-with-8-18-ha-av-library/17055).  

Thank you for the link. The older ffmpeg problem is probably the issue. I applied the APT Pinning and ran `apt install ffmpeg=7:5.1.3-1` to force "downgrade" from 8:4.3.6-0+deb11u1+rpt2. Now I have to restart homeassistant again and have it re-run the ha-av install which takes a while.

Nekmo, 2023-08-20:
> I'm currently retesting, as only ARMv6, ARMv7 and RISC-V should be affected by the issue. PyPI provides `x86_64` and `aarch64` wheels for the `ha-av` module: https://pypi.org/project/ha-av/#files EDIT: Yes it is.
> 
> If you compile the full set of FFmpeg libraries and headers, and install of of them, so that those appear below `/usr/local/` + headers in `/usr/local/include/`, it should work. Those should override the DEB package from Debian, but you can try to remove it:
> 
> ```shell
> apt autopurge libavdevice-dev
> pip3 install -U ha-av
> ```
> 
> Note that while this is a good test, the module is otherwise built by the Home Assistant service internally during service start, into its isolated `pyenv`. The above command installs it system-wide, which also requires a system-wide Python instance.
> 
> I am just not keen to implement compilation of FFmpeg from source only for HA on Buster and Bullseye. Especially those architectures (32-bit and RISC-V) are usually significantly slower than x86 and aarch64, and FFmpeg, AFAIK, is huge, which makes the already very long install/compile process of HA again much longer. I think it would be much better to just upgrade to Bookworm or switch to an ARMv8 image where possible.
> 
> Hence "solution" currently is to just disable the HA install option on Buster and Bullseye ARMv6/7/RISC-V systems.

I have run the command without reading the apt command message. Now my system is destroyed. Now I am trying to recover the system.

MichaIng, 2023-08-20:
So you are on a Buster or Bullseye system? I really recommend to upgrade to Bookworm and have FFmpeg 5 and all its libraries/headers installed from Debian repo. This is now done automatically, fixed with DietPi v8.20.

And what do you mean by "destroyed"? Did `ha-av` not compile successfully with the manually compiled FFmpeg? Above indeed I do not see a case where this worked, not sure why, probably the default build is missing some needed library or headers. Hence the suggestion to just upgrade to Bookworm.

Joulinar, 2023-08-22:
@maury77 I created a new issue for your as your problem is not related to HA directly

Joulinar, 2023-08-22:
As this has nothing to do with HA, open a new issue and share the entire error message.

maury77, 2023-08-22:
> As this has nothing to do with HA, open a new issue and share the entire error message.

I solved with permit downgrade

But sorry ma after upgrade diepi , upgrade raspiabn, but I have the same problem

Unable to install package ha-av==10.1.1: error: subprocess-exited-with-error × Getting requirements to build wheel did not run successfully. │ exit code: 1 ╰─> [2 lines of output] pkg-config returned flags we don't understand: -pthread Building PyAV against static FFmpeg libraries is not supported. [end of output] note: This error originates from a subprocess, and is likely not a problem with pip. error: subprocess-exited-with-error × Getting requirements to build wheel did not run successfully. │ exit code: 1 ╰─> See above for output. note: This error originates from a subprocess, and is likely not a problem with pip.

Joulinar, 2023-08-22:
did you reinstall `FFmpeg` as well as HA?

maury77, 2023-08-22:
Yes

MichaIng, 2023-08-22:
`ffmpeg` is not required, only the development headers `libavdevice-dev` and its dependencies.

Did you remove all remains of the self-compiled FFmpeg? I.e. is at least `/usr/local/include/` empty?

maury77, 2023-08-22:
> `ffmpeg` is not required, only the development headers `libavdevice-dev` and its dependencies.
> 
> Did you remove all remains of the self-compiled FFmpeg? I.e. is at least `/usr/local/include/` empty?

root@sdomotica:/usr/local/include# ls -al
totale 16
drwxr-xr-x  4 pi    1001 4096  1 lug 14.11 .
drwxrwsr-x 12 root staff 4096  1 feb  2022 ..
drwxr-xr-x  6 pi    1001 4096  1 feb  2022 node
drwxr-xr-x  2 root root  4096 26 set  2020 openssl
root@sdomotica:/usr/local/include# 

how to remove self compile libraries?

maury77, 2023-08-23:
I SOLVED !! I delete all video library from /usr/local/lib e sub directories arm-linux-gnueabihf e pkgconfig

and Now 

homeassistant@sdomotica:~$ pip install ha-av --upgrade
Collecting ha-av
  Downloading ha-av-10.1.1.tar.gz (3.6 MB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 3.6/3.6 MB 5.3 MB/s eta 0:00:00
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
  Preparing metadata (pyproject.toml) ... done
Building wheels for collected packages: ha-av
  Building wheel for ha-av (pyproject.toml) ... done
  Created wheel for ha-av: filename=ha_av-10.1.1-cp311-cp311-linux_armv7l.whl size=10079153 sha256=07528b2bc6a4dfbb4913c3ac7b6c650f7004a6c2833cb9c7dc150608203ba29b
  Stored in directory: /tmp/pip-ephem-wheel-cache-31mrf7oq/wheels/aa/6e/1b/d0272abbfc216dcc4e92c39b43af3beccbcf087c7038cfb432
Successfully built ha-av
Installing collected packages: ha-av
Successfully installed ha-av-10.1.1

MichaIng, 2023-09-01:
Great, so as I expected some conflict with the manually compiled libs. Surely some flag was missing to compile with the features or headers needed for `ha-av`.

MichaIng, 2023-09-21:
Okay, this won't be solved by HA. In the original PyAV lib there is even a PR open for FFmpeg 6 support which breaks FFmpeg 5, while theoretically preserving the particular needed code for FFmpeg 4.3 support 😄. So I think we need to live with the fact that HA cannot be easily installed on Bullseye anymore. Given that Bookworm is stable and we have a good upgrade script, I think it is acceptable.

closed 2023-09-21 by MichaIng