MichaIng/DietPi #6451: Ability to select between 5GHZ/2.4GHZ WiFi Band

actraiser30, 2023-06-30:
Hello there

I have a Raspberry Pi 4 and installed the WIFI Hotspot Software

but upon i go in the dietpi-config in the network adapter settings i cannot configure the WiFI Hotspot to work in 5GHZ mode

only 2.4GHZ is possible

Can you add this to a future update so a toggle between 5GHZ/2.4GHZ to exists where supported by the onboard WiFi??

thanks

symbios24, 2023-07-02:
At least can you tell me how to configure the hotpot manual to work at 5ghz and not at 2.4ghz ??

symbios24, 2023-07-05:
Anyway you are busy probably as a suggestion i say if its not possible to add a 2.4/5ghz band selection in wifi hotspot what abour add 2 installation scripts for wifi hotspot one to install hotspor in 2.4ghz band and one ro install it in 5ghz band. Thank you

MichaIng, 2023-08-10:
In `/etc/hostapd/hostapd.conf` one must set `hw_mode=a` to enable 5 GHz support. But it seems to be either or, not both. The `channel` then also needs to be adjusted. The numbers for WiFi 5 start at 30 or so? However `channel=0` should always work, as it then selects the least crowded one automatically.

actraiser30, 2023-08-11:
Is it possible to have two installation scripts for wifi hotspot one for
2.4ghz and the other for 5ghz ?? To make it easier for people?? Or just in
The Install of wifi hotspot to prompt you what mode to use 2.4 or 5ghz ??
Thanks

MichaIng, 2023-08-11:
> Is it possible to have two installation scripts for wifi hotspot one for 2.4ghz and the other for 5ghz ??

Not sure what your mean with two scripts? Like for the other hotspot related choices, I'll add a `dietpi.txt` setting to choose between 2.4 GHz and 5 GHz and in the same turn to enable WiFi 4 and WiFi 5 support (WiFi 4 is implied/needed for 5 GHz). If I find the time, I'll also add interactive input boxes to the install process.

MichaIng, 2023-08-20:
This has been added for next DietPi release: #6558
Beta starts within the next hours.

closed 2023-08-26 by MichaIng