MichaIng/DietPi #6452: DietPi has encountered an error

fabiodantas, 2023-07-01:
#### Details:
- Date           | Fri Jun 30 20:38:31 -03 2023
- Bug report     | 563b2cae-f271-4271-81d3-67c6f7372106
- DietPi version | v8.18.2 (MichaIng/master)
- Image creator  | 
- Pre-image      | 
- Hardware       | RPi 3 Model B (armv7l) (ID=3)
- Kernel version | `Linux DietPi 6.1.21-v7+ #1642 SMP Mon Apr  3 17:20:52 BST 2023 armv7l GNU/Linux`
- Distro         | bullseye (ID=6,RASPBIAN=0)
- Command        | `phpenmod redis`
- Exit code      | 2
- Software title | DietPi-Software
#### Steps to reproduce:

1. ...
2. ...
#### Expected behaviour:

- ...
#### Actual behaviour:

- ...
#### Extra details:

- ...
#### Additional logs:
```
/usr/sbin/phpenmod: 31: .: cannot open /usr/lib/php/php-helper: No such file

Joulinar, 2023-07-01:
which application you are trying to install?

fabiodantas, 2023-07-01:
When I try to install nextcloud, but I have more apps like plex, transmission and other apps with web interface.

Obter o Outlook para Android<https://aka.ms/AAb9ysg>

Joulinar, 2023-07-02:
can you check following

```
ls -la /usr/lib/php/
```

closed 2023-09-03 by MichaIng