MichaIng/DietPi #6453: Dietpi-VPN Not Connecting

paeore, 2023-07-01:
### Creating a bug report/issue

#### Required Information
- DietPi version | `v8.18.2`
- Distro version | `buster 1`
- Kernel version | `Linux DietPi 5.10.103-v7l+ #1529 SMP Tue Mar 8 12:24:00 GMT 2022 armv7l GNU/Linux`
- SBC model | `RPi 4 Model B (armv7l)`
- Power supply used | `5V 1A `
- SD card used | `120gb Kingston SSD`

#### Additional Information (if applicable)
- Software title | `DietPi-VPN`

#### Steps to reproduce
1. Start/Restart VPN connection

#### Expected behaviour
- VPN should connect

#### Actual behaviour
-  Connection times out after 20 attempts

#### Extra details
```
Connection failed/timeout: uk1842.nordvpn.com.udp.ovpn
Please verify account details are correct.
If problems persist, please check the service status: "systemctl -l status dietpi-vpn"
```
After running `systemctl -l status dietpi-vpn` the output was:
```
● dietpi-vpn.service - VPN Client (DietPi)
   Loaded: loaded (/etc/systemd/system/dietpi-vpn.service; enabled; vendor preset: enabled)
   Active: inactive (dead) since Sat 2023-07-01 14:17:20 BST; 18min ago
  Process: 603 ExecStart=/usr/sbin/openvpn --suppress-timestamps --nobind --config /etc/openvpn/client.ovpn (code=exited, status=0/SUCCESS)
 Main PID: 603 (code=exited, status=0/SUCCESS)
   Status: "Pre-connection initialization successful"
```

But still the VPN remains disconnected.

I tried Updating the the `VPN Server List` and connecting to a number of different servers from different countries trying both the UDP and TCP protocals, but still the VPN remains disconnected.

Tried running `sudo openvpn /etc/openvpn/client.ovpn`  that @MichaIng advised in [this post](https://dietpi.com/forum/t/dietpi-vpn-blocking-443/13348/13), with the output being:

```
sudo openvpn /etc/openvpn/client.ovpn
Sat Jul  1 14:38:23 2023 OpenVPN 2.4.7 arm-unknown-linux-gnueabihf [SSL (OpenSSL)] [LZO] [LZ4] [EPOLL] [PKCS11] [MH/PKTINFO] [AEAD] built on Apr 28 2021
Sat Jul  1 14:38:23 2023 library versions: OpenSSL 1.1.1n  15 Mar 2022, LZO 2.10
Sat Jul  1 14:38:23 2023 WARNING: --ping should normally be used with --ping-restart or --ping-exit
Sat Jul  1 14:38:23 2023 NOTE: the current --script-security setting may allow this configuration to call user-defined scripts
Sat Jul  1 14:38:23 2023 Outgoing Control Channel Authentication: Using 512 bit message hash 'SHA512' for HMAC authentication
Sat Jul  1 14:38:23 2023 Incoming Control Channel Authentication: Using 512 bit message hash 'SHA512' for HMAC authentication
Sat Jul  1 14:38:23 2023 TCP/UDP: Preserving recently used remote address: [AF_INET]185.169.255.33:1194
Sat Jul  1 14:38:23 2023 Socket Buffers: R=[180224->180224] S=[180224->180224]
Sat Jul  1 14:38:23 2023 UDP link local: (not bound)
Sat Jul  1 14:38:23 2023 UDP link remote: [AF_INET]185.169.255.33:1194
Sat Jul  1 14:38:23 2023 TLS: Initial packet from [AF_INET]185.169.255.33:1194, sid=8bd1cf44 a1eaf473
Sat Jul  1 14:38:23 2023 WARNING: this configuration may cache passwords in memory -- use the auth-nocache option to prevent this
Sat Jul  1 14:38:23 2023 VERIFY OK: depth=2, C=PA, O=NordVPN, CN=NordVPN Root CA
Sat Jul  1 14:38:23 2023 VERIFY OK: depth=1, O=NordVPN, CN=NordVPN CA8
Sat Jul  1 14:38:23 2023 VERIFY KU OK
Sat Jul  1 14:38:23 2023 Validating certificate extended key usage
Sat Jul  1 14:38:23 2023 ++ Certificate has EKU (str) TLS Web Server Authentication, expects TLS Web Server Authentication
Sat Jul  1 14:38:23 2023 VERIFY EKU OK
Sat Jul  1 14:38:23 2023 VERIFY X509NAME OK: CN=uk1842.nordvpn.com
Sat Jul  1 14:38:23 2023 VERIFY OK: depth=0, CN=uk1842.nordvpn.com
Sat Jul  1 14:38:23 2023 Control Channel: TLSv1.3, cipher TLSv1.3 TLS_AES_256_GCM_SHA384, 4096 bit RSA
Sat Jul  1 14:38:23 2023 [uk1842.nordvpn.com] Peer Connection Initiated with [AF_INET]185.169.255.33:1194
Sat Jul  1 14:38:24 2023 SENT CONTROL [uk1842.nordvpn.com]: 'PUSH_REQUEST' (status=1)
Sat Jul  1 14:38:29 2023 SENT CONTROL [uk1842.nordvpn.com]: 'PUSH_REQUEST' (status=1)
Sat Jul  1 14:38:29 2023 AUTH: Received control message: AUTH_FAILED
Sat Jul  1 14:38:29 2023 SIGTERM[soft,auth-failure] received, process exiting
```

The last line states `SIGTERM[soft,auth-failure] received, process exiting`. Hadn't changed my Nord account password since I created the account but I went ahead and reset my password thinking it might fix the connection issue. Rebooted the Pi and chnaged my password via `sudo dietpi-vpn` but when applying the chnages I still am unable to connect.

Let me know if you need any further information.
Paeore

salemdar, 2023-07-03:
Apparently they changed their manual auth method to service credentials.

>  Due to the recent change in the authentication process, using your email and password for the manual connection method will no longer work.
> 
> You will need to use the service credentials instead.
> 
> You can find your service credentials by following these steps:
> 
>     Please log in to your Nord Account by following this link: https://my.nordaccount.com/dashboard/nordvpn/
> 
>     Click on the NordVPN tab on the left panel which is under the Services tab.
> 
>     Scroll down and locate the Manual Setup tab, then click on Set up NordVPN manually:
> 
>     Verify your email by entering the one-time code you will receive in your registered email inbox. If you are unable to find the email, please make sure to also check your spam/junk folder.
> 
>     Copy your service credentials by using the buttons on the right.

[Source](https://www.reddit.com/r/nordvpn/comments/14l7r7k/anyone_else_unable_to_login/)

Joulinar, 2023-07-04:
@MichaIng maybe we would need to give a hint about the new login method within our configuration dialogue?

paeore, 2023-07-08:
Thank you @salemdar all working now after your helpful post.

MichaIng, 2023-09-03:
Good point. Let's add the hint + link to the input box.

MichaIng, 2026-10-26:
Done in #8399. The article at https://support.nordvpn.com/hc/en-us/articles/19685514639633 is a good source about where to find/create service credentials.