MichaIng/DietPi #6458: New CM4 compatible board from Waveshare!

DrCWO, 2023-07-03:
Hello,
Waveshare sent me this promotion today with a quite unexpensive new CM4 compatible board.
https://www.waveshare.com/core3566.htm?___store=default

Any idea if you will support it with Dietpi in future?

Best DrCWO

MichaIng, 2023-07-03:
Ah, I misunderstood, not a baseboard for the CM4 but a module for CM4 baseboards 😄.

Hmm, they offer dedicated images with HDMI output or DSI output enabled. The SoC has full mainline kernel support, but there is not device tree yet: https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git/tree/arch/arm64/boot/dts/rockchip
So I guess at least for the DSI display and probably some other hardware features, the vendor kernel would be required. For U-Boot it is most likely the same.

Could you find any kernel and bootloader sources for this module?

DrCWO, 2023-07-04:
I did not yet find the time to get into it. It seems like an interesting alternative for the Raspberry CM4 module especially regarding the price tag!
I saw that they offer Linux for it but I am not willing to use a Chinese Linux. Nobody knows what will be hidden in it 👎 
As I am travelling the rest of the week I can't help anyway at the moment...

Akkiesoft, 2023-08-11:
I also have Core3566.
I'm looking for a third party distribution that works with Core3566.
If that's DietPi, I'm glad.

FYI. The SDK can be downloaded here. This includes device tree, u-boot.
https://wiki.luckfox.com/Core3566/Datasheets

MichaIng, 2023-09-01:
Where can I find the source code for the kernel and U-Boot builds of these images? Probably I am blind 😄.

Akkiesoft, 2023-09-01:
In fact, it's a pain in the neck to get them out... 😅 

The work required an Ubuntu environment(I used Ubuntu Server 22.04).

1. Open https://wiki.luckfox.com/Core3566/Datasheets
2. Select "virtual machine file and Core3566-SDK"
3. Download "LUCKFOX_LINUX419_SDK_RELEASE_20230626.tar.gz"
    * or: Download the vbox image and extract it according to the following page: https://wiki.luckfox.com/Core3566/Core3566-SDK
4. Extract it according to the following page: https://wiki.luckfox.com/Core3566/Core3566-SDK#5-compiling-the-sdk
    * If you do not use vbox images, you will need to obtain a repo tool that is executable according to your Ubuntu environment.

When the repo command succeeds, the following file is created.

```
$ find . | grep dts | grep core3566
./kernel/arch/arm64/boot/dts/rockchip/core3566-rk3566.dtsi
./kernel/arch/arm64/boot/dts/rockchip/core3566-mipi-camera-imx219-cam0.dtsi
./kernel/arch/arm64/boot/dts/rockchip/core3566-mipi-touchscreen-800-480.dtsi
./kernel/arch/arm64/boot/dts/rockchip/core3566-rk3568.dtsi
./kernel/arch/arm64/boot/dts/rockchip/core3566-rockchip-pinconf.dtsi
./kernel/arch/arm64/boot/dts/rockchip/core3566-mipi-lp4x-v1-linux.dts
./kernel/arch/arm64/boot/dts/rockchip/core3566-rk3568-dram-default-timing.dtsi
./kernel/arch/arm64/boot/dts/rockchip/core3566-rk3568-pinctrl.dtsi
./kernel/arch/arm64/boot/dts/rockchip/core3566-luckfox-rk3566.dtsi
./kernel/arch/arm64/boot/dts/rockchip/core3566-base.dtsi
./kernel/arch/arm64/boot/dts/rockchip/core3566-hdmi-lp4x-v1-linux.dts
./kernel/arch/arm64/boot/dts/rockchip/core3566-mipi-camera-imx219-cam1.dtsi
./kernel/arch/arm64/boot/dts/rockchip/core3566-hdmi-screen.dtsi
./u-boot/arch/arm/dts/luckfox-core3566.dts
```

MichaIng, 2023-09-01:
But those are only the device tree sources. What about the kernel itself?

Is there no online repository with the source code for the entire kernel laid out transparently? This VM image extraction hassle is kinda distributor-hostile. Given the likely small amount of users, I'm not gonna invest my limited time into this. There are more open image requests where implementation is easier and with more potential users 🤔.

Akkiesoft, 2023-09-01:
Unfortunately, there does not appear to be an online repository. And I very much understand that your time is limited.

Core3566 has very poor support and probably very few buyers, so I don't think it is worth supporting.

On the other hand, I would be very grateful if you could provide support for us who have purchased such a helpless board. Otherwise, I will either be patient and use the official OS, or throw the board in the trash then buy a Raspberry Pi...

MichaIng, 2023-09-03:
Or ask the vendor for a public/transparent kernel source repository as a  customer 😉.

b-durgaharikiran, 2025-05-03:
I should have seen these posts before buying one. Dumb ass me. wasted Rs 6k on this dumb compute module

Akkiesoft, 2025-05-03:
@b-durgaharikiran You can use Armbian if you can buildung image yourself.