MichaIng/DietPi #6460: Unknown module, dietpi-https, error while using dietpi-letsencrypt

midniteca, 2023-07-05:
#### Details:
- Date           | Tue Jul  4 23:19:33 EDT 2023
- Bug report     | 0f471e48-e51d-4dd5-9630-d720c5f5a98f
- DietPi version | v8.19.1 (MichaIng/master)
- Image creator  |
- Pre-image      |
- Hardware       | RPi 3 Model B (aarch64) (ID=3)
- Kernel version | `Linux DietPi 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023 aarch64 GNU/Linux`
- Distro         | bullseye (ID=6,RASPBIAN=0)
- Command        | `lighty-enable-mod dietpi-https`
- Exit code      | 2
- Software title | DietPi-LetsEncrypt
#### Steps to reproduce:

1. ... Run  DietPi-LetsEncrypt
2. ... Entered domain we wanted an SSL cert for
3. ... Other options left as default with the exception for "Redirect" was set to On.
#### Expected behaviour:

- Cert is obtained successfully, http to https redirect is configured in lighttpd and no errors are returned
#### Actual behaviour:

- Letsencrypt reports that the cert is successfully obtained.
- Getting an error saying "unknown module: dietpi-https", assuming this is dietpi trying to setup https in lighttpd
- No signs of https being setup within /etc/lighttpd/lighttpd.conf
#### Extra details:

Webserver is the default setup done by installing the "LLSP: Lighttpd + SQLite  + PHP" option via dietpi-software
root@DietPi:/# pwd
/
root@DietPi:/# find | fgrep -i dietpi-https | wc -l
0

#### Additional logs:
```
Ignoring unknown module: dietpi-https
Run "service lighttpd force-reload" to enable changes
```

Joulinar, 2023-07-05:
This is known https://github.com/MichaIng/DietPi/issues/6455#issuecomment-1616775923

However there should be a live patch out to fix this alteady

midniteca, 2023-07-05:
That is correct, this isn't a new installation of dietpi. I'm just trying to create a certificate via the dietpi-letsencrypt script.

Just to be sure, I did try running dietpi-update to see if the live patch mentioned in #6455 exists but dietpi-update doesn't offer any live patches for me to apply.

Also tried manually running the command mentioned in https://github.com/MichaIng/DietPi/issues/6455#issuecomment-1616775923 but that modifies dietpi-software where my error looks to be occurring from dietpi-letsencrypt, I did try ordering another certificate after running that command and have encountered the same error as before:

`[FAILED] DietPi-LetsEncrypt | lighty-enable-mod dietpi-https`
`Ignoring unknown module: dietpi-https
Run "service lighttpd force-reload" to enable changes
`

Joulinar, 2023-07-05:
ok basically same issue cause by the same typo.

https://github.com/MichaIng/DietPi/blame/d84586b8afa5404d2b309ed71647926fd960da83/dietpi/dietpi-letsencrypt#L115

Workaround would be similar

```
sed -i '/setenv.*dietpi-https/s/dietpi-https/setenv/' /boot/dietpi/dietpi-letsencrypt
```

midniteca, 2023-07-05:
Tested that workaround and dietpi-letsencrypt looks to be working for me now. Thanks @Joulinar 

For record:

Patch applied:

`root@DietPi:/boot/dietpi# cp dietpi-letsencrypt dietpi-letsencrypt.bak
root@DietPi:/boot/dietpi# sed -i '/setenv.*dietpi-https/s/dietpi-https/setenv/' /boot/dietpi/dietpi-letsencrypt
root@DietPi:/boot/dietpi# diff dietpi-letsencrypt dietpi-letsencrypt.bak
115c115
<                               [[ -f '/etc/lighttpd/conf-enabled/05-setenv.conf' ]] || G_EXEC lighty-enable-mod setenv
---
>                               [[ -f '/etc/lighttpd/conf-enabled/05-setenv.conf' ]] || G_EXEC lighty-enable-mod dietpi-https`

Sniplet of the dietpi-letencrypt output after patch applied:

`[  OK  ] DietPi-LetsEncrypt | lighty-enable-mod setenv
[ INFO ] DietPi-LetsEncrypt | Checking for required APT packages: lighttpd-mod-openssl
[ INFO ] DietPi-LetsEncrypt | APT update, please wait...
Hit:1 https://deb.debian.org/debian bullseye InRelease
Hit:2 https://deb.debian.org/debian bullseye-updates InRelease
Hit:3 https://deb.debian.org/debian-security bullseye-security InRelease
Hit:4 https://deb.debian.org/debian bullseye-backports InRelease
Hit:5 https://archive.raspberrypi.org/debian bullseye InRelease
Reading package lists...
[  OK  ] DietPi-LetsEncrypt | APT update
[ INFO ] DietPi-LetsEncrypt | APT install lighttpd-mod-openssl, please wait...
Reading package lists...
Building dependency tree...
Reading state information...
The following NEW packages will be installed:
  lighttpd-mod-openssl
0 upgraded, 1 newly installed, 0 to remove and 5 not upgraded.
Need to get 17.8 kB of archives.
After this operation, 69.6 kB of additional disk space will be used.
Get:1 https://deb.debian.org/debian bullseye/main arm64 lighttpd-mod-openssl arm64 1.4.59-1+deb11u2 [17.8 kB]
debconf: delaying package configuration, since apt-utils is not installed
Fetched 17.8 kB in 0s (79.6 kB/s)
                                 Selecting previously unselected package lighttpd-mod-openssl.
(Reading database ... 24103 files and directories currently installed.)
Preparing to unpack .../lighttpd-mod-openssl_1.4.59-1+deb11u2_arm64.deb ...
Unpacking lighttpd-mod-openssl (1.4.59-1+deb11u2) ...
Setting up lighttpd-mod-openssl (1.4.59-1+deb11u2) ...
[  OK  ] DietPi-LetsEncrypt | APT install lighttpd-mod-openssl
[  OK  ] DietPi-LetsEncrypt | sed -i s/, "Options" => "-SessionTicket"// /etc/lighttpd/conf-available/50-dietpi-https.conf
[  OK  ] DietPi-LetsEncrypt | lighty-enable-mod dietpi-https
[  OK  ] DietPi-LetsEncrypt | Desired setting in /etc/lighttpd/lighttpd.conf was already set:   "mod_redirect",
[  OK  ] DietPi-LetsEncrypt | lighty-enable-mod dietpi-https_redirect
[ INFO ] DietPi-LetsEncrypt | Applying HTTPS domain to known web application configs
[  OK  ] DietPi-LetsEncrypt | systemctl restart lighttpd
[  OK  ] DietPi-LetsEncrypt | eval echo -e '#!/bin/dash\nsystemctl reload lighttpd' > /etc/letsencrypt/renewal-hooks/deploy/dietpi-lighttpd.sh
[  OK  ] DietPi-LetsEncrypt | chmod +x /etc/letsencrypt/renewal-hooks/deploy/dietpi-lighttpd.sh

Press any key to return to the DietPi-LetsEncrypt menu ...`

Joulinar, 2023-07-05:
ok great

@MichaIng 
I guess we need another Live Patch 🤣

MichaIng, 2023-07-06:
Jep, same copy&paste error. Thanks for reporting, will write a live patch for this.

MichaIng, 2023-07-06:
Live patch merged: #6462

closed 2023-07-06 by MichaIng