MichaIng/DietPi #6461: Can't install ffmpeg on fresh installation first boot

ramoneiggy, 2023-07-06:
### Creating a bug report/issue

#### Required Information
- DietPi version | 8.19.1
- SBC model | Rpi4

#### Additional Information (if applicable)
- Software title | ffmpeg 
- Was the software title installed freshly or updated/migrated?
- Fresh install during first boot
- Can this issue be replicated on a fresh installation of DietPi?
- yes, it was during fresh install

47fae8cf-5c69-4988-b5ea-392b8145d9b9

#### Steps to reproduce
1. First boot, choose to install anything that needs ffmpeg

#### Expected behaviour
normal boot

#### Actual behaviour
This is error log I get during first boot install

 APT install ffmpeg                                                                                                │
│  - Command: apt-get -y --allow-change-held-packages install ffmpeg                                                │
│  - Exit code: 100                                                                                                 │
│  - DietPi version: v8.19.1 (MichaIng/master) | HW_MODEL: 4 | HW_ARCH: 3 | DISTRO: 7                               │
│  - Error log:                                                                                                     │
│ Reading package lists...                                                                                          │
│ Building dependency tree...                                                                                       │
│ Reading state information...                                                                                      │
│ Some packages could not be installed. This may mean that you have                                                 │
│ requested an impossible situation or if you are using the unstable                                                │
│ distribution that some required packages have not yet been created                                                │
│ or been moved out of Incoming.                                                                                    │
│ The following information may help to resolve the situation:                                                      │
│                                                                                                                   │
│ The following packages have unmet dependencies:                                                                   │
│  ffmpeg : Depends: libavcodec58 (= 8:4.3.6-0+deb11u1+rpt2)                                                        │
│           Depends: libavfilter7 (= 8:4.3.6-0+deb11u1+rpt2)                                                        │
│  libavdevice58 : Depends: libavcodec58 (= 8:4.3.6-0+deb11u1+rpt2)                                                 │
│                  Depends: libavfilter7 (= 8:4.3.6-0+deb11u1+rpt2)                                                 │
│  libavformat58 : Depends: libavcodec58 (= 8:4.3.6-0+deb11u1+rpt2)                                                 │
│                  Depends: libsrt1.4-gnutls (>= 1.4.2) but it is not installable                                   │
│ E: Unable to correct problems, you have held broken packages.

Joulinar, 2023-07-06:
Yes, it's a challenge with current Raspbian apt repository as they don't offer Bookworm packages yet. Therefore we keep them on Bullseye but unfortunately this is causing issues for `ffmpeg` now 😞  https://dietpi.com/forum/t/the-following-packages-have-been-kept-back-ffmpeg/17234/18

MichaIng, 2023-07-10:
Raspberry Pi repository (the one where kernel, firmware, RPi specific builds come from), not "Raspbian" (the ARMv6 clone of Debian) 😉.

Please try this:
```sh
cat << '_EOF_' > /etc/apt/preferences.d/dietpi-ffmpeg
Package: ffmpeg* libav* libpostproc* libsw*
Pin: origin archive.raspberrypi.org
Pin-Priority: -1
_EOF_
apt update
apt install ffmpeg
ffmpeg -version
```

MichaIng, 2023-07-20:
Solved for next DietPi version, including a simple way to enforce the "downgrade" (actually upgrade) to Debian's FFmpeg 5.1 packages via regular APT upgrade: #6492

Live patch up: #6495

MichaIng, 2023-07-20:
Live patch merged.

closed 2023-07-20 by MichaIng