MichaIng/DietPi #6463: Bookworm | Mirror warnings

Arghh, 2023-07-07:
### Creating a bug report/issue

#### Required Information
- DietPi version | `G_DIETPI_VERSION_CORE=8
G_DIETPI_VERSION_SUB=19
G_DIETPI_VERSION_RC=1
G_GITBRANCH='master'
G_GITOWNER='MichaIng'
G_LIVE_PATCH_STATUS[0]='applied'
G_LIVE_PATCH_STATUS[1]='applied'`
- Distro version | `bookworm 1`
- Kernel version | `Linux PiHole 6.1.21-v7+ #1642 SMP Mon Apr  3 17:20:52 BST 2023 armv7l GNU/Linux`
- SBC model | `RPi 3 Model B (armv7l` or (EG: RPi3)
- Power supply used | (EG: 5V 1A RAVpower)
- SD card used | (EG: SanDisk ultra)

#### Steps to reproduce
Update from bullseye to bookworm with the script from dietpi blogpost.

#### Expected behaviour
No warnings while running sudo apt update. 

#### Actual behaviour
Warning while running sudo apt update.
I only have 2 mirrors configurated:
Hit:1 https://mirror.netcologne.de/raspbian/raspbian bookworm InRelease
Hit:2 https://archive.raspberrypi.org/debian bullseye InRelease

`W: https://mirror.netcologne.de/raspbian/raspbian/dists/bookworm/InRelease: Key is stored in legacy trusted.gpg keyring (/etc/apt/trusted.gpg), see the DEPRECATION section in apt-key(8) for details.

W: Skipping acquire of configured file 'non-free-firmware/binary-armhf/Packages' as repository 'https://mirror.netcologne.de/raspbian/raspbian bookworm InRelease' doesn't have the component 'non-free-firmware' (component misspelt in sources.list?)

W: https://archive.raspberrypi.org/debian/dists/bullseye/InRelease: Key is stored in legacy trusted.gpg keyring (/etc/apt/trusted.gpg), see the DEPRECATION section in apt-key(8) for details.)`

#### Extra details
Everything is working fine. So no real problem just a question can the migration to new keys can be done automatically? Also the non-free-firmware package warning is mirror related or?
Otherwise smooth update from bullseye to bookworm. Thank you for all that you do! : )

Joulinar, 2023-07-07:
Is your system already runnig quite a while? Was it upgraded from Buster before? Because in my test system, I don't have massages for `legacy trusted.gpg keyring`

Can you check source configuration

```
for i in /etc/apt/sources.list{,.d/*.list}; do echo "$i:"; cat "$i"; done
```

MichaIng, 2023-07-07:
On old images this may be indeed the case, but we ship new images with all keys in `/etc/apt/sources.list.d` since a while.

The RPi repo key is stored in `/etc/apt/sources.list.d` OOTB and the Raspbian repo key can be linked there (as we do in our installer since a while). Afterwards you can remove the legacy combined keyring:
```sh
ln -sf /usr/share/keyrings/raspbian-archive-keyring.gpg /etc/apt/trusted.gpg.d/raspbian-archive-keyring.gpg
rm /etc/apt/trusted.gpg
apt update
```

I wasn't aware that Raspbian didn't follow Debian with the `non-free-firmware` suite split. Hmm, not sure whether it will remain like this or whether we should remove the suite for ARMv6 images for now?

Joulinar, 2023-07-07:
@MichaIng we could perhaps remove them for the time being to avoid these kinds of warnings and check again at a later date?

MichaIng, 2023-07-07:
You mean the `non-free-firmware` component? The problem is, if we remove it, and Raspbian then made the split, users won't get any firmware updates and might run into dependency errors until we re-add it. So from a security and safety point of view it is better to leave it.

Raspbian also did not add the Trixie suite yet, so I hope that they will align with Debian regarding all the suite and component structure.

Arghh, 2023-07-07:
> Is your system already runnig quite a while? Was it upgraded from Buster before? Because in my test system, I don't have massages for `legacy trusted.gpg keyring`
> 
> Can you check source configuration
> 
> ```
> for i in /etc/apt/sources.list{,.d/*.list}; do echo "$i:"; cat "$i"; done
> ```

yes, since 2020.
yes, I upgraded from Buster to Bullseye and now Booworm

answer to your query for source configuration:
```
/etc/apt/sources.list:
deb https://mirror.netcologne.de/raspbian/raspbian/ bookworm main contrib non-free non-free-firmware
/etc/apt/sources.list.d/raspi.list:
deb https://archive.raspberrypi.org/debian/ bullseye main
```

Joulinar, 2023-07-08:
> /etc/apt/sources.list:
deb https://mirror.netcologne.de/raspbian/raspbian/

It's not related to the issue, but why do you configure Netcologne as mirror directly?

Arghh, 2023-07-08:
what to do you mean? i chose NetCologne as my apt mirror in dietpi-config in network options because Im from Germany. should we just leave it as global mirror director (default)?

Joulinar, 2023-07-09:
Usually there should be no need to specify a dedicated mirror. It should be fine to use global configuration. This should redirect you to the closed mirror automatically.

MichaIng, 2023-07-10:
For the Debian repo, I would agree, as the mirror director works quite well and transparent. For the Raspbian repo I used the very same local mirror for two reasons:
1. I am still not sure if/whether/how (well) `raspbian.raspberrypi.org` really works.
2. It is plain HTTP, being a theoretical security and/or privacy issue. APT package lists are signed and checked separately, and as the public key is preinstalled, there is no way that a malicious package can be installed because of this. But communication is unencrypted and could be answered from any other server (in which case APT would throw an error).