MichaIng/DietPi #6465: Unable to update

itsdukenguyen, 2023-07-09:
#### Details:                                                   - Date           | Sun Jul  9 05:26:33 BST 2023                 - DietPi version | v8.18.2 (MichaIng/master)                    - Image creator  |                                              - Pre-image      |                                              - Hardware       | RPi 4 Model B (aarch64) (ID=4)
- Kernel version | `Linux DietPi 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023 aarch64 GNU/Linux`
- Distro         | bullseye (ID=6,RASPBIAN=0)                   - Command        | `apt-get -y --allow-change-held-packages autopurge rclone`                                                   - Exit code      | 100                                          - Software title | DietPi-Software                              #### Steps to reproduce:                                                             1. ...                                                          2. ...                                                          #### Expected behaviour:                                                                            - ...                                                           #### Actual behaviour:                                                                               - ...                                                           #### Extra details:                                                                                                       - ...                                                           #### Additional logs:                                           ```                                                             Reading package lists...                                        Building dependency tree...                                     Reading state information...                                    The following packages will be REMOVED:                           rclone*                                                       0 upgraded, 0 newly installed, 1 to remove and 0 not upgraded.
2 not fully installed or removed.                               After this operation, 52.9 MB disk space will be freed.
(Reading database ... 36743 files and directories currently installed.)                                                         Removing rclone (1.61.1) ...                                    Setting up mariadb-server-10.5 (1:10.5.19-0+deb11u2) ...        String found where operator expected at /usr/lib/aarch64-linux-gnu/perl-base/File/Temp.pm line 2337, near "#pod If atda5ion to thE MELIUM recurity checkc, also c�eck for$txe#�od posskbidi} gf ``chown() giveaway�'"                                             (Might be a runaway multi-line '' string starting on line 2332)                                                                       (Missing semicolon on previous line?)                   Bareword found where operator expected at /usr/lib/aarch64-linux-gnu/perl-base/File/Temp.pm line 2337, near "#pod If atda5ion to thE MELIUM recurity checkc, also c�eck for$txe#�od posskbidi} gf ``chown() giveaway�' tskng"                                           (Missing operator before tskng?)                        Bareword found where operator expected at /usr/lib/aarch64-linux-gnu/perl-base/File/Temp.pm line 2347, near "3pod"                      (Missing operator before pod?)                          Bareword found where operator expected at /usr/lib/aarch64-linux-gnu/perl-base/File/Temp.pm line 2348, near "2Pem"                      (Missing operator before Pem?)                          Unrecognized character �; marked by <-- HERE after  File:2Pem<-- HERE near column 17 at /usr/lib/aarch64-linux-gnu/perl-base/File/Temp.pm line 2348.                                            Compilation failed in require at /usr/bin/deb-systemd-helper line 87.                                                           BEGIN failed--compilation aborted at /usr/bin/deb-systemd-helper line 87.                                                       dpkg: error processing package mariadb-server-10.5 (--configure):                                                                installed mariadb-server-10.5 package post-installation script subprocess returned error exit status 255
dpkg: dependency problems prevent configuration of mariadb-server:
 mariadb-server depends on mariadb-server-10.5 (>= 1:10.5.19-0+deb11u2); however:
  Package mariadb-server-10.5 is not configured yet.            
dpkg: error processing package mariadb-server (--configure):     dependency problems - leaving unconfigured
Errors were encountered while processing:                        mariadb-server-10.5
 mariadb-server                                                 E: Sub-process /usr/bin/dpkg returned an error code (1)

Joulinar, 2023-07-09:
strange output. Might be file system corruption. Could you check for kernel error messages

```
dmesg -l 0,1,2,3
```

MichaIng, 2025-05-31:
Closing this issue. Feel free to reopen if the problem persists.

closed 2025-05-31 by MichaIng