MichaIng/DietPi #6468: ProFTPD services failed to start after reboot

Rmy1992, 2023-07-11:
- DietPi version | v8.19.1
- SBC model | Raspberry Pi 4 Model B
- Power supply used | Original power supply
- SD card used | Kingston
#### Additional Information (if applicable)
- Software title | ProFTPD

#### Steps to reproduce

1. sudo reboot now
2. ProFTPD service failed to start
3. DietPi-Services  ( (Re)start ALL enabled services )
4. ProFTPD started and working

#### Expected behaviour

ProFTPD services should start after reboot

Joulinar, 2023-07-11:
Can you share following right after reboot

```
journalctl -u proftpd
```

Rmy1992, 2023-07-11:
root@DietPi:~# journalctl -u proftpd
-- No entries --

root@DietPi:~# systemctl status proftpd.service
○ proftpd.service - ProFTPD FTP Server
     Loaded: loaded (/lib/systemd/system/proftpd.service; enabled; preset: enabled)
     Active: inactive (dead)
       Docs: man:proftpd(8)

Rmy1992, 2023-07-11:
> can you try following
> 
> ```
> systemctl disbale proftpd.service
> systemctl enable proftpd.service
> reboot
>Synchronizing state of proftpd.service with SysV service script with /lib/systemd/systemd-sysv-install.
Executing: /lib/systemd/systemd-sysv-install disable proftpd
Removed "/etc/systemd/system/multi-user.target.wants/proftpd.service".
Synchronizing state of proftpd.service with SysV service script with /lib/systemd/systemd-sysv-install.
Executing: /lib/systemd/systemd-sysv-install enable proftpd
Created symlink /etc/systemd/system/multi-user.target.wants/proftpd.service → /lib/systemd/system/proftpd.service.
> ```
> 
> 
> ```
>root@DietPi:~# systemctl status proftpd.service
○ proftpd.service - ProFTPD FTP Server
     Loaded: loaded (/lib/systemd/system/proftpd.service; enabled; preset: enabled)
     Active: inactive (dead)
       Docs: man:proftpd(8)
> ```

Joulinar, 2023-07-11:
looks like it service is not trying to start even if enabled. Can you share service configuration.

```
systemctl cat proftpd.service
```

Rmy1992, 2023-07-11:
root@DietPi:~# systemctl cat proftpd.service
# /lib/systemd/system/proftpd.service
[Unit]
Description=ProFTPD FTP Server
Documentation=man:proftpd(8)
Wants=network-online.target
After=network-online.target nss-lookup.target local-fs.target remote-fs.target

[Service]
Type=forking
Environment=OPTIONS= CONFIG_FILE=/etc/proftpd/proftpd.conf
EnvironmentFile=-/etc/default/proftpd
ExecStartPre=/usr/sbin/proftpd --configtest -c $CONFIG_FILE $OPTIONS
ExecStart=/usr/sbin/proftpd -c $CONFIG_FILE $OPTIONS
ExecReload=/bin/kill -HUP $MAINPID
PIDFile=/run/proftpd.pid

[Install]
WantedBy=multi-user.target

MichaIng, 2023-07-12:
Strange, the command and output clearly shows that it has been enabled, but the status and journal look like it is not enabled. If it failed, there would be some output about this, in both of them. Does the following again produce the same output as before?
```sh
systemctl enable proftpd
```
And is is really enabled after reboot?
```sh
ls -l /etc/systemd/system/multi-user.target.wants/proftpd.service
```
The status output actually indicates this. Are all other services starting as expected?
```sh
systemctl | grep 'service '
```

Do you use a Bullseye or Bookworm and 32-bit or 64-bit system? Basically completing the bug report template would be helpful for trying to replicate:
```sh
uname -m
dpkg --print-architecture
echo $G_DISTRO_NAME
```
I started a test install on all combinations: https://github.com/MichaIng/DietPi/actions/runs/5536385121
RISC-V is expected to fail, as latest systemd somehow fails to be started within a container.
... works well there.

Rmy1992, 2023-07-12:
> ```shell
> root@DietPi:~# systemctl enable proftpd
Synchronizing state of proftpd.service with SysV service script with /lib/systemd/systemd-sysv-install.
Executing: /lib/systemd/systemd-sysv-install enable proftpd
> ```
> 
> And is is really enabled after reboot?
> 
> ```shell
> root@DietPi:~# ls -l /etc/systemd/system/multi-user.target.wants/proftpd.service
lrwxrwxrwx 1 root root 35 Jul 11 16:12 /etc/systemd/system/multi-user.target.wan                                                                                                                                                             ts/proftpd.service -> /lib/systemd/system/proftpd.service
> ```
> 
> The status output actually indicates this. Are all other services starting as expected?
> 
> ```shell
> root@DietPi:~# systemctl | grep 'service '
  console-setup.service                                                                                                                                                                                                                                                                                           loaded acti                                                                                                                                                             ve exited    Set console font and keymap
  cron.service                                                                                                                                                                                                                                                                                                    loaded acti                                                                                                                                                             ve running   Regular background program processing daemon
  dietpi-postboot.service                                                                                                                                                                                                                                                                                         loaded acti                                                                                                                                                             ve exited    DietPi-PostBoot
  dietpi-preboot.service                                                                                                                                                                                                                                                                                          loaded acti                                                                                                                                                             ve exited    DietPi-PreBoot
  dietpi-ramlog.service                                                                                                                                                                                                                                                                                           loaded acti                                                                                                                                                             ve exited    DietPi-RAMlog
  dropbear.service                                                                                                                                                                                                                                                                                                loaded acti                                                                                                                                                             ve running   Lightweight SSH server
  fake-hwclock.service                                                                                                                                                                                                                                                                                            loaded acti                                                                                                                                                             ve exited    Restore / save the current clock
  getty@tty1.service                                                                                                                                                                                                                                                                                              loaded acti                                                                                                                                                             ve running   Getty on tty1
  ifup@eth0.service                                                                                                                                                                                                                                                                                               loaded acti                                                                                                                                                             ve exited    ifup for eth0
[... cut ...]
       Docs: man:proftpd(8)

> same result, after reboot ProFTPD service doesn't start
> Need manually Dietpi-Services ( (Re)start ALL enabled services )
> OK  ] DietPi-Services | restart : proftpd
[  OK  ] DietPi-Services | restart : minidlna
[  OK  ] DietPi-Services | restart : qbittorrent
[  OK  ] DietPi-Services | restart : cron
[  OK  ] DietPi-Services | restart : dropbear
[ INFO ] DietPi-Services | skip : dietpi-vpn (disabled)
[ INFO ] DietPi-Services | skip : dietpi-cloudshell (disabled)

MichaIng, 2023-07-13:
Spooky 👻. `systemctl` does not show the service at all, i.e. it was not loaded at all during boot, even that `systemctl status/enable/cat proftpd` does find/handle it correctly.

Not idea how this is possible. Probably reinstalling the package helps:
```sh
apt install --reinstall proftpd-basic
```

Rmy1992, 2023-07-13:
root@DietPi:~# apt install --reinstall proftpd-basic
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
0 upgraded, 0 newly installed, 1 reinstalled, 0 to remove and 0 not upgraded.
Need to get 455 kB of archives.
After this operation, 0 B of additional disk space will be used.
Get:1 https://deb.debian.org/debian bookworm/main arm64 proftpd-basic all 1.3.8+dfsg-4 [455 kB]
Fetched 455 kB in 0s (1026 kB/s)
debconf: delaying package configuration, since apt-utils is not installed
(Reading database ... 19467 files and directories currently installed.)
Preparing to unpack .../proftpd-basic_1.3.8+dfsg-4_all.deb ...
Unpacking proftpd-basic (1.3.8+dfsg-4) over (1.3.8+dfsg-4) ...
Setting up proftpd-basic (1.3.8+dfsg-4) ...

sudo reboot now

root@DietPi:~# systemctl status proftpd.service
○ proftpd.service - ProFTPD FTP Server
     Loaded: loaded (/lib/systemd/system/proftpd.service; enabled; preset: enabl                                                                                                                                                             ed)
     Active: inactive (dead)
       Docs: man:proftpd(8)

And thank you for your time MichaIng & Joulinar !

MichaIng, 2023-07-13:
Is there an override file?
```sh
ls -l /etc/systemd/system/proftpd.service
```
If not, could you try to create one and see whether this loads at boot?
```sh
cat /lib/systemd/system/proftpd.service > /etc/systemd/system/proftpd.service
systemctl disable proftpd
systemctl enable proftpd
reboot
```

Rmy1992, 2023-07-13:
root@DietPi:~# ls -l /etc/systemd/system/proftpd.service
ls: cannot access '/etc/systemd/system/proftpd.service': No such file or directory

root@DietPi:~# cat /lib/systemd/system/proftpd.service > /etc/systemd/system/proftpd.service
systemctl disable proftpd
systemctl enable proftpd
reboot
Synchronizing state of proftpd.service with SysV service script with /lib/systemd/systemd-sysv-install.
Executing: /lib/systemd/systemd-sysv-install disable proftpd
Removed "/etc/systemd/system/multi-user.target.wants/proftpd.service".
Synchronizing state of proftpd.service with SysV service script with /lib/systemd/systemd-sysv-install.
Executing: /lib/systemd/systemd-sysv-install enable proftpd
Created symlink /etc/systemd/system/multi-user.target.wants/proftpd.service → /etc/systemd/system/proftpd.service.

sudo reboot now

root@DietPi:~# ls -l /etc/systemd/system/proftpd.service
-rw-r--r-- 1 root root 515 Jul 14 05:18 /etc/systemd/system/proftpd.service

root@DietPi:~# systemctl status proftpd.service
○ proftpd.service - ProFTPD FTP Server
     Loaded: loaded (/etc/systemd/system/proftpd.service; enabled; preset: enabled)
     Active: inactive (dead)
       Docs: man:proftpd(8)

MichaIng, 2023-07-13:
Okay I have no idea, even with the service completely recreated at a different location it is like it does not exist at boot 🤔. Do you have a dropin config for it?
```sh
ls -l /etc/systemd/system/proftpd.service.d
```

Rmy1992, 2023-07-14:
root@DietPi:~# ls -l /etc/systemd/system/proftpd.service.d
ls: cannot access '/etc/systemd/system/proftpd.service.d': No such file or directory

Joulinar, 2023-07-14:
ok it's a bug on Debian Bookworm. Hopefully, they will release updated version `1.3.8+dfsg-`5 soon

MichaIng, 2023-09-03:
Solved with deb12u1 patch: https://metadata.ftp-master.debian.org/changelogs//main/p/proftpd-dfsg/proftpd-dfsg_1.3.8+dfsg-4+deb12u1_changelog

closed 2023-09-03 by MichaIng