MichaIng/DietPi #6470: Upgrade fails from 18.17.2 to 18.19.1 even though network is working

mfrazzz, 2023-07-11:
### Creating a bug report/issue

#### Required Information
- DietPi version | 18.17.2
- Distro version | bullseye 0
- Kernel version | Linux Sonarr 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023 aarch64 GNU/Linux
- SBC model | RPi 4 Model B (aarch64)
- Power supply used | (EG: 5V 1A RAVpower)
- SD card used | Samsung EVO

#### Steps to reproduce

1. ... sudo dietpi-update
2. ...

#### Expected behaviour

- ... Should allow me to do the update

#### Actual behaviour

- ... Fails saying it can't connect, but other internet functions work

#### Extra details

- ...

#### Details:
- Date           | Tue Jul 11 11:18:43 MDT 2023
- DietPi version | v8.17.2 (MichaIng/master)
- Image creator  |
- Pre-image      |
- Hardware       | RPi 4 Model B (aarch64) (ID=4)
- Kernel version | `Linux Sonarr 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023 aarch64 GNU/Linux`
- Distro         | bullseye (ID=6,RASPBIAN=0)
- Command        | `ping -4nc 1 -W 10 9.9.9.9`
- Exit code      | 1
- Software title | DietPi-Update
#### Steps to reproduce:

1. ...
2. ...
#### Expected behaviour:

- ...
#### Actual behaviour:

- ...
#### Extra details:

- ...
#### Additional logs:
```
PING 9.9.9.9 (9.9.9.9) 56(84) bytes of data.
From 172.27.1.246 icmp_seq=1 Destination Port Unreachable
ping: sendmsg: Operation not permitted

--- 9.9.9.9 ping statistics ---
1 packets transmitted, 0 received, +1 errors, 100% packet loss, time 0ms
`

Joulinar, 2023-07-11:
Can you try following on CLI

```
ping -4nc 1 -W 10 9.9.9.9
ping -4nc 1 -W 10 8.8.8.8
ping -4nc 1 -W 10 1.1.1.1
```

mfrazzz, 2023-07-11:
Not sure what happened.  Did multiple reboots and turning wifi on and off, and finally got it to connect and complete the upgrade.

closed 2023-07-11 by mfrazzz