MichaIng/DietPi #6472: unattended install (automatic) of wireguard (172) - choose weither is Server or Client

nudeldudel, 2023-07-12:
With unattended_pivpn.conf there is a way to install pivpn autimaticly.

For my usecase I need to install Wireguard (172) automatic, but it fault because of the Question, weither it shall be a Server or a CLient.

Is ther any way to jump over this step, set the chose by enviroment oder anything else to install it tuomaticly using the dietpi.txt

If not, is there a chance to impelent it?

Target: the automatic installed device is one of many clients in a VPN an fetch ist CONF after installing wireguard by a postinstall script.

MichaIng, 2023-07-12:
There is currently no such possibility, but I agree that it would be great. There is a section in [dietpi.txt](https://github.com/MichaIng/DietPi/blob/dev/dietpi.txt#L231) where a commented setting could be added to enforce either server of client install of WireGuard.

nudeldudel, 2023-07-15:
Ok, thanks for the clarification.