MichaIng/DietPi #6476: Mounted drives lost/unavailable

rainfallsevensamurai, 2023-07-13:
### Creating a bug report/issue

#### Required Information
- DietPi version | 
```
G_DIETPI_VERSION_CORE=8
G_DIETPI_VERSION_SUB=19
G_DIETPI_VERSION_RC=1
G_GITBRANCH='master'
G_GITOWNER='MichaIng'
G_LIVE_PATCH_STATUS[0]='applied'
G_LIVE_PATCH_STATUS[1]='applied'
```

- Distro version | `bullseye`

- Kernel version | `Linux DietPi 6.1.11-meson64 #23.02.2 SMP PREEMPT Sat Feb 18 00:07:55 UTC 2023 aarch64 GNU/Linux`
- SBC model | `Odroid C4/HC4 (aarch64)`
- Power supply used | 
- SD card used | None

#### Additional Information (if applicable)
- Software title | (EG: Nextcloud)
- Was the software title installed freshly or updated/migrated?
- Can this issue be replicated on a fresh installation of DietPi?

- Bug report ID | `echo $G_HW_UUID`

#### Steps to reproduce

1. Mount a drive from Synology NAS using dietpi_Drive_Manager
2. Check ls -al /mnt/, drive is ready and usable
3. Wait
4. Check again, drive is no longer available:

```
drwxrwx---  2 dietpi dietpi    0 Mar 23 10:48 audio
drwxrwxr-x 12 dietpi dietpi 4096 Apr  1 17:57 dietpi_userdata
drwxrwx---  2 dietpi dietpi    0 Aug 29  2022 downloads
drwxrwxr-x  2 dietpi dietpi 4096 Jul 31  2022 ftp_client
drwxrwxr-x  2 dietpi dietpi 4096 Jul 31  2022 nfs_client
drwxrwx---  2 dietpi dietpi    0 Aug 29  2021 video
```

#### Expected behaviour

- The drives should continue to be available and usable for radarr or Plex or whatever, this has been working unattended for months

#### Actual behaviour

- The drives are not readable and applications fail

#### Extra details

- Since latest updates

Joulinar, 2023-07-13:
There is nothing on our scripts that automatically unmount drives/shares. What is the mount point of your share? 

Better to check using `df -h` for available shares.

Can you have a look to systems logs using `journalctl` or `dmesg` for related messages?

rainfallsevensamurai, 2023-07-13:
`dmesg`:

```
[   12.897358] CIFS: Attempting to mount \\192.168.1.x\downloads
[   32.899046] CIFS: Attempting to mount \\192.168.1.x\video
[   33.425788] CIFS: Attempting to mount \\192.168.1.x\audio
```

It seems to work now, however it takes a long time to mount. Could this be related to Synology?

MichaIng, 2023-07-13:
Could be some network and/or drive bottleneck. Did you check logs at the Synology?

AFAIK there are no other related system logs in `journalctl` aside of the CIFS kernel messages you see in `dmesg` already, right?

MichaIng, 2025-05-31:
Closing this issue. Feel free to reopen if the problem persists.

closed 2025-05-31 by MichaIng