MichaIng/DietPi #6477: Deluge error "findCaller() takes from 1 to 2 positional arguments but 3 were given"

rainfallsevensamurai, 2023-07-13:
### Creating a bug report/issue

#### Required Information
- DietPi version | 

```
G_DIETPI_VERSION_CORE=8
G_DIETPI_VERSION_SUB=19
G_DIETPI_VERSION_RC=1
G_GITBRANCH='master'
G_GITOWNER='MichaIng'
G_LIVE_PATCH_STATUS[0]='applied'
G_LIVE_PATCH_STATUS[1]='applied'
```

- Distro version | `bullseye`
- Kernel version | `Linux DietPi 6.1.11-meson64 #23.02.2 SMP PREEMPT Sat Feb 18 00:07:55 UTC 2023 aarch64 GNU/Linux`
- SBC model | `Odroid C4/HC4 (aarch64)` 
- Power supply used | (EG: 5V 1A RAVpower)
- SD card used | (EG: SanDisk ultra)

#### Additional Information (if applicable)
- Software title | Deluge
- Was the software title installed freshly or updated/migrated? Freshly reinstalled
- Can this issue be replicated on a fresh installation of DietPi?

- Bug report ID | `echo $G_HW_UUID`

#### Steps to reproduce

1. Install Deluge
2. Add user deluge
3. Change ownership of folder /mnt/dietpi_userdata/deluge to 'deluge'
4. Change auth file in /mnt/dietpi_userdata/deluge/.config/deluge/auth
5. Restart service
6. run `sudo -u deluge deluge-console`

#### Expected behaviour

- Working as it has been working for months

#### Actual behaviour

- Error:

```
root@DietPi:/mnt/dietpi_userdata/deluge# sudo -u deluge deluge-console
Unhandled error in Deferred:

Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/twisted/internet/defer.py", line 654, in _runCallbacks
    current.result = callback(current.result, *args, **kw)
  File "/usr/lib/python3/dist-packages/deluge/ui/client.py", line 408, in __on_connect_fail
    self.daemon_info_deferred.errback(reason)
  File "/usr/lib/python3/dist-packages/twisted/internet/defer.py", line 501, in errback
    self._startRunCallbacks(fail)
  File "/usr/lib/python3/dist-packages/twisted/internet/defer.py", line 568, in _startRunCallbacks
    self._runCallbacks()
--- <exception caught here> ---
  File "/usr/lib/python3/dist-packages/twisted/internet/defer.py", line 654, in _runCallbacks
    current.result = callback(current.result, *args, **kw)
  File "/usr/lib/python3/dist-packages/deluge/ui/console/modes/connectionmanager.py", line 90, in _on_connect_fail
    self.report_message('Failed to connect!', result)
  File "/usr/lib/python3/dist-packages/deluge/ui/console/widgets/popup.py", line 61, in report_message
    self.push_popup(MessagePopup(self, title, message))
  File "/usr/lib/python3/dist-packages/deluge/ui/console/widgets/popup.py", line 354, in __init__
    lns = format_utils.wrap_string(self.message, self.width - 3, 3, True)
  File "/usr/lib/python3/dist-packages/deluge/ui/console/utils/format_utils.py", line 207, in wrap_string
    s1 = string.split('\n')
builtins.AttributeError: 'Failure' object has no attribute 'split'

Temporarily disabling observer LegacyLogObserverWrapper(<bound method TwistedLoggingObserver.emit of <deluge.log.TwistedLoggingObserver object at 0xffffa3ae0130>>) due to exception: [Failure instance: Traceback: <class 'TypeError'>: findCaller() takes from 1 to 2 positional arguments but 3 were given
/usr/lib/python3/dist-packages/twisted/internet/defer.py:962:__del__
/usr/lib/python3/dist-packages/twisted/logger/_logger.py:190:failure
/usr/lib/python3/dist-packages/twisted/logger/_logger.py:144:emit
--- <exception caught here> ---
/usr/lib/python3/dist-packages/twisted/logger/_observer.py:131:__call__
/usr/lib/python3/dist-packages/twisted/logger/_legacy.py:93:__call__
/usr/lib/python3/dist-packages/deluge/log.py:204:emit
/usr/lib/python3.9/logging/__init__.py:1489:critical
/usr/lib/python3.9/logging/__init__.py:1573:_log
]
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/twisted/internet/defer.py", line 962, in __del__
    log.failure(format,
  File "/usr/lib/python3/dist-packages/twisted/logger/_logger.py", line 190, in failure
    self.emit(level, format, log_failure=failure, **kwargs)
  File "/usr/lib/python3/dist-packages/twisted/logger/_logger.py", line 144, in emit
    self.observer(event)
--- <exception caught here> ---
  File "/usr/lib/python3/dist-packages/twisted/logger/_observer.py", line 131, in __call__
    observer(event)
  File "/usr/lib/python3/dist-packages/twisted/logger/_legacy.py", line 93, in __call__
    self.legacyObserver(event)
  File "/usr/lib/python3/dist-packages/deluge/log.py", line 204, in emit
    getattr(LoggingLoggerClass, event_dict['log_level'].name)(
  File "/usr/lib/python3.9/logging/__init__.py", line 1489, in critical
    self._log(CRITICAL, msg, args, **kwargs)
  File "/usr/lib/python3.9/logging/__init__.py", line 1573, in _log
    fn, lno, func, sinfo = self.findCaller(stack_info, stacklevel)
builtins.TypeError: findCaller() takes from 1 to 2 positional arguments but 3 were given
```

#### Extra details

- ...

rainfallsevensamurai, 2023-07-13:
Extra information: I setup deluge to use a virtual network to connect via VPN like this:

ip link add name vethhost0 type veth peer name vethvpn0
ip link set vethvpn0 netns deluge
ip addr add 10.0.0.1/24 dev vethhost0
ip netns exec deluge ip addr add 10.0.0.2/24 dev vethvpn0
ip link set vethhost0 up
ip netns exec deluge ip link set vethvpn0 up

ip netns exec deluge ip route add default via 10.0.0.1

MichaIng, 2023-07-13:
Hmm, since the service runs as `debian-deluged` user, does it even run after you changed permissions to the new `deluge` user?
```sh
journalctl -u deluged
```
Does it work when you use the pre-created user, in case after reverting the owner change?
```sh
sudo -u debian-deluged deluge-console
```

rainfallsevensamurai, 2023-08-04:
Sorry for the delay...

I can't use the default user because a script which configures wireguard settings, doesn't accept long usernames. Bad excuse, I know. 

The service is running as user 'deluge':
[image]

Status:
```
root@DietPi:/mnt/dietpi_userdata/dxld-mullvad-master# systemctl status deluged.service
● deluged.service - Deluge Daemon (DietPi)
     Loaded: loaded (/etc/systemd/system/deluged.service; enabled; vendor preset: enabled)
     Active: active (running) since Fri 2023-08-04 07:33:39 BST; 32min ago
       Docs: man:deluged
   Main PID: 20284 (deluged)
      Tasks: 4 (limit: 4215)
     Memory: 87.6M
        CPU: 32.446s
     CGroup: /system.slice/deluged.service
             └─20284 /usr/bin/python3 /usr/bin/deluged -d

Aug 04 07:33:50 DietPi deluged[20284]:     observer(event)
Aug 04 07:33:50 DietPi deluged[20284]:   File "/usr/lib/python3/dist-packages/twisted/logger/_legacy.py", line 93, in __call__
Aug 04 07:33:50 DietPi deluged[20284]:     self.legacyObserver(event)
Aug 04 07:33:50 DietPi deluged[20284]:   File "/usr/lib/python3/dist-packages/deluge/log.py", line 204, in emit
Aug 04 07:33:50 DietPi deluged[20284]:     getattr(LoggingLoggerClass, event_dict['log_level'].name)(
Aug 04 07:33:50 DietPi deluged[20284]:   File "/usr/lib/python3.9/logging/__init__.py", line 1489, in critical
Aug 04 07:33:50 DietPi deluged[20284]:     self._log(CRITICAL, msg, args, **kwargs)
Aug 04 07:33:50 DietPi deluged[20284]:   File "/usr/lib/python3.9/logging/__init__.py", line 1573, in _log
Aug 04 07:33:50 DietPi deluged[20284]:     fn, lno, func, sinfo = self.findCaller(stack_info, stacklevel)
Aug 04 07:33:50 DietPi deluged[20284]: builtins.TypeError: findCaller() takes from 1 to 2 positional arguments but 3 were given
root@DietPi:/mnt/dietpi_userdata/dxld-mullvad-master#
```

rainfallsevensamurai, 2023-08-04:
Fixed by manually editing /usr/lib/python3/dist-packages/deluge/log.py based on this post: https://git.deluge-torrent.org/deluge/commit/?h=develop&id=351664ec071daa04161577c6a1c949ed0f2c3206 and systemctl restart deluged

closed 2023-08-04 by rainfallsevensamurai

MichaIng, 2023-09-01:
Strange. The commit sounds like it is a general issue with Python >3.8. But Bullseye ships Python 3.9 and all these years I do not remeber anyone facing this error, hence am 99.99% sure such major incompatibility with the Debian version's default Python version would have been patches by Debian package maintainers, or that the upstream version with the upstream fix would be provided.

Did you change anything else in default configs apart of the user name, so that e.g. this function is now called while it is not on a fresh install or so?