MichaIng/DietPi #6478: Error with Kodi while upgrading to Bookworm

Niaz89, 2023-07-13:
#### Details:
- Date           | Thu Jul 13 14:23:40 CEST 2023
- DietPi version | v8.19.1 (MichaIng/master)
- Image creator  | 
- Pre-image      | 
- Hardware       | RPi 4 Model B (aarch64) (ID=4)
- Kernel version | `Linux RPi8 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023 aarch64 GNU/Linux`
- Distro         | bullseye (ID=6,RASPBIAN=0)
- Command        | `apt-get -y --with-new-pkgs upgrade`
- Exit code      | 100
- Software title | dietpi-bookworm-upgrade
#### Steps to reproduce:

1. I've ran the upgrade script
#### Expected behaviour:

- upgrade to Bookworm without errors
#### Actual behaviour:

 Errors were encountered while processing:
 /tmp/apt-dpkg-install-qkzdWu/232-kodi_2%3a20.1+dfsg-1_arm64.deb
 /tmp/apt-dpkg-install-qkzdWu/233-kodi-bin_2%3a20.1+dfsg-1_arm64.deb
 /tmp/apt-dpkg-install-qkzdWu/257-kodi-data_2%3a20.1+dfsg-1_all.deb
E: Sub-process /usr/bin/dpkg returned an error code (1)
#### Extra details:

- After this error, I've decided to uninstall Kodi. Before the uninstall the upgrade to Bookworm seemingly continued. Not sure in what state my system is. For example, Nextcloud is not working.
#### Additional logs:
Because of the limit of 65536 characters. the log is here:
https://qtext.io/sv99
If you have better idea where to share, let me know.

Niaz89, 2023-07-13:
Error when trying to access Nextcloud:

Internal Server Error

The server encountered an internal error and was unable to complete your request.
Please contact the server administrator if this error reappears multiple times, please include the technical details below in your report.
More details can be found in the server log.

qBittorent Web UI is accesible. SMB share is working.

MichaIng, 2023-07-13:
We saw these kind of APT errors before when there was a file conflict, i.e. two packages serving the same file: https://dietpi.com/forum/t/upgrading-dietpi-from-bullseye-to-bookworm/15963/16

But very unlikely in case of Kodi, it could have other reasons. I guess reinstalling Kodi now will work well.

To check about the Nextcloud error:
```sh
journalctl -u php8.2-fpm
```
Which webserver are you using?

Niaz89, 2023-07-13:
Oh, yeah, Pastebin. That's the service I couldn't remember. Here: https://pastebin.com/EabkDLXg

 DietPi-Services
─────────────────────────────────────────────────────
 Mode: status 

[  OK  ] DietPi-Services | avahi-daemon         active (running) since Thu 2023-07-13 16:53:29 CEST; 5h 56min ago
[  OK  ] DietPi-Services | nmbd                 active (running) since Thu 2023-07-13 16:53:31 CEST; 5h 56min ago
[  OK  ] DietPi-Services | smbd                 active (running) since Thu 2023-07-13 16:53:31 CEST; 5h 56min ago
[FAILED] DietPi-Services | × redis-server.service - Advanced key-value store
     Loaded: loaded (/lib/systemd/system/redis-server.service; enabled; preset: enabled)
    Drop-In: /etc/systemd/system/redis-server.service.d
             └─dietpi.conf
     Active: failed (Result: exit-code) since Thu 2023-07-13 16:53:33 CEST; 5h 56min ago
       Docs: http://redis.io/documentation,
             man:redis-server(1)
    Process: 889 ExecStart=/usr/bin/redis-server /etc/redis/redis.conf --supervised systemd --daemonize no (code=exited, status=1/FAILURE)
   Main PID: 889 (code=exited, status=1/FAILURE)
     Status: "Redis is loading..."
        CPU: 195ms

Jul 13 16:53:33 RPi8 systemd[1]: redis-server.service: Scheduled restart job, restart counter is at 5.
Jul 13 16:53:33 RPi8 systemd[1]: Stopped redis-server.service - Advanced key-value store.
Jul 13 16:53:33 RPi8 systemd[1]: redis-server.service: Start request repeated too quickly.
Jul 13 16:53:33 RPi8 systemd[1]: redis-server.service: Failed with result 'exit-code'.
Jul 13 16:53:33 RPi8 systemd[1]: Failed to start redis-server.service - Advanced key-value store.
[  OK  ] DietPi-Services | mariadb              active (running) since Thu 2023-07-13 16:53:31 CEST; 5h 56min ago
[  OK  ] DietPi-Services | php7.4-fpm           active (running) since Thu 2023-07-13 16:53:32 CEST; 5h 56min ago
[  OK  ] DietPi-Services | apache2              active (running) since Thu 2023-07-13 16:53:32 CEST; 5h 56min ago
[  OK  ] DietPi-Services | qbittorrent          active (running) since Thu 2023-07-13 16:53:30 CEST; 5h 56min ago
[  OK  ] DietPi-Services | cron                 active (running) since Thu 2023-07-13 16:53:31 CEST; 5h 56min ago
[  OK  ] DietPi-Services | ssh                  active (running) since Thu 2023-07-13 16:53:30 CEST; 5h 56min ago
[ INFO ] DietPi-Services | dietpi-vpn           inactive (dead)
[ INFO ] DietPi-Services | dietpi-cloudshell    inactive (dead)
[  OK  ] DietPi-Services | dietpi-ramlog        active (exited) since Thu 2023-07-13 16:53:30 CEST; 5h 56min ago
[  OK  ] DietPi-Services | dietpi-preboot       active (exited) since Thu 2023-07-13 16:53:30 CEST; 5h 56min ago
[  OK  ] DietPi-Services | dietpi-postboot      active (exited) since Thu 2023-07-13 16:53:30 CEST; 5h 56min ago
[ INFO ] DietPi-Services | dietpi-wifi-monitor  inactive (dead)

I have Apache. I think the whole problem is with PHP. Since when I do:
/etc/php/8.2# dir
I get:
cli  mods-available  phpdbg

There's no FPM. I think there was some hiccup in the script because of the error and the FPM didn't get installed and thus the config couldn't get copied.

But "php -v" gives me:
PHP 8.2.7 (cli) (built: Jun  9 2023 19:37:27) (NTS)
Copyright (c) The PHP Group
Zend Engine v4.2.7, Copyright (c) Zend Technologies
    with Zend OPcache v8.2.7, Copyright (c), by Zend Technologies

But if I go to domain/phpinfo.php it says "PHP Version 7.4.33"

MichaIng, 2023-07-13:
Interesting, the RPi Foundation splits files differently across `kodi`, `kodi-bin` and `kodi-data` packages than Debian does 😄. You need to remove those packages first:
```sh
apt remove kodi kodi-bin kodi-data
```
Then repeat the upgrade script, after which I guess/hope all services run again as they should.

MichaIng, 2023-07-13:
This should patch should avoid the issue. But need to be merged into the master branch first: https://github.com/MichaIng/DietPi/commit/0107477

Niaz89, 2023-07-14:
I've restored a backup, removed Kodi first and ran thew script again. Everything went perfectly this time. Thank you!

closed 2023-07-29 by MichaIng